syssi/esphome-jk-bms #356: B2A20S20P fault connectionn

alesssionico, 2023-07-26:
Hello, I tried to install as per the guide https://github.com/syssi/esphome-jk-bms/blob/main/esp32-example.yaml everything is installed correctly, using Raspberry p3+ usb module with FTDI chip and jk module RS485.
After installing everything following the installation wizard esp + Raspberry, enter the MQTT user and password, SSD and WiFi password and although everything is successful, this error string comes out

`SUCCESS] Took 1900.72 seconds ========================
INFO Successfully compiled program.
esptool.py v4.6.2
Serial port /dev/ttyUSB0
Connecting......................................
ERROR Running command failed: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 460800 --port /dev/ttyUSB0 --chip esp32 write_flash -z --flash_size detect 0x10000 /data/jk-bms/.pioenvs/jk-bms/firmware.bin 0x1000 /data/jk-bms/.pioenvs/jk-bms/bootloader.bin 0x8000 /data/jk-bms/.pioenvs/jk-bms/partitions.bin 0x9000 /data/jk-bms/.pioenvs/jk-bms/ota_data_initial.bin locally.
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
esptool.py v4.6.2
Serial port /dev/ttyUSB0
Connecting......................................
ERROR Running command failed: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 115200 --port /dev/ttyUSB0 --chip esp32 write_flash -z --flash_size detect 0x10000 /data/jk-bms/.pioenvs/jk-bms/firmware.bin 0x1000 /data/jk-bms/.pioenvs/jk-bms/bootloader.bin 0x8000 /data/jk-bms/.pioenvs/jk-bms/partitions.bin 0x9000 /data/jk-bms/.pioenvs/jk-bms/ota_data_initial.bin locally.

`

syssi, 2023-07-26:
The flash tool is unable to talk to your ESP32 via `/dev/ttyUSB0`. Which ESP32 board do you own? Are you able to flash the ESP in general? Which GPIOs did you use to connect the BMS?

alesssionico, 2023-07-26:
hi, yes i flashed following the.information step by step when creating
a new connection. all components are installed on the home assistant.
as a module I'm using a common rs485 to usb converter, so I didn't
change the gpio parameters that were set. I attach FTDI converter
image

syssi, 2023-07-26:
Could you rephrase your feedback and describe your setup in detail? I hope you are using a FTDI based UART-TTL to USB adapter and no RS485 converter. If the converter isn't part of your ESP board please don't forget to boot the ESP into flash mode.

alesssionico, 2023-07-26:
you are right, i used FTDI rs485 usb converter and not usb UART. I
used a converter I had at home. This is not good because I have no
driver install on home assistant or because it can not communicate ?
at this point it is better to buy an esp32 module and I solve it since
the problem is only of communication. what do you think?

alesssionico, 2023-07-26:
[image]
[logs_jj_run.txt](https://github.com/syssi/esphome-jk-bms/files/12168974/logs_jj_run.txt)

syssi, 2023-08-09:
You have to use a "UART-TTL to USB" adapter to flash your ESP if it's not equipped with such a chip (ch430, pl2303, etc.). If you buy a ESP32 dev board the microcontroller can be flashed using a micro USB cable without hassle.