syssi/esphome-jk-bms #360: CRC Check Fail after months of working almost properly

DragotaDarius, 2023-08-02:
Hi,
I have been using your implementation with an esp8266(nodemcu v3 Variant) with direct connection to the bms and It was working almost perfectly, excluding the during the charging where I was always getting the crc check fail.
Here is a small dump of the logs, but I cannot see anything relevant other than nan for every value. 
I want to mention that nothing changed on the hardware side, since it started to not work.
[logs_esphome-web-0bbb4b_run (1).txt](https://github.com/syssi/esphome-jk-bms/files/12236527/logs_esphome-web-0bbb4b_run.1.txt)

DragotaDarius, 2023-08-02:
Here is the logs with the raw data to and from the bms:
[logs_esphome-web-0bbb4b_logs.txt](https://github.com/syssi/esphome-jk-bms/files/12236616/logs_esphome-web-0bbb4b_logs.txt)

syssi, 2023-08-02:
Some ideas:

1. Try to reduce the cable length between the ESP and the BMS
2. Try to use a ESP32 just for testing. There a some users here which cannot communicate with the BMS using a ESP8266 but a ESP32 works flawlessly for unknown reasons.

DragotaDarius, 2023-08-02:
> Some ideas:
> 
> 1. Try to reduce the cable length between the ESP and the BMS
> 2. Try to use a ESP32 just for testing. There a some users here which cannot communicate with the BMS using a ESP8266 but a ESP32 works flawlessly for unknown reasons.

I will try by the end of this week, but it is strange that it worked for a long time and suddenly stopped :)
The length is very short, around 10-15cm if I remember correctly.

syssi, 2023-08-09:
I've updated my setup to ESPHome 2023.7.1 + `esphome-jk-bms.git@main` today to reproduce your issue.

syssi, 2023-08-10:
My setup isn't affected. The ESP is stable (no availability drop-outs) and no CRC errrors since 24h.

DragotaDarius, 2023-08-10:
Thank you for your help. 
Sadly I couldn't manage to solve the error and I have finally switched to the esp32 and I am finally able to also view the stats while charging, which didn't work at all in the previous setup.
[image]
Thank you for your help.

syssi, 2023-08-10:
Perfect! In other words: The hardware UART of th ESP32 works much better as the one of the ESP8266?

closed 2023-08-10 by syssi

DragotaDarius, 2023-08-10:
> Perfect! In other words: The hardware UART of th ESP32 works much better as the one of the ESP8266?

I think we can definetely say that. 
I kept the exact same wires, just resoldered to the esp32 and it works since I moved to the new "platform".
This is a screenshot of the logs, which look very good so far:
[image]

And to maybe help other people, here is the exact picture of the esp32 variant that I have:
[image]

Here is also my .yaml for anyone that may need it.
I had to put `power_save_mode: none` because I had poor coverage with the default, which I think it is something like "Light" or similar.
Even tho the board is not the right one, so far it worked without a problem.
I wasn't able to use the ble because of the interference with the wifi connection, by having my esp32 under 20cm of reinforced concrete, in the underground garage.

`substitutions:
  name: esphome-web-68c0ec
  # friendly_name: BMS ESP32
  device_description: "Monitor a JK-BMS via UART-TTL"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO01
  rx_pin: GPIO03

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.5.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

ota:

logger:
  level: INFO

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

api:
  encryption:
    key: "Redacted, Inser yors"
  reboot_timeout: 10s

uart:
  - id: uart_0
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

jk_modbus:
[... 191 lines cut ...]
switch:
  - platform: jk_bms
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

text_sensor:
  - platform: jk_bms
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    battery_type:
      name: "${name} battery type"
    password:
      name: "${name} password"
    device_type:
      name: "${name} device type"
    software_version:
      name: "${name} software version"
    manufacturer:
      name: "${name} manufacturer"
    total_runtime_formatted:
      name: "${name} total runtime formatted"`

GeminiAntares, 2024-09-01:
Hello,
i know the issue is already closed. However, I am expiring same problems. First, I was running it with ESP8266, then I switched to ESP32 (even same board as used by Dragados). And yes, it got much better, but still having this problem intermitten (average every 5 minutes). Same thing here: wondering since it was working perfectly fine (ESP8266) and the behavior changed. I see changes have been made in "jk_modbus.cpp". Any older version available?