syssi/esphome-jk-bms #361: help with remaining capacityy template

magnetus26, 2023-08-06:
hello, Im trying to create a temple to display my battery
remaining capacity in Ah.

here is my code:

```
- platform: template 
   name: " ${devicename} rem. capacity " 
    id: rc 
    unit_of_measurement: Ah 
    device_class: energy 
    state_class: total_increasing 
    accuracy_decimals: ${accuracy} 
    lambda: |- 
          float i = 0.0;  
          float e = id(bv).state; 
          float w = id(kWhc).state; 
          if (w > 0) { 
            i =  w / e; 
          } 
          return {i}; 
```
but I get the flollowing error:

[image]

I do have a similar template for battery charge in kWh and that one works:

```
- platform: template
    name: " ${devicename} charge "
    id: kWhc
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: ${accuracy}
    lambda: |-
          float i = 264 + id(consumed_amp_hours).state; 
          float e = id(bv).state;
          float w = 0.0;
          if (i > 0) {
            w =  i * e;
          }
          return {w};
I just added the id:kWhc to name: " ${devicename} charge "
```
ps. 
this is from my smartshunt device from [VictronSmartShunt-ESPHOME](https://github.com/KinDR007/VictronSmartShunt-ESPHOME)
please advice.
best regards,
Miguel

syssi, 2023-08-07:
Could you provide your complete YAML? It will help to reproduce the issue.

Please make the mentioned snippet is part of the `sensor:` section and neatly indented:

```
- platform: template 
  name: " ${devicename} rem. capacity " 
  id: rc 
  unit_of_measurement: Ah 
  device_class: energy 
  state_class: total_increasing 
  accuracy_decimals: ${accuracy} 
  lambda: |- 
    float i = 0.0;  
    float e = id(bv).state; 
    float w = id(kWhc).state; 
    if (w > 0) { 
      i =  w / e; 
    } 
    return {i};
```

magnetus26, 2023-08-08:
the template is working but its not pretty. i think it colud be improved.

```
esphome:
  name: "${lower_devicename}"
  platform: ESP8266
  board: d1_mini
  comment: "Monitor and a Victron Smart Shunt via TTL"

external_components:
  - source: github://KinDR007/VictronSmartShunt-ESPHOME@main
    refresh: 0s

substitutions:
  lower_devicename: "smartshunt"
  devicename: "SmartShunt"
  config_version: "v2021.06.02.001"
  wifi_fast_connect: "true"
  accuracy: "3"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
 
  manual_ip:
   static_ip: 192.168.12.27
   gateway: 192.168.12.1
   subnet: 255.255.255.0
   dns1: 192.168.12.1
   dns2: 1.1.1.1

  ap:
    ssid: "${devicename} Hotspot"
    password: !secret wifi_ap_password

captive_portal:

logger:
  baud_rate: 0

api:

ota:

web_server:
  port: 80

time:
  - platform: homeassistant

uart:
  id: uart_0
  tx_pin: D8
  rx_pin: D7
  baud_rate: 19200
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256

victron_smart_shunt:
  uart_id: uart_0

sensor:
  - platform: victron_smart_shunt
    battery_voltage:
      name: "Battery Voltage"
      id: bv
      accuracy_decimals: ${accuracy}

    battery_current:
      name: "Battery Current"
      id: bc 
      accuracy_decimals: ${accuracy}

    instantaneous_power:
      name: "instantaneous power"
      id: instantaneous_power
      filters:
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}

    time_to_go:
      name: "time to go"
      id: time_to_go

    consumed_amp_hours:
      name: "consumed amp hours"
      id: consumed_amp_hours
      unit_of_measurement: Ah
      accuracy_decimals: ${accuracy}

    min_battery_voltage:
      name: "Min battery voltage"
      id: min_battery_voltage
      accuracy_decimals: ${accuracy}

    max_battery_voltage:
      name: "Max battery voltage"
      id: max_battery_voltage
      accuracy_decimals: ${accuracy}

    amount_of_charged:
      name: "Amount of charged"
      id:  amount_of_charged   
      filters:
        - multiply: 0.000001
      unit_of_measurement: MWh
      device_class: energy
      state_class: total_increasing
      accuracy_decimals: ${accuracy}
      
    bmv_alarm_text:
      name: "BMV alarm"
      id: bmv_alarm
     
    last_full_charge:
      name: "Time since last full charge"
      id: last_full_charge

    deepest_discharge:
      name: "Depth of the deepest discharge"
      id: deepest_discharge
      unit_of_measurement: Ah
      accuracy_decimals: ${accuracy}

    last_discharge:
      name: "Depth of the last discharge"
      id: last_discharge
      unit_of_measurement: Ah
      accuracy_decimals: ${accuracy}

    state_of_charge:
      id: state_of_charge
      name: "SoC"

    discharged_energy:
      name: "Amount of discharged energy"
      id: discharged_energy   
      filters:
        - multiply: 0.000001
      unit_of_measurement: MWh
      device_class: energy
      state_class: total_increasing
      accuracy_decimals: ${accuracy}
      
  - platform: template
    name: " ${devicename} discharge "
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: ${accuracy}
    lambda: |-
          float i = id(consumed_amp_hours).state;
          float e = id(bv).state;
          float w = 0.0;
          if (i < 0) {
            w = i * e;
          }
          return {w * -1} ;
        
     
    
  - platform: template
    name: " ${devicename} charge "
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: ${accuracy}
    lambda: |-
          float i = 264 + id(consumed_amp_hours).state; 
          float e = id(bv).state;
          float w = 0.0;
          if (i > 0) {
            w =  i * e;
          }
          return {w} ;

  - platform: template
    name: " ${devicename} Rem. Capacity "
    unit_of_measurement: Ah
    device_class: energy
    state_class: total_increasing
    lambda: |-
          float i = 264 + id(consumed_amp_hours).state; 
          float e = id(bv).state;
          float w = 0.0;
          if (i > 0) {
            w =  i * e;
          }
          return {i} ;

switch:
  - platform: restart
    icon: mdi:reload-alert
    name: "${devicename} Restart"
```

[image]
[image]

thanks
Miguel

syssi, 2023-08-09:
I would multiply the `state of charge` against the `total capacity`: 264 * 0.78 = 205,92 Ah. I guess this will be more accurate.

closed 2023-08-09 by syssi

magnetus26, 2023-08-09:
i dont have a `total capacity` sensor. i just grab that number from my battery pack data.
how can i create that template to multiply by `state of charge` ?
 do i need to change my code form the lambda down?
thanks
Miguel

syssi, 2023-08-10:
You could try something like this:

```
  - platform: template
    name: " ${devicename} Rem. Capacity"
    unit_of_measurement: Ah
    device_class: energy
    state_class: total_increasing
    lambda: |-
      if (id(state_of_charge).state) {
          return 264 * (id(state_of_charge).state * 0.01f); 
      }
      return NAN;
```