syssi/esphome-jk-bms #362: CRC check failed! with JK_B2A8S20P and SW V11.262

Blackforrestpilot, 2023-08-12:
Hi,
i'm allready using an JK_B2A8S20P with an ESP8266 over UART. It runs since several weeks without any issues on SW V11.17.

Now i'm building a new unit with an other BMS with SW V11.262 and getting several failure messages.

[image]

the new ESP8266 runs on my first BMS without issues.
I've allready checked the wiring, but without success.
[jk-bms-kopie.yaml.txt](https://github.com/syssi/esphome-jk-bms/files/12327939/jk-bms-kopie.yaml.txt)

any ideas, then changing to an ESP32?

Best regards
Christoph

Blackforrestpilot, 2023-08-13:
additional info:
my "old" unit based on version 1.2.0 is running with the new BMS with SW V11.262 without any problems
This old unit was build with ESPHome 2022.4.0

regards
Christoph

syssi, 2023-08-13:
Are you able to downgrade your ESPHome installation? Could you try component version 1.5.0 plus ESPHome 2022.4.0?

Blackforrestpilot, 2023-08-13:
i tested on an different PC, which i've used to setup the first unit.
Python version is 3.11.0
esphome version is 2022.10.2

unfortunately i get the same result with "CRC check failed!" and "Invalid header"
[image]

I've ordered some new ESP and will test my luck with new hardware.
i will keep you informed

Blackforrestpilot, 2023-08-14:
I am now completely confused
This afternoon I flashed the 2nd ESP8266 with the old program and connected it to my first battery with BMS (SW V11.17). There it works without any errors or abnormalities.
Then I flashed the previous ESP with the new program and connected it to the new BMS (SW V11.262).
And there it runs without any problems. No CRC or header errors.
Only sometimes there is an error message in the log that took a little too long.
[image]

Very confusing.
now that I don't know how to narrow down the cause further, I suggest closing here.

regards
Christoph

syssi, 2023-11-01:
A CRC error every now and then is fine. Let's close this issue.

closed 2023-11-01 by syssi

ldiegos, 2024-02-22:
Hi, 
May I open it again?
I'm having the same problem with the JK_B2A8S20P with Hardware version V11.XW and software V11.287
I'm using:
  project:
    name: "syssi.esphome-jk-bms"
    # version: 2.0.0
    version: 1.5.0

Always i have this messages and the information is not updated:
[image]
[image]

Only restarting the ESP8266 and a few minutes after I receive the cell state once:
[image]

Could you help me?
Using the BlueTooth connection the ESP8266 show everything but the connection is not stable and the JK is bipping all the time, so i want to have it connected with the UART port (GPS connector)

Kind regards
Luis.