syssi/esphome-jk-bms #363: Mischmasch data over MT

proasnet, 2023-08-17:
Dear @syssi ,
my friend has the same BMS and esp as I have.
I have not any issue, but my friend has from BMS mischmash data about few days.
Before, all working fine, but few days ago, I was called for interesting problem.
Some data from BMS has mischmasch values, other data is good.
I tried build new firmware from the same config and reflash node. Nothing help.
Please, know you how help to my friend? Thanak you very much.
```
substitutions:
  name: jk-bms
  bms0: "${name} bms0"
  #bms1: "${name} bms1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  #bms0_mac_address: C8:47:8C:EE:F4:F5
  bms0_mac_address: C8:47:8C:E8:E4:A4
  #bms1_mac_address: C8:47:8C:E4:52:9D
  bms0_protocol_version: JK02
  #bms1_protocol_version: JK02

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.2.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

logger:
  level: debug

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
  #- mac_address: ${bms1_mac_address}
  #  id: client1

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${bms0_protocol_version}
    throttle: 10s
    id: bms0
  #- ble_client_id: client1
  #  protocol_version: ${bms1_protocol_version}
  #  throttle: 10s
  #  id: bms1

binary_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balancing:
      name: "${bms0} balancing"
    charging:
      name: "${bms0} charging"
    discharging:
      name: "${bms0} discharging"

  #- platform: jk_bms_ble
  #  jk_bms_ble_id: bms1
  #  balancing:
  #    name: "${bms1} balancing"
  #  charging:
  #    name: "${bms1} charging"
  #  discharging:
  #    name: "${bms1} discharging"

button:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    retrieve_settings:
      name: "${bms0} retrieve settings"
    retrieve_device_info:
      name: "${bms0} retrieve device info"

  #- platform: jk_bms_ble
  #  jk_bms_ble_id: bms1
  #  retrieve_settings:
  #    name: "${bms1} retrieve settings"
  #  retrieve_device_info:
  #    name: "${bms1} retrieve device info"
      
  - platform: restart
    name: "ESP Restart"

number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balance_trigger_voltage:
      name: "${bms0} balance trigger voltage"
    balance_starting_voltage:
      name: "${bms0} balance starting voltage"
    power_off_voltage:
      name: "${bms0} power off voltage"
    # ...

  #- platform: jk_bms_ble
  #  jk_bms_ble_id: bms1
  #  balance_trigger_voltage:
  #    name: "${bms1} balance trigger voltage"
  #  balance_starting_voltage:
  #    name: "${bms1} balance starting voltage"
  #  power_off_voltage:
  #    name: "${bms1} power off voltage"
    # ...

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    delta_cell_voltage:
      name: "${bms0} delta cell voltage"
    average_cell_voltage:
      name: "${bms0} average cell voltage"
    total_voltage:
      name: "${bms0} total voltage"
    current:
      name: "${bms0} current"
    power:
      name: "${bms0} power"
    temperature_sensor_1:
      name: "${bms0} temperature sensor 1"
    temperature_sensor_2:
      name: "${bms0} temperature sensor 2"
    power_tube_temperature:
      name: "${bms0} power tube temperature"
    state_of_charge:
      name: "${bms0} state of charge"
    capacity_remaining:
      name: "${bms0} capacity remaining"
    charging_cycles:
      name: "${bms0} charging cycles"
    cell_voltage_1:
      name: "${bms0} cell voltage 1"
    cell_voltage_2:
      name: "${bms0} cell voltage 2"
    cell_voltage_3:
      name: "${bms0} cell voltage 3"
    cell_voltage_4:
      name: "${bms0} cell voltage 4"
    cell_voltage_5:
      name: "${bms0} cell voltage 5"
    cell_voltage_6:
      name: "${bms0} cell voltage 6"
    cell_voltage_7:
      name: "${bms0} cell voltage 7"
    cell_voltage_8:
      name: "${bms0} cell voltage 8"
    cell_voltage_9:
      name: "${bms0} cell voltage 9"
    cell_voltage_10:
      name: "${bms0} cell voltage 10"
    cell_voltage_11:
      name: "${bms0} cell voltage 11"
    cell_voltage_12:
      name: "${bms0} cell voltage 12"
    cell_voltage_13:
      name: "${bms0} cell voltage 13"
    cell_voltage_14:
      name: "${bms0} cell voltage 14"
    cell_voltage_15:
      name: "${bms0} cell voltage 15"
    cell_voltage_16:
      name: "${bms0} cell voltage 16"
    # ...

  #- platform: jk_bms_ble
  #  jk_bms_ble_id: bms1
  #  delta_cell_voltage:
  #    name: "${bms1} delta cell voltage"
  #  average_cell_voltage:
  #    name: "${bms1} average cell voltage"
  #  total_voltage:
  #    name: "${bms1} total voltage"
  #  current:
  #    name: "${bms1} current"
  #  power:
  #    name: "${bms1} power"
  #  temperature_sensor_1:
  #    name: "${bms1} temperature sensor 1"
  #  temperature_sensor_2:
  #    name: "${bms1} temperature sensor 2"
  #  power_tube_temperature:
  #    name: "${bms1} power tube temperature"
  #  state_of_charge:
  #    name: "${bms1} state of charge"
  #  capacity_remaining:
  #    name: "${bms1} capacity remaining"
  #  charging_cycles:
  #    name: "${bms1} charging cycles"
  #  cell_voltage_1:
  #    name: "${bms1} cell voltage 1"
  #  cell_voltage_2:
  #    name: "${bms1} cell voltage 2"
  #  cell_voltage_3:
  #    name: "${bms1} cell voltage 3"
  #  cell_voltage_4:
  #    name: "${bms1} cell voltage 4"
  #  cell_voltage_5:
  #    name: "${bms1} cell voltage 5"
  #  cell_voltage_6:
  #    name: "${bms1} cell voltage 6"
  #  cell_voltage_7:
  #    name: "${bms1} cell voltage 7"
  #  cell_voltage_8:
  #    name: "${bms1} cell voltage 8"
  #  cell_voltage_9:
  #    name: "${bms1} cell voltage 9"
  #  cell_voltage_10:
  #    name: "${bms1} cell voltage 10"
  #  cell_voltage_11:
  #    name: "${bms1} cell voltage 11"
  #  cell_voltage_12:
  #    name: "${bms1} cell voltage 12"
  #  cell_voltage_13:
  #    name: "${bms1} cell voltage 13"
  #  cell_voltage_14:
  #    name: "${bms1} cell voltage 14"
  #  cell_voltage_15:
  #    name: "${bms1} cell voltage 15"
  #  cell_voltage_16:
  #    name: "${bms1} cell voltage 16"
    # ...
    
switch:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    charging:
      name: "${bms0} charging"
    discharging:
      name: "${bms0} discharging"
    balancer:
      name: "${bms0} balancer"

  #- platform: jk_bms_ble
  #  jk_bms_ble_id: bms1
  #  charging:
  #    name: "${bms1} charging"
  #  discharging:
  #    name: "${bms1} discharging"
  #  balancer:
  #    name: "${bms1} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${bms0} enable bluetooth connection"
  #- platform: ble_client
  #  ble_client_id: client1
  #  name: "${bms1} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"

  #- platform: jk_bms_ble
  #  jk_bms_ble_id: bms1
  #  errors:
  #    name: "${bms1} errors"
  #  total_runtime_formatted:
  #    name: "${bms1} total runtime formatted"
```

[image]

syssi, 2023-08-17:
Please use protocol version `JK02_32S` instead of `JK02`.

proasnet, 2023-08-17:
Thank you. Now working fine.

closed 2023-08-17 by proasnet