syssi/esphome-jk-bms #369: I have problems to install the lovelace support in the HA.

moscito1010, 2023-09-02:
Hello
I have problems to install the lovelace support in the HA.
HA runs for me as a Docker container without a supervisor.
HACS could be installed but I can't get any further with lovelace.
Does anyone have a simple guide on how to install lovelace in a HA Docker container? What I have found so far was out of date or the instructions found were apparently not usable for my system.
thank you very much

syssi, 2023-11-01:
The user interface of Home Assistant is called Lovelace and doesn't need to be installed. It's just there. What are you looking for?

szheliab, 2023-11-02:
> Hello I have problems to install the lovelace support in the HA. HA runs for me as a Docker container without a supervisor. HACS could be installed but I can't get any further with lovelace. Does [...]

Hi
Probably you're looking for `frontend` configuration: https://www.home-assistant.io/integrations/frontend/

closed 2023-11-04 by syssi