syssi/esphome-jk-bms #371: Errors in the log, problem or expected behavior?

NeliusDK, 2023-09-10:
Just trying to use this to connect my 3 JK BMS to Home Assistant.
It seems to be working, but i get some errors in the log and i cant figure out where they're comming from.

Full log and full config / code is attached.

[17:04:56][I][jk_bms_ble:1039]:   Con. wire resistance 24: 0.000000 Ohm
[17:04:56][W][component:204]: Component esp32_ble took a long time for an operation (0.33 s).
[17:04:56][W][component:205]: Components should block for at most 20-30ms.
[17:04:56][I][jk_bms_ble:467]: Cell info frame (version 3, 300 bytes) received
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 average cell voltage': Sending state 3.11400 V with 3 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 delta cell voltage': Sending state 0.06800 V with 3 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 mosfet temperature': Sending state 25.20000 °C with 1 decimals of accuracy
[17:04:56][D][jk_bms_ble:562]: Wire resistance warning bitmask: 0x00 0x00 0x00 0x00
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 total voltage': Sending state 49.82600 V with 2 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 current': Sending state -1.73100 A with 2 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 power': Sending state -86.24882 W with 2 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 temperature sensor 1': Sending state 22.50000 °C with 1 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 temperature sensor 2': Sending state 22.80000 °C with 1 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[17:04:56][D][text_sensor:064]: 'JK-BMS1 jk-reader-1 errors': Sending state ''
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 balancing current': Sending state 0.00000 A with 2 decimals of accuracy
[17:04:56][D][binary_sensor:036]: 'JK-BMS1 jk-reader-1 balancing': Sending state OFF
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 state of charge': Sending state 4.00000 % with 0 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 capacity remaining': Sending state 13.62100 Ah with 0 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 total battery capacity setting': Sending state 280.00000 Ah with 0 decimals of accuracy
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 charging cycles': Sending state 109.00000  with 0 decimals of accuracy
[17:04:56][D][jk_bms_ble:647]: Unknown158: 0x64 0x00 (always 0x64 0x00?)
[17:04:56][D][jk_bms_ble:650]: Unknown160: 0x00 0x00 (always 0xC5 0x09?)
[17:04:56][D][sensor:094]: 'JK-BMS1 jk-reader-1 total runtime': Sending state 15726000.00000 s with 0 decimals of accuracy
[17:04:56][D][text_sensor:064]: 'JK-BMS1 jk-reader-1 total runtime formatted': Sending state '182d '
[17:04:56][D][binary_sensor:036]: 'JK-BMS1 jk-reader-1 charging': Sending state ON
[17:04:56][D][binary_sensor:036]: 'JK-BMS1 jk-reader-1 discharging': Sending state ON
[17:04:56][D][jk_bms_ble:663]: Unknown168: 00.00.00.00.00.00.00.00.00.00.00.00.00.00 (14)
[17:04:56][D][jk_bms_ble:677]: Unknown189: 0x01 0x64
[17:04:56][D][jk_bms_ble:681]: Unknown192: 0x08
[17:04:56][D][jk_bms_ble:683]: Unknown193: 0x00 0x8B (0x00 0x8D)
[17:04:56][D][jk_bms_ble:685]: Unknown195: 0x67 0x43 (0x21 0x40)
[17:04:56][W][component:204]: Component esp32_ble took a long time for an operation (0.23 s).
[17:04:56][W][component:205]: Components should block for at most 20-30ms.

[logs_jk-reader-1_run.txt](https://github.com/syssi/esphome-jk-bms/files/12568531/logs_jk-reader-1_run.txt)
[JK esphome ble backup 03-09-2023.txt](https://github.com/syssi/esphome-jk-bms/files/12568540/JK.esphome.ble.backup.03-09-2023.txt)

syssi, 2023-09-10:
This is debug output (`[D]`) and can be ignored.

NeliusDK, 2023-09-10:
Ahh, perfect. Thanks!

closed 2023-09-10 by NeliusDK