syssi/esphome-jk-bms #373: Wrong data

ayasystems, 2023-09-11:
I have wrong data without change any thing of installation, for exaple resistance of cell numer 1

LOG:

[0;38m[VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.01.00.00.00.00.14 (22)
[0m
[0;32m[I][jk_bms_ble:467]: Cell info frame (version 2, 300 bytes) received
[0m
[0;38m[VV][jk_bms_ble:468]:   55.AA.EB.90.02.EF.4B.0D.53.0D.4D.0D.4D.0D.4E.0D.4D.0D.4F.0D.4B.0D.4B.0D.4B.0D.4E.0D.4E.0D.4B.0D.53.0D.54.0D.4B.0D.53.0D.4B.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.03.00.4E.0D.09.00.10.00.2D.00.2E.00.2D.00.2E.00.2E.00.2D.00.2D.00.2D.00.2D.00.2D.00.2D.00.2D.00.2D.00.2D.00.2D.00.2D.00.2D.00.2C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.38.01.00.00.00.00 (150)
[0m
[0;38m[VV][jk_bms_ble:469]:   79.EF.00.00.00.00.00.00.00.00.00.00.11.01.14.01.00.00.08.00.BD.F7.02.63.A7.45.04.00.C0.45.04.00.3C.00.00.00.A7.8B.03.01.64.00.00.00.98.75.F2.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.46.04.00.00.00.00.8F.51.41.40.90.00.00.00.F2.17.D2.1B.00.01.00.01.00.05.00.00.F6.A5.4C.09.00.00.00.00.38.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.01.00.00.00.00.14 (150)
[0m
[0;36m[D][sensor:094]: 'bms cell voltage 1': Sending state 3.40300 V with 3 decimals of accuracy
[0m
[0;37m[V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[0m
[0;37m[V][json:056]: Size after shrink 108 bytes
[0m
[0;36m[D][sensor:094]: 'bms cell resistance 1': Sending state 0.00000 Ω with 3 decimals of accuracy
[0m
[0;37m[V][json:036]: Attempting to allocate 512 bytes for JSON serialization

I have running JK BMS with BLE connection from 5 mounths ago but the data is wrong after the last day 08.09.2023.... 
[image]

[image]

With the android app all data is showed fine, no error founded.

[jk.log](https://github.com/syssi/esphome-jk-bms/files/12574867/jk.log)

ayasystems, 2023-09-11:
I'm ussing 1.5.0 version 

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.5.0

syssi, 2023-09-11:
Could you try to use protocol version `JK02_32S`:

```
jk_bms_ble:
  - ble_client_id: client0
    protocol_version: JK02_32S
    throttle: 5s
    id: bms0
```

The protocol version auto-detection is buggy and can cause this issue.

ayasystems, 2023-09-11:
All is working fine with JK02_32S... Thanks!!

syssi, 2023-09-11:
Perfect! I will remove the unstable auto-detection soon.

closed 2023-09-11 by syssi

ayasystems, 2023-09-12:
;) DO2KSM EA4GKQ 73s