syssi/esphome-jk-bms #375: BMS suddenly stopped reporting most sensor values

Jalle19, 2023-09-18:
All of a sudden, this morning my BMS stopped reporting most values to esphome-jk-bms. I updated to the latest version, but it didn't help. Most values are 0, but some values are read correctly (individual cell voltages), and some values are read incorrectly (e.g. charging cycles, should be 60).

```
[11:04:03][D][sensor:094]: 'jk-bms cell voltage 1': Sending state 3.26700 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms cell resistance 1': Sending state 0.00000 Ω with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms cell voltage 2': Sending state 3.26800 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms cell resistance 2': Sending state 0.00000 Ω with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms cell voltage 3': Sending state 3.26300 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms cell resistance 3': Sending state 0.00000 Ω with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms cell voltage 4': Sending state 3.26100 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms cell resistance 4': Sending state 0.01500 Ω with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms min cell voltage': Sending state 3.26100 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms max cell voltage': Sending state 3.26800 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms average cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms delta cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms max voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms min voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[11:04:03][D][jk_bms_ble:557]: Unknown112: 0x00 0x00
[11:04:03][D][jk_bms_ble:562]: Wire resistance warning bitmask: 0x00 0x00 0x00 0x00
[11:04:03][D][sensor:094]: 'jk-bms total voltage': Sending state 0.00000 V with 2 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms current': Sending state 0.00000 A with 2 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms power': Sending state 0.00000 W with 2 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms charging power': Sending state 0.00000 W with 2 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms temperature sensor 1': Sending state 0.00000 °C with 1 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms temperature sensor 2': Sending state 0.00000 °C with 1 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms power tube temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[11:04:03][D][text_sensor:064]: 'jk-bms errors': Sending state ''
[11:04:03][D][sensor:094]: 'jk-bms balancing current': Sending state 0.00000 A with 2 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms capacity remaining': Sending state 25034.75391 Ah with 0 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms total battery capacity setting': Sending state 0.00000 Ah with 0 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms charging cycles': Sending state 13053.00000  with 0 decimals of accuracy
[11:04:03][D][sensor:094]: 'jk-bms total charging cycle capacity': Sending state 282.84802 Ah with 3 decimals of accuracy
[11:04:03][D][jk_bms_ble:647]: Unknown158: 0x5B 0xAB (always 0x64 0x00?)
[11:04:03][D][jk_bms_ble:650]: Unknown160: 0xFF 0xFF (always 0xC5 0x09?)
[11:04:03][D][sensor:094]: 'jk-bms total runtime': Sending state 21037382.00000 s with 0 decimals of accuracy
[11:04:03][D][text_sensor:064]: 'jk-bms total runtime formatted': Sending state '243d 11h'
[11:04:03][D][jk_bms_ble:663]: Unknown168: 00.00.00.00.00.27.6F.B2.01.00.C0.45.04.00.3C.00.00.00.AC.13.01.01.64.00.00.00.B3.F2.44.01.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (46)
[11:04:03][D][jk_bms_ble:677]: Unknown189: 0x01 0x64
[11:04:03][D][jk_bms_ble:681]: Unknown192: 0x00
[11:04:03][D][jk_bms_ble:683]: Unknown193: 0x00 0xB3 (0x00 0x8D)
[11:04:04][D][jk_bms_ble:685]: Unknown195: 0xF2 0x44 (0x21 0x40)
[11:04:04][W][component:204]: Component esp32_ble took a long time for an operation (0.34 s).
[11:04:04][W][component:205]: Components should block for at most 20-30ms.
```

[image]
[image]

I've power cycled the ESP32 and checked in the Jikong BMS app what's going on - there all the values appear correctly.

syssi, 2023-09-18:
Please set the `protocol_version` to `JK02_32S` and try again.

Jalle19, 2023-09-28:
Sorry for taking so long to reply, but yeah, that did the trick!
```
[15:33:33][I][jk_bms_ble:467]: Cell info frame (version 3, 300 bytes) received
[15:33:33][D][sensor:094]: 'jk-bms cell voltage 1': Sending state 3.33400 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms cell resistance 1': Sending state 0.03800 Ω with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms cell voltage 2': Sending state 3.32500 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms cell resistance 2': Sending state 0.03900 Ω with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms cell voltage 3': Sending state 3.32200 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms cell resistance 3': Sending state 0.03900 Ω with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms cell voltage 4': Sending state 3.32300 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms cell resistance 4': Sending state 0.03700 Ω with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms min cell voltage': Sending state 3.32200 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms max cell voltage': Sending state 3.33400 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms average cell voltage': Sending state 3.32600 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms delta cell voltage': Sending state 0.01300 V with 3 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms max voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms min voltage cell': Sending state 2.00000  with 0 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms power tube temperature': Sending state 38.80000 °C with 1 decimals of accuracy
[15:33:33][D][jk_bms_ble:562]: Wire resistance warning bitmask: 0x00 0x00 0x00 0x00
[15:33:33][D][sensor:094]: 'jk-bms total voltage': Sending state 13.30600 V with 2 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms current': Sending state 24.25300 A with 2 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms power': Sending state 322.71045 W with 2 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms charging power': Sending state 322.71045 W with 2 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms temperature sensor 1': Sending state 32.70000 °C with 1 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms temperature sensor 2': Sending state 31.80000 °C with 1 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[15:33:33][D][text_sensor:064]: 'jk-bms errors': Sending state ''
[15:33:33][D][sensor:094]: 'jk-bms balancing current': Sending state 0.00000 A with 2 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms state of charge': Sending state 9.00000 % with 0 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms capacity remaining': Sending state 25.26700 Ah with 0 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms total battery capacity setting': Sending state 280.00000 Ah with 0 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms charging cycles': Sending state 66.00000  with 0 decimals of accuracy
[15:33:33][D][sensor:094]: 'jk-bms total charging cycle capacity': Sending state 18609.94531 Ah with 3 decimals of accuracy
[15:33:33][D][jk_bms_ble:647]: Unknown158: 0x64 0x00 (always 0x64 0x00?)
[15:33:33][D][jk_bms_ble:650]: Unknown160: 0x00 0x00 (always 0xC5 0x09?)
[15:33:33][D][sensor:094]: 'jk-bms total runtime': Sending state 22175926.00000 s with 0 decimals of accuracy
[15:33:33][D][text_sensor:064]: 'jk-bms total runtime formatted': Sending state '256d 15h'
[15:33:33][D][jk_bms_ble:663]: Unknown168: 00.00.00.00.00.00.00.00.00.00.00.00.00.00 (14)
[15:33:33][D][jk_bms_ble:677]: Unknown189: 0x01 0x64
[15:33:33][D][jk_bms_ble:681]: Unknown192: 0x00
[15:33:33][D][jk_bms_ble:683]: Unknown193: 0x00 0x3F (0x00 0x8D)
[15:33:33][D][jk_bms_ble:685]: Unknown195: 0x34 0x40 (0x21 0x40)
[15:33:33][W][component:204]: Component esp32_ble took a long time for an operation (0.33 s).
[15:33:33][W][component:205]: Components should block for at most 20-30ms.
```

closed 2023-09-28 by Jalle19