syssi/esphome-jk-bms #377: No answer from JKBMS

Tikaminou, 2023-09-29:
Hello,
I do my best, but I don’t have any answer from the BMS ☹
Do you have any idea?
Thank you for your help.

[image]

[image]

[image]
esp32-wroom-32u

[image]

substitutions:
   name: esphome-jkbms
   device_description: "Monitor a JK-BMS via UART-TTL"
   external_components_source: github://syssi/esphome-jk-bms@main
   tx_pin: GPIO1
   rx_pin: GPIO3

esphome:
  name: ${name}
  platform: ESP32
  board: esp32dev
  friendly_name: ESPHome JKBMS
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.5.0
  
external_components:
  - source: ${external_components_source}
    refresh: 0s

api:
  encryption:
    key: "azerty"

ota:
  password: "azerty"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Esphome-Jkbms Fallback Hotspot"
    password: "azerty"
    
logger:
  baud_rate: 0
  level: DEBUG

uart:
  id: uart_0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    
jk_modbus:
  - id: modbus0
    uart_id: uart_0
    rx_timeout: 50ms

jk_bms:
  - id: bms0
    jk_modbus_id: modbus0
    update_interval: 10s

binary_sensor:
  - platform: jk_bms
    balancing:
      name: "${name} balancing"

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    actual_battery_capacity:
      name: "${name} actual battery capacity"

INFO ESPHome 2023.8.3 INFO Reading configuration /config/esphome/esphome-jkbms.yaml... INFO Updating https://github.com/syssi/esphome-jk-bms.git@main INFO Starting log output from esphome-jkbms.local using esphome API INFO Successfully connected to esphome-jkbms.local 
[10:57:08][I][app:102]: ESPHome version 2023.8.3 compiled on Sep 29 2023, 10:30:17 
[10:57:08][I][app:104]: Project syssi.esphome-jk-bms version 1.5.0 
[10:57:08][C][wifi:543]: WiFi: 
[10:57:08][C][wifi:379]: Local MAC: A8:42:E3:4B:EA:8C 
[10:57:08][C][wifi:380]: SSID: 'Freebox-0B7205'[redacted] 
[10:57:08][C][wifi:381]: IP Address: 192.168.1.15 
[10:57:08][C][wifi:383]: BSSID: 3A:07:16:A0:4E:D4[redacted] 
[10:57:08][C][wifi:384]: Hostname: 'esphome-jkbms' 
[10:57:08][C][wifi:386]: Signal strength: -66 dB ▂▄▆█ 
[10:57:08][C][wifi:390]: Channel: 6 
[10:57:08][C][wifi:391]: Subnet: 255.255.255.0 
[10:57:08][C][wifi:392]: Gateway: 192.168.1.254 
[10:57:08][C][wifi:393]: DNS1: 192.168.1.254 
[10:57:08][C][wifi:394]: DNS2: 0.0.0.0 
[10:57:08][C][logger:301]: Logger: 
[10:57:08][C][logger:302]: Level: DEBUG 
[10:57:08][C][logger:303]: Log Baud Rate: 0 
[10:57:08][C][logger:305]: Hardware UART: UART0 
[10:57:08][C][uart.arduino_esp32:124]: UART Bus 0: 
[10:57:08][C][uart.arduino_esp32:125]: TX Pin: GPIO1 
[10:57:08][C][uart.arduino_esp32:126]: RX Pin: GPIO3 
[10:57:08][C][uart.arduino_esp32:128]: RX Buffer Size: 384 
[10:57:08][C][uart.arduino_esp32:130]: Baud Rate: 115200 baud 
[10:57:08][C][uart.arduino_esp32:131]: Data Bits: 8 
[10:57:08][C][uart.arduino_esp32:132]: Parity: NONE 
[10:57:08][C][uart.arduino_esp32:133]: Stop bits: 1 
[10:57:08][C][jk_modbus:113]: JkModbus: 
[10:57:08][C][jk_modbus:114]: RX timeout: 50 ms 
[10:57:08][C][jk_bms:593]: JkBms: 
[10:57:08][C][jk_bms:594]: Address: 0x4E 
[10:57:08][C][jk_bms:595]: Fake traffic enabled: NO 
[10:57:08][C][jk_bms:596]: Minimum Cell Voltage 'esphome-jkbms min cell voltage' 
[10:57:08][C][jk_bms:596]: Device Class: 'voltage' 
[10:57:08][C][jk_bms:596]: State Class: 'measurement' 
[10:57:08][C][jk_bms:596]: Unit of Measurement: 'V' 
[10:57:08][C][jk_bms:596]: Accuracy Decimals: 3 
[10:57:08][C][jk_bms:597]: Maximum Cell Voltage 'esphome-jkbms max cell voltage' 
[10:57:08][C][jk_bms:597]: Device Class: 'voltage' 
[10:57:08][C][jk_bms:597]: State Class: 'measurement' 
[10:57:08][C][jk_bms:597]: Unit of Measurement: 'V' 
[10:57:08][C][jk_bms:597]: Accuracy Decimals: 3 
[10:57:08][C][jk_bms:683]: Balancing 'esphome-jkbms balancing' 
[10:57:08][C][mdns:112]: mDNS: 
[10:57:08][C][mdns:113]: Hostname: esphome-jkbms 
[10:57:08][C][ota:093]: Over-The-Air Updates: 
[10:57:08][C][ota:094]: Address: esphome-jkbms.local:3232 
[10:57:08][C][ota:097]: Using Password. 
[10:57:08][C][api:138]: API Server: 
[10:57:08][C][api:139]: Address: esphome-jkbms.local:6053 
[10:57:08][C][api:141]: Using noise encryption: YES 
[10:57:10][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28 
[10:57:20][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28 
[10:57:30][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28 
[10:57:40][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28

I unplugged from the BMS and put a wire between TX & RX: 
[10:39:40][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28 
[10:39:40][I][jk_bms:071]: Status frame received 
[10:39:40][D][sensor:094]: 'esphome-jkbms min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy 
[10:39:40][D][sensor:094]: 'esphome-jkbms max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy 
[10:39:40][D][binary_sensor:036]: 'esphome-jkbms balancing': Sending state OFF 
[10:39:40][D][sensor:094]: 'esphome-jkbms actual battery capacity': Sending state 0.00000 Ah with 0 decimals of accuracy 
[10:39:40][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28 

Remark: could you please confirm that I don’t need this device between the ESP32 and the JKBMS?
[image]

donulf, 2023-10-12:
same exact issue here :-(  I have an esp32-wroom-32 board too.
 after confirm successful installation, my log says:

[13:09:37][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:09:42][D][binary_sensor:036]: 'jk-bms online status': Sending state OFF
[13:09:42][D][text_sensor:064]: 'jk-bms errors': Sending state 'Offline'
[13:09:42][D][sensor:094]: 'jk-bms min cell voltage': Sending state nan V with 3 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms max cell voltage': Sending state nan V with 3 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms min voltage cell': Sending state nan  with 0 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms max voltage cell': Sending state nan  with 0 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms delta cell voltage': Sending state nan V with 3 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms average cell voltage': Sending state nan V with 3 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms power tube temperature': Sending state nan °C with 0 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms temperature sensor 1': Sending state nan °C with 0 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms temperature sensor 2': Sending state nan °C with 0 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms total voltage': Sending state nan V with 2 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms current': Sending state nan A with 2 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms power': Sending state nan W with 2 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms charging power': Sending state nan W with 2 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms discharging power': Sending state nan W with 2 decimals of accuracy
[13:09:42][D][sensor:094]: 'jk-bms capacity remaining': Sending state nan % with 0 decimals of accuracy
[13:09:42][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:09:47][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:09:52][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:09:57][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:10:03][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:10:08][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:10:12][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:10:17][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
and so on

nigel914, 2023-10-12:
I am also having the same issue I was previously using a USB to TTL connection to my rPI running solar assistant and that worked.

Tikaminou, 2023-10-19:
Hello,
Issue solved: My JKBMS GPS port not working (but it is maybe my fault :)).
I tried with another JKBMS, it works, without any modification of the EPS32 hardware nor ESPhome software.
Regards.

closed 2023-10-19 by Tikaminou

braincore, 2024-05-03:
@Tikaminou How did you conclude that it's your GPS port? I've measured my TX via an oscilloscope to verify its correctness, but I don't see any RX.

Tikaminou, 2024-05-10:
Hello Braincore, unfortunately I am not an expert. I had no more idea so I bought another JKBMS and It works. I conclude that the issue was related to the JKBMS/GPS port.

syssi, 2024-07-24:
I've recently learned the USB-to-TTL connector attached to GPIO1/GPIO3 causes the issue pretty often. Please use GPIOs which aren't attached to other chips on your dev board.