syssi/esphome-jk-bms #380: 02-32s results in no error text coming through

JohnGalt1717, 2023-10-16:
The 

```
errors:
      name: "${name} errors"
```

Isn't coming through nor is total_runtime_formatted as soon as I moved to 02-32s because the logs showed I was supposed to be using it for version.

It pulled the latest version of the source code if that matters.

syssi, 2023-11-01:
Could you increase the log level to `VERY_VERBOSE` and provide some RAW traffic? I'm interested in these log messages to reproduce the issue locally:

```
ESP_LOGI(TAG, "Cell info frame (version %d, %d bytes) received", frame_version, data.size());
ESP_LOGVV(TAG, "  %s", format_hex_pretty(&data.front(), 150).c_str());
ESP_LOGVV(TAG, "  %s", format_hex_pretty(&data.front() + 150, data.size() - 150).c_str());
```
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L467-L469

JohnGalt1717, 2023-11-02:
[logs_bmses3_logs.txt](https://github.com/syssi/esphome-jk-bms/files/13244073/logs_bmses3_logs.txt)
Here you got. Took a while because it crashes Edge with that much data ;)

I'm pretty sure it's JK02_32S because it literally told me that the default isn't right in the logs and it's a newer unit with the push button, heater, and Canbus option.

syssi, 2023-11-05:
This value isn't correct?

```
[16:10:07][V][sensor:043]: 'BMS 5 total runtime': Received new state 18297532.000000
[16:10:07][D][sensor:094]: 'BMS 5 total runtime': Sending state 18297532.00000 s with 0 decimals of accuracy
...
[16:10:02][V][text_sensor:013]: 'Balancer 5 total runtime formatted': Received new state 211d 18h
[16:10:02][D][text_sensor:064]: 'Balancer 5 total runtime formatted': Sending state '211d 18h'
```

And this value isn't available via MQTT too?

```
[16:10:02][V][sensor:043]: 'BMS 5 errors bitmask': Received new state 4096.000000
[16:10:02][D][sensor:094]: 'BMS 5 errors bitmask': Sending state 4096.00000  with 0 decimals of accuracy
...
[16:10:02][V][text_sensor:013]: 'Balancer 5 errors': Received new state Cell Overvoltage
[16:10:02][D][text_sensor:064]: 'Balancer 5 errors': Sending state 'Cell Overvoltage'
```

I'm asking because the log looks good in general. So may be I didn't got the point / understand the issue?

JohnGalt1717, 2023-11-05:
That’s the heltec balancer that’s also attached. It’s bms 5 that doesn’t report errors properly. It does report error mask.

syssi, 2023-11-06:
@JohnGalt1717 Could you provide your YAML configuration? I would like to try to replicate your setup.

JohnGalt1717, 2023-11-06:
substitutions:
  device_name: bmses3
  battery_name: "Battery 5"
  battery_id: battery5  
  name: "BMS 5"
  bms1_id: bms5  
  mac_address1: xxxxx 
  bal1_id: bal5
  bal1_name: "Balancer 5"
  mac_address_bal:xxxx
  protocol_version1: JK02_32S
  external_components_source: github://syssi/esphome-jk-bms@main

esphome:
  name: bmses3
  friendly_name: BMS Host 3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf    
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10" 

external_components:
  - source: ${external_components_source}
    refresh: 0s

api:

ota:
  password: !secret esphome_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: ${device_name}.hancocks.email

logger:
  level: INFO

web_server:
  port: 80

preferences:
  flash_write_interval: 0.5h

time:
  - platform: homeassistant
    id: homeassistant_time

esp32_ble_tracker:
  scan_parameters:
    interval: 500ms
    window: 150ms    
  on_ble_advertise:
    then:
      - lambda: |-
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());

ble_client:
  - mac_address: ${mac_address1}
    id: ${bms1_id}
  - mac_address: ${mac_address_bal}
    id: ${bal1_id}

[... 343 lines cut ...]
      name: "${bal1_name} balancer"
      

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bal1_name} errors"
    total_runtime_formatted:
      name: "${bal1_name} total runtime formatted"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bal0
    operation_status:
      name: "${bal1_name} operation status"
    total_runtime_formatted:
      name: "${bal1_name} total runtime formatted"
    buzzer_mode:
      name: "${bal1_name} buzzer mode"
    battery_type:
      name: "${bal1_name} battery type"      

interval:
  - interval: 30min
    then:
      - button.press: retrieve_settings_button

syssi, 2023-11-06:
Okay, issue identified:

```
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bal1_name} errors"
    total_runtime_formatted:
      name: "${bal1_name} total runtime formatted"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bal0
    operation_status:
      name: "${bal1_name} operation status"
    total_runtime_formatted:
      name: "${bal1_name} total runtime formatted"
```

Both sensors (Heltec vs JK) using the same name:

```
      name: "${bal1_name} total runtime formatted"
      name: "${bal1_name} total runtime formatted"
```

Please use something like

```
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bal0
    operation_status:
      name: "${bal1_name} operation status"
    total_runtime_formatted:
      name: "${bal1_name} total runtime formatted"
```

instead.

JohnGalt1717, 2023-11-06:
That fixed it. I feel stupid now. Find and Replace struck.

closed 2023-11-06 by JohnGalt1717

syssi, 2023-11-06:
No problem. You are welcome! :-)