syssi/esphome-jk-bms #387: Connection to two NEEY balancers breaks down sporadically

Elektrofix-OL, 2023-11-04:
I have the same problem and have now gone back to ESPHome version 2023.5.4, but after an indefinite period of time (3 hours or 7 days) I no longer receive any data.
My Yamel: 
```
substitutions:`
  name: speicher
  device0: akku1
  device1: akku2
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: 00:e4:4c:21:d9:cd
  mac_address2: 3c:a5:51:85:e8:4c

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG
  
# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
#api:
#  encryption:
#    key: "q+W4YeOHnu8kOBLz0LYw7uUlWzzRE+NMk6CpfZE856k="

mqtt:
   broker: !secret mqtt_host
   username: !secret mqtt_username
   password: !secret mqtt_password
   id: mqtt_client

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0
  - mac_address: ${mac_address2}
    id: client1

heltec_balancer_ble:
  - ble_client_id: client0
[... 298 lines cut ...]
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms1
    # Not implemented
    # errors:
    #   name: "errors ${device1}"
    operation_status:
      name: "operation status ${device1}"
    total_runtime_formatted:
      name: "total runtime formatted ${device1}"
    buzzer_mode:
      name: "buzzer mode ${device1}"
    battery_type:
      name: "battery type ${device1}"

interval:
  - interval: 30min
    then:
      - button.press: retrieve_settings_button0
      - button.press: retrieve_settings_button1
``` 
The displays in the LOG:
[image]
Before time to time comes:
[image]
If I unplug the ESP32 and restart it, everything is OK. I've already changed the board, no success. What am I doing wrong?

syssi, 2023-11-05:
Unfortunately I've no idea what's going on here. Could you try to interface a single balancer? I assume there is a timing issue which gets worth if the BLE traffic increases.

SchwedeGS, 2023-11-15:
Did you rename your neey's?
That's maybe solve this problem. Use different names

stefanK74, 2023-11-27:
Exactly the same issue here. 
There are 2 ESP32s connecting via BT to 2 Neey balancers each. However, I skipped the single-cell datapoints to reduce traffic. 
I´ll try a single ESP32 for each of the balancers and report back.

syssi, 2023-11-27:
Please make sure you are using the latest ESPHome release. The `ble_client` implementation was buggy in the past so multiple connections wasn't possible / all traffic was routed to the first BLE client connection and did confuse the remote device probably.

stefanK74, 2023-12-07:
It seems as the ESP32s run stable when they have to deal with 1 BT connection only.
Using the lastest ESPhome release and multiple BT connections still led to sporadic disconnects.
That's why I now use a separate ESP32 for each of the 4 balancers.

DutchKillerbee, 2024-02-18:
FYI
I experienced the same with WEMOS-S3-MINI with these settings:

esphome:
  name: ${name}
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git
    board: lolin_s3_mini
    framework: espidf
    board_build.flash_mode: dio
    build_unflags: -Werror=all

esp32:
  variant: esp32s3
  board: lolin_s3_mini
  flash_size: 4MB
  framework:
    type: esp-idf
    version: latest
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

closed 2024-08-16 by syssi