syssi/esphome-jk-bms #388: does not create a utility meter assistant in HA, does not reset the utility meter

amgstone, 2023-11-06:
does not create a utility meter assistant in HA, does not reset _the
[image]
[image]
_ utility meter

syssi, 2023-11-06:
It's super important to provide a time source. Without the time the component is unable to reset the counter at midnight:

https://github.com/syssi/esphome-jk-bms/blob/main/yaml-snippets/esp32-ble-energy-dashboard.yaml#L43-L44

amgstone, 2023-11-06:
time:
  - platform: sntp
    id: my_time
    timezone: Europe/Kiev

syssi, 2023-11-06:
Please close this issue if it's resolved.

amgstone, 2023-11-06:
but it doesn't work like that

amgstone, 2023-11-06:
time:
  - platform: homeassistant
    id: my_time

now wrote like this

syssi, 2023-11-07:
This:

```
time:
  - platform: sntp
```

receives the time from NTP servers. This retrieves the time from Home Assistant:

```
time:
  - platform: homeassistant
    id: my_time
```

Does the `bms0 charging energy today` sensor reset on midnight now?

amgstone, 2023-11-08:
time:
  - platform: sntp
reset in 2 am

syssi, 2023-11-08:
`timezone: Europe/Kiev` should fix this.

amgstone, 2023-11-08:
when I type it, it stops working, does not reset the counter

syssi, 2024-10-21:
Did you resolve the issue? I'm unable to replicate it.

amgstone, 2024-10-22:
yes, having solved the task by flashing esp

closed 2024-10-22 by syssi