syssi/esphome-jk-bms #390: support for new range of JK BMS

PMPoulsen, 2023-11-08:
Hi,

Do you have plans for supporting the new range of BMS's from JK e.g.  JK-PB1A16S10P?

PMPoulsen, 2023-11-08:
I will contact them on Aliexpress to see what they can provide. I have ordered but not yet received.
There are new features in this BMS hence I expect that they have updated the protocol. I would guess still modbus over RS-232 and the JKBMS app do also support these new BMS's

PMPoulsen, 2023-11-08:
yes and the protocols for the RS485 / CAN busses should support multiple inverters as well

PMPoulsen, 2023-11-09:
I have received an example on the protocol from JKBMS, how ever I don't know how useful this is as the new parameters for e.g. the communication settings do not seems to be described
[Jikong-BMS-RS485-Communication example.pdf](https://github.com/syssi/esphome-jk-bms/files/13304890/Jikong-BMS-RS485-Communication.example.pdf)

testpaul999, 2023-11-21:
Hi,
If I can help I have one of this new BMS an can send RAW protocol if needed.
Regards

syssi, 2023-11-22:
I'm happy about every detail about the new models. Please provide as much input as you can.

testpaul999, 2023-11-26:
Hi, 
Now I have had time and here some files.
Hopefully the names are self explanatory.
[sending 1.4A charge.txt](https://github.com/syssi/esphome-jk-bms/files/13466828/sending.1.4A.charge.txt)
[resive 1_4A charge.txt](https://github.com/syssi/esphome-jk-bms/files/13466829/resive.1_4A.charge.txt)
[sending_site_no load.txt](https://github.com/syssi/esphome-jk-bms/files/13466830/sending_site_no.load.txt)
[sendig 1_3A load.txt](https://github.com/syssi/esphome-jk-bms/files/13466831/sendig.1_3A.load.txt)
[resive_site_no load.txt](https://github.com/syssi/esphome-jk-bms/files/13466832/resive_site_no.load.txt)
[resive 1_3A load.txt](https://github.com/syssi/esphome-jk-bms/files/13466833/resive.1_3A.load.txt)
[image]
[image]
[image]
Collected via rs485 adapter 
Regards

syssi, 2023-11-27:
Cool! Could you provide some more screenshots (of the different tabs) of the `JK-BMS-MONITOR` app? It's the first time I see this application.

syssi, 2023-11-27:
I've extracted all Modbus requests from the capture above:

```
01 10 10 70 00 02 04 00 01 00 00 68 8B
01 10 10 74 00 02 04 00 01 00 00 69 78
01 10 10 84 00 02 04 0D DE 00 00 55 6A
01 10 16 0C 00 01 02 00 00 D1 5D
01 10 16 1C 00 01 02 00 00 D3 CD
01 10 16 1E 00 01 02 00 00 D2 2F
01 10 16 20 00 01 02 00 00 D6 F1
01 10 16 22 00 01 02 00 00 D7 13
01 10 16 24 00 01 02 00 00 D7 75
 |  |  |  |  |  |  |  |  |  |  |
 |  |  |  |  |  |  |  |  |  |  CRC
 |  |  |  |  |  |  |  |  |  CRC
 |  |  |  |  |  |  |  |  Data
 |  |  |  |  |  |  |  Data
 |  |  |  |  |  |  Data length
 |  |  |  |  |  Number of registers low
 |  |  |  |  Number of registers high
 |  |  |  Start address low
 |  |  Start address high
 |  Function code
 Device address
```

testpaul999, 2023-11-28:
Hi, screenshots from PC software 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

regards

jrventer, 2023-11-30:
I just got two JK_PB2A16S20P BMS so busy getting them ready to test. Managed to get the JK Modbus protocol document however in Chinese but most of the important parts are english as well. 
[JK_BMS RS485 Modbus v1_0.pdf](https://github.com/syssi/esphome-jk-bms/files/13508230/JK_BMS.RS485.Modbus.v1_0.pdf)
Will be testing these soon might save time reverse engineering using the JK BMS monitor. At least with modbus should be straight forward integration now.

testpaul999, 2023-12-02:
I‘m to be anxious to get the esp32 run with this new bms.
Are the BT integration available as well?

regards

jrventer, 2023-12-02:
> I‘m to be anxious to get the esp32 run with this new bms. Are the BT integration available as well?
> 
> regards

I have not tested the BT integration but it works through the same mobile app via BT. Not sure if the BT protocol is a different version but I assume most would work the same as the other BMSs. There is obviously a couple of extra features and values that will have to be added. I have done some initial testing with the RS485 using standard ESPHOME modbus components and it seems to work without any issues.

jrventer, 2023-12-02:
Small sample of collecting SoC% via modbus

`# Configure uart that will be used
uart:
  id: mod_bus
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 115200
  rx_buffer_size: 384

modbus:
  id: jk_modbus

modbus_controller:
  - id: jk_modbus_bat1pack1
    address: 0x01   ## address of the ModBUS slave device on the bus
    modbus_id: jk_modbus
    setup_priority: -10
    update_interval: "5s"
    command_throttle: "50ms"
sensor:
  - platform: modbus_controller # 0x1200 index 0x90 Battery Voltage
    modbus_controller_id: jk_modbus_bat1pack1
    name: "Bat 1 Pack 1 Battery Voltage"
    id: battery_voltage
    register_type: holding
    address: 0x1290
    unit_of_measurement: "V"
    accuracy_decimals: 3
    device_class: voltage
    state_class: measurement
    filters:
      - multiply: 0.001
    value_type: U_DWORD_R`

testpaul999, 2023-12-14:
Hello,

have build esp32-ble for this new jk bms.
It run without problem and display all parameters accept the new one‘s.
Is there any beta decode for maximize to all parameters?

regards.

testpaul999, 2024-01-04:
Happy new year 🎇
Is there any progress within implement the newer jk inverter BMS?

regards

silviudc, 2024-01-17:
I have a  jk-pb2a16s20p bms on v14.17, and would like to connect via an esp32 and ble and home assistant, have people managed to get it to work? Which yaml file to use?

syssi, 2024-01-17:
If you want to use BLE please try this one: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml

If you want to talk to your BMS using RS485 this is a first draft: https://github.com/syssi/esphome-jk-bms/blob/add-jk-pbx-support/esp32-jk-pb-modbus-example.yaml

silviudc, 2024-01-17:
Thank you. And use protocol JK02_32S in the config?

testpaul999, 2024-01-17:
@silviudc 
see here have build this https://github.com/syssi/esphome-jk-bms/issues/390#issuecomment-1855716298
yes this will work with this protocol.
you will see all informations accept the new parameter like RCV and RFV.
hopefully Syssi will improve the BT protocol as well.

regards

silviudc, 2024-01-17:
Yes I got it to work, it is great.
The only thing missing, those two values, a toggle for float mode on/off and temperature sensor 3 and 4, only see 1 and 2

syssi, 2024-01-17:
@silviudc Could you increase the log level to `VERY_VERBOSE` and provide a ESPHome log which includes some RAW data.

silviudc, 2024-01-17:
[logs_esp32-jkbms-bt_logs.txt](https://github.com/syssi/esphome-jk-bms/files/13968754/logs_esp32-jkbms-bt_logs.txt)

syssi, 2024-01-18:
The (for me) important parts of the logs for analysis later on:

```
# logs_esp32-jkbms-bt_logs.txt
# The device info frame isn't part of the log

# Cell info frame (version 3, 300 bytes) received
55.AA.EB.90.02.2B.C8.0C.CA.0C.CD.0C.CD.0C.CF.0C.D0.0C.D0.0C.CC.0C.C7.0C.CB.0C.CB.0C.CC.0C.CD.0C.D1.0C.D1.0C.CF.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.CD.0C.0A.00.0E.0C.5B.00.3C.00.3E.00.3B.00.3F.00.3C.00.3F.00.3C.00.40.00.3D.00.40.00.3D.00.40.00.3E.00.40.00.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7C.00.00.00.00.00. CD.CC.00.00.38.DF.35.00.2C.F3.FE.FF.71.00.68.00.00.00.08.00.00.00.00.51.A4.75.07.00.C0.27.09.00.02.00.00.00.37.6E.17.00.64.00.00.00.7C.0E.08.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9D.03.00.00.BA.00.8E.8A.40.40.00.00.00.00.7A.14.00.00.00.01.01.01.00.06.00.00.62.88.06.00.00.00.00.00.7C.00.71.00.74.00.A1.03.77.D0.9C.07.3D.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.59

# Cell info frame (version 3, 300 bytes) received
55.AA.EB.90.02.37.C8.0C.CA.0C.CD.0C.CD.0C.CF.0C.D1.0C.D0.0C.CC.0C.C7.0C.CB.0C.CB.0C.CC.0C.C6.0C.CF.0C.D1.0C.CF.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.CD.0C.0B.00.0E.0C.5B.00.3C.00.3E.00.3B.00.3F.00.3C.00.3F.00.3C.00.40.00.3D.00.40.00.3D.00.40.00.3E.00.40.00.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7C.00.00.00.00.00. CF.CC.00.00.17.94.35.00.F4.F8.FE.FF.71.00.68.00.00.00.08.00.00.00.00.51.2A.75.07.00.C0.27.09.00.02.00.00.00.B1.6E.17.00.64.00.00.00.82.0E.08.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9D.03.00.00.B6.00.8E.8A.40.40.00.00.00.00.7B.14.00.00.00.01.01.01.00.06.00.00.A4.88.06.00.00.00.00.00.7C.00.71.00.74.00.A1.03.7D.D0.9C.07.3D.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.0C
```

syssi, 2024-01-18:
@silviudc Could you provide another log including the `device info frame` and `settings frame` you can request this data using the button entities:

```
button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"
```

testpaul999, 2024-01-18:
How do I Catch the log file? I don’t use homeassistant only the web-ui

syssi, 2024-01-18:
If you have to command line tool (`pip3 install esphome`) installed you can look into the logs using:

```
esphome logs yournode.yaml
```

testpaul999, 2024-01-18:
Ok here they are. Have push the both buttons twice
[jk bms_2.log](https://github.com/syssi/esphome-jk-bms/files/13975263/jk.bms_2.log)

syssi, 2024-01-18:
Good job!

```
# jk bms_2.log

# Device info frame (300 bytes) received
55.AA.EB.90.03.6B.4A.4B.5F.50.42.32.41.31.36.53.31.35.50.00.00.00.31.34.2E.58.41.00.00.00.31.34.2E.32.30.00.00.00.88.A3.01.00.94.00.00.00.4A.4B.5F.50.42.32.41.31.36.53.31.35.50.00.00.00.31.32.33.34.00.00.00.00.00.00.00.00.00.00.00.00.32.33.31.31.31.38.00.00.33.30.39.32.35.37.32.31.33.34.00.30.30.30.30.00.49.6E.70.75.74.20.55.73.65.72.64.61.74.61.00.00.31.32.33.34.35.37.00.00.00.00.00.00.00.00.00.00.49.6E.70.75.74.20.55.73.65.72.64.61.74.61.00.00.FE.FF.FF.FF.AF.E9.01.02.00.00.00.00.90.1F.00.00.00.00.C0.D8.E7.FE.1F.00.00.01.00.00.00.00.00.00.00.00.01.04.CF.03.00.00.00.00.00.00.00.00.00.00.00.00.00.00.DF.07.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0B.00.01.00.00.00.00.00.00.00.00.09.00.00.00.0B.00.00.00.00.00.00.00.00.00.00.00.80.51.00.00.0A.50.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.9F.E9.FE.03.00.00.00.00.00.00.00.DE

# Vendor ID: JK_PB2A16S15P
# Hardware version: 14.XA
# Software version: 14.20
# Uptime: 107400 s
# Power on count: 148
# Device name: JK_PB2A16S15P
# Device passcode: 1234
# Manufacturing date: 231118
# Serial number: 3092572134
# Passcode: 0000
# User data: Input Userdata
# Setup passcode: 123457

# Settings frame (300 bytes) received
55.AA.EB.90.01.9B.AC.0D.00.00.F0.0A.00.00.B7.0B.00.00.41.0E.00.00.79.0D.00.00.03.00.00.00.7A.0D.00.00.F1.0A.00.00.7E.0D.00.00.48.0D.00.00.C4.09.00.00.20.CB.00.00.03.00.00.00.3C.00.00.00.20.CB.00.00.2C.01.00.00.3C.00.00.00.05.00.00.00.D0.07.00.00.BC.02.00.00.58.02.00.00.BC.02.00.00.58.02.00.00.32.00.00.00.64.00.00.00.E8.03.00.00.20.03.00.00.10.00.00.00.01.00.00.00.01.00.00.00.01.00.00.00.D0.07.00.00.DC.05.00.00.7A.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.05.00.00.00.60.E3.16.00.00.02.3C.32.18.FE.FF.FF.FF.BF.E9.01.02.00.00.00.00.9E
```

testpaul999, 2024-01-18:
You are welcome 🙂

syssi, 2024-01-18:
> in my mind it would be better do both requests at connect or?

This is already the case. The first beep on (re-)connect is the request of the `device info frame`. The second beep (0.5s later) is the request of the `settings frame`. May be you won't see it at the log because it's pretty early after boot up. You could disable the BLE client connection temporary. As soon you re-enable the connection you should hear the two beeps again and see the `settings frame` at the log. This is a feature request of a small improvement in case settings gets changed: https://github.com/syssi/esphome-jk-bms/issues/292

testpaul999, 2024-01-18:
one thing more ;-)
I can change the parameter like "balance trigger voltage" from the esphome Web-UI and see this at the PC who connected vi RS485 all good.
But if I change something at the PC this will never appear at the esphome WEB-UI only by pressing the "retrive settings" the parameter who changesd will shown.

syssi, 2024-01-18:
That's correct and would be improved/fixed by: https://github.com/syssi/esphome-jk-bms/issues/292

At the moment the settings are retrieved once (on connect). You expect are periodically update which isn't the case at the moment.

syssi, 2024-01-18:
If you extend your YAML a bit you could retrieve the settings periodically if you like:

```
button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
      id: retrieve_settings_button
    retrieve_device_info:
      name: "${name} retrieve device info"

interval:
  - interval: 10min
    then:
      - button.press: retrieve_settings_button
```

Your BMS will beep on every settings update / command.

silviudc, 2024-01-18:
@syssi Any luck with finding the temp 3 and temp 4 values, or the float on/off value from the frames above?

testpaul999, 2024-01-18:
@syssi 
thx for the sugesstion maybe every 12h let's see.
:-)

syssi, 2024-01-18:
> Any luck with finding the temp 3 and temp 4 values, or the float on/off value from the frames above?

I will have a look at the weekend. To implement the float switch a btsnoop capture of the commands is required. I will try to identify the missing temperature measurements.

syssi, 2024-01-18:
```
55.AA.EB.90.02.7F.AB.0C.9C.0C.97.0C.9E.0C.A0.0C.9C.0C.9E.0C.9E.0C.96.0C.9E.0C.8D.0C.96.0C.A2.0C.9C.0C.96.0C.A7.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.9D.0C.1E.00.00.0A.68.00.68.00.7A.00.73.00.72.00.85.00.70.00.67.00.82.00.77.00.65.00.66.00.7E.00.78.00.74.00.9C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AE.00.00.00.00.00.C8.C9.00.00.00.00.00.00.00.00.00.00.B1.00.B2.00.00.00.00.00.00.00.00.34.13.04.00.00.D0.07.00.00.00.00.00.00.00.00.00.00.64.00.00.00.EB.A3.01.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.E2.04.00.00.00.00.F6.C7.40.40.00.00.00.00.2D.14.00.00.00.01.01.01.00.06.00.00.9A.0F.00.00.00.00.00.00.AE.00.B3.00.B4.00.90.03.2D.27.9D.07.18.06.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.45
55.AA.EB.90.02.89.AB.0C.9C.0C.97.0C.9E.0C.A0.0C.9C.0C.9E.0C.9F.0C.97.0C.9E.0C.8D.0C.95.0C.A2.0C.9C.0C.96.0C.A7.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.9D.0C.1F.00.00.0A.68.00.68.00.7A.00.73.00.72.00.85.00.70.00.67.00.82.00.77.00.65.00.66.00.7E.00.78.00.74.00.9C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AE.00.00.00.00.00.C8.C9.00.00.00.00.00.00.00.00.00.00.B2.00.B2.00.00.00.00.00.00.00.00.34.13.04.00.00.D0.07.00.00.00.00.00.00.00.00.00.00.64.00.00.00.F0.A3.01.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.E2.04.00.00.00.00.F6.C7.40.40.00.00.00.00.2D.14.00.00.00.01.01.01.00.06.00.00.D0.0F.00.00.00.00.00.00.AE.00.B3.00.B4.00.90.03.32.27.9D.07.18.06.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.92
55.AA.EB.90.02.93.AB.0C.9C.0C.97.0C.9E.0C.A0.0C.9C.0C.9E.0C.9F.0C.96.0C.9E.0C.8D.0C.96.0C.A2.0C.9C.0C.96.0C.A7.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.9D.0C.1E.00.00.0A.68.00.68.00.7A.00.73.00.72.00.85.00.70.00.67.00.82.00.77.00.65.00.66.00.7E.00.78.00.74.00.9C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AE.00.00.00.00.00.C8.C9.00.00.00.00.00.00.00.00.00.00.B2.00.B1.00.00.00.00.00.00.00.00.34.13.04.00.00.D0.07.00.00.00.00.00.00.00.00.00.00.64.00.00.00.F6.A3.01.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.E2.04.00.00.00.00.F6.C7.40.40.00.00.00.00.2D.14.00.00.00.01.01.01.00.06.00.00.07.10.00.00.00.00.00.00.AE.00.B3.00.B4.00.90.03.38.27.9D.07.18.06.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.DE
55.AA.EB.90.02.9B.AB.0C.9C.0C.97.0C.9E.0C.9F.0C.9C.0C.9E.0C.9E.0C.96.0C.9E.0C.8C.0C.96.0C.A2.0C.9C.0C.96.0C.A9.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.9D.0C.1F.00.00.0A.68.00.68.00.7A.00.73.00.72.00.85.00.70.00.67.00.82.00.77.00.65.00.66.00.7E.00.78.00.74.00.9C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AE.00.00.00.00.00.CA.C9.00.00.00.00.00.00.00.00.00.00.B2.00.B2.00.00.00.00.00.00.00.00.34.13.04.00.00.D0.07.00.00.00.00.00.00.00.00.00.00.64.00.00.00.FB.A3.01.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.E2.04.00.00.00.00.F6.C7.40.40.00.00.00.00.2D.14.00.00.00.01.01.01.00.06.00.00.37.10.00.00.00.00.00.00.AE.00.B3.00.B4.00.90.03.3D.27.9D.07.18.06.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.23
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      ^^^^^ ^^^^^                                                                                                                                                                                                                                                                               ^^^^^^^^^^^                                                                                                                        
55.AA.EB.90.02.2B.C8.0C.CA.0C.CD.0C.CD.0C.CF.0C.D0.0C.D0.0C.CC.0C.C7.0C.CB.0C.CB.0C.CC.0C.CD.0C.D1.0C.D1.0C.CF.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.CD.0C.0A.00.0E.0C.5B.00.3C.00.3E.00.3B.00.3F.00.3C.00.3F.00.3C.00.40.00.3D.00.40.00.3D.00.40.00.3E.00.40.00.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7C.00.00.00.00.00.CD.CC.00.00.38.DF.35.00.2C.F3.FE.FF.71.00.68.00.00.00.08.00.00.00.00.51.A4.75.07.00.C0.27.09.00.02.00.00.00.37.6E.17.00.64.00.00.00.7C.0E.08.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9D.03.00.00.BA.00.8E.8A.40.40.00.00.00.00.7A.14.00.00.00.01.01.01.00.06.00.00.62.88.06.00.00.00.00.00.7C.00.71.00.74.00.A1.03.77.D0.9C.07.3D.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.59
[... cut ...]                                                                                                                                                                                                            temp1 temp2                                                                                                                                                                                                                                                                               temp3 temp4
```

testpaul999, 2024-01-19:
Hello,
how I can do this "To implement the float switch a btsnoop capture of the commands is required." Witch LOG-Level I need for?

syssi, 2024-01-19:
Do you own a Android device? This is a short summary what needs to be done: https://github.com/syssi/esphome-jk-bms/issues/276#issuecomment-1466304352

As soon you are able to create a btsnoop capture and retrieve it from your Android device you have to spin up the Android app of the BMS and toggle the `float switch` a few times. The traffic capture will include these commands. If you can provide such an capture I will extract the commands.

testpaul999, 2024-01-19:
hopefully this are what you need
[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/13992869/btsnoop_hci.log)

testpaul999, 2024-01-19:
where i must put in tihs?
Can be tested using:

substitutions:
  external_components_source: github://syssi/esphome-jk-bms@add-new-temperature-sensors

syssi, 2024-01-19:
> hopefully this are what you need
> [btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/13992869/btsnoop_hci.log)

Wow! You are fast. These are the commands which are part of your btsnoop capture:

```
aa:55:90:eb:97:00:97:a2:55:53:be:f1:fc:f9:79:6b:52:14:13:f3  # command 0x97 COMMAND_DEVICE_INFO
aa:55:90:eb:96:00:e9:e2:2d:51:8e:1f:56:08:57:27:a7:05:d4:62  # command 0x96 COMMAND_CELL_INFO
aa:55:90:eb:96:00:d0:52:82:77:75:1b:99:4a:ed:58:3d:6a:52:dc  # command 0x96 COMMAND_CELL_INFO
aa:55:90:eb:30:04:00:00:00:00:36:d5:24:4a:68:8e:ad:95:5f:be  # command 0x30 Update register 0x30 to 0x00 (off)
aa:55:90:eb:30:04:01:00:00:00:3c:35:b5:c4:8c:dd:6c:11:32:b1  # command 0x30 Update register 0x30 to 0x01 (on)
aa:55:90:eb:30:04:00:00:00:00:3d:e2:b4:b4:59:cf:ce:23:3d:8b  # command 0x30 Update register 0x30 to 0x00 (off)
```

It looks like register `0x30` is the `float switch` register address. I will prepare a `switch entity` tomorrow.

syssi, 2024-01-19:
Could you provide a screenshot of the float switch at the Android app? Could you explain the purpose of the float switch? I would like to understand the meaning and would like to know `float switch` is a good name or not. ;-)

testpaul999, 2024-01-19:
[image]
The float switch will enable a lower voltage after absorption 
Here the RFV(V)
[image]
After a config amount of time
[image]
RCV Time (h) is the absorbtion time ( for top balancing the cells)
RFV Time (h) is the float time

syssi, 2024-01-19:
@testpaul999 I don't understand your last statement about the temperature sensor.

syssi, 2024-01-19:
@testpaul999 Do you have the chance to provide another btsnoop capture of the `heating switch`? This would be awesome!

testpaul999, 2024-01-19:
@syssi 
ich meine damit, das die Temeratursensoren vertauscht sind.
wenn ich Hardwareseitig den Sensor1 erwärme geht die Temperatur in der Anzeige bei Temp2 hoch und umgekehrt.
Das selbe gilt für 3 und 4.

give me a moment for the log file "heating switch"

Regards ;-)

syssi, 2024-01-19:
Alright! I will swap the measurements. :-)

testpaul999, 2024-01-19:
22:06 heating switch
22:07 disable temp. sensor
22:08 display always on
22:09 samrt sleep
22:10 disable PCL 
22:11timed store Data
22:12 emergency
22:13 Time Calibration

[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/13994559/btsnoop_hci.log)

:-)

syssi, 2024-01-20:
```
# heating switch
aa:55:90:eb:27:04:01:00:00:00:36:d5:24:4a:68:8e:ad:95:5f:b6
aa:55:90:eb:27:04:00:00:00:00:3c:35:b5:c4:8c:dd:6c:11:32:a7
aa:55:90:eb:27:04:01:00:00:00:3d:e2:b4:b4:59:cf:ce:23:3d:83
aa:55:90:eb:27:04:00:00:00:00:27:df:a7:f9:96:fc:1e:e0:66:41

# disable temp. sensor
aa:55:90:eb:28:04:01:00:00:00:2c:0e:7b:8c:ca:30:42:8f:bc:6f
aa:55:90:eb:28:04:00:00:00:00:9f:7b:ce:d1:b8:b1:87:ec:8a:c5
aa:55:90:eb:28:04:01:00:00:00:d6:bb:2e:15:63:0e:3c:dc:a4:a8
aa:55:90:eb:28:04:00:00:00:00:3a:7a:06:20:a7:93:1b:34:dd:e6
aa:55:90:eb:28:04:01:00:00:00:4c:f5:ec:88:96:68:d6:68:a0:38
aa:55:90:eb:28:04:00:00:00:00:09:6f:8e:93:47:c9:41:db:ac:17

# display always on
aa:55:90:eb:2b:04:01:00:00:00:cf:97:89:f3:51:05:79:71:2c:f8
aa:55:90:eb:2b:04:00:00:00:00:0e:0d:60:6b:59:d5:59:e1:6c:63
aa:55:90:eb:2b:04:01:00:00:00:c1:b9:55:63:42:44:71:55:0b:33
aa:55:90:eb:2b:04:00:00:00:00:ba:97:e6:68:67:fe:71:5b:50:c9
aa:55:90:eb:2b:04:01:00:00:00:76:55:c2:22:63:4f:02:ce:a8:83
aa:55:90:eb:2b:04:00:00:00:00:d8:a7:0a:45:6a:43:1a:4d:85:10
aa:55:90:eb:2b:04:01:00:00:00:5f:3e:db:6a:f8:73:d1:61:5a:83
aa:55:90:eb:2b:04:00:00:00:00:d0:53:35:21:c9:8a:63:ec:6d:31

# smart sleep 
aa:55:90:eb:2d:04:01:00:00:00:b2:ee:bc:da:47:64:e5:0d:4e:cd
aa:55:90:eb:2d:04:00:00:00:00:a8:27:1b:2f:06:59:8a:f0:53:f0
aa:55:90:eb:2d:04:01:00:00:00:7d:c2:34:d7:13:87:8c:b3:d1:a0
aa:55:90:eb:2d:04:00:00:00:00:97:18:3e:84:4a:ac:c0:a5:f4:6b

# disable PCL
aa:55:90:eb:2e:04:01:00:00:00:26:0b:02:f3:b3:a8:10:62:af:4f
aa:55:90:eb:2e:04:00:00:00:00:e8:ec:a0:3c:6a:e2:f0:c1:f5:4e
aa:55:90:eb:2e:04:01:00:00:00:78:ce:a9:c9:66:41:08:6a:0c:8a
aa:55:90:eb:2e:04:00:00:00:00:35:2c:b1:2a:d1:3c:ab:c5:ef:54

# timed stored data
aa:55:90:eb:2f:04:01:00:00:00:54:55:53:83:3f:bf:24:fa:2b:74
aa:55:90:eb:2f:04:00:00:00:00:86:eb:6c:7c:65:3b:a6:ae:21:1b
aa:55:90:eb:2f:04:01:00:00:00:67:b7:8c:73:ec:38:a4:96:0a:33
aa:55:90:eb:2f:04:00:00:00:00:e0:42:cf:51:16:a5:24:9a:e4:4c

# emergency 
aa:55:90:eb:6b:04:01:00:00:00:e3:3e:df:8e:45:cc:fb:41:b1:76
aa:55:90:eb:6b:04:00:00:00:00:b6:e7:60:58:cf:97:64:c2:04:ce
aa:55:90:eb:6b:04:01:00:00:00:9c:68:9a:a6:c8:5c:f5:99:f2:d2
aa:55:90:eb:6b:04:00:00:00:00:9b:bd:0d:00:22:ca:e0:b0:10:da

# time calibration
aa:55:90:eb:6c:04:f0:12:9f:07:2d:2c:52:de:e3:b9:be:3c:89:3a
aa:55:90:eb:6c:04:f3:12:9f:07:d6:a0:4d:da:bc:35:f5:63:7d:f8
aa:55:90:eb:6c:04:f5:12:9f:07:d1:d8:18:c4:75:55:d1:f4:77:22
```

testpaul999, 2024-01-20:
One point more for the v14

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("**_JK__**", 0) == 0) {

at this point you will use jk_ this are the responds from the new jk bms

syssi, 2024-01-20:
Does somebody know the meaning of "Disable PCL module"?

syssi, 2024-01-20:
I've merged the new switch entities into `main`. Please give it a try: https://github.com/syssi/esphome-jk-bms/commit/9187dc54370bcf47a8ce37c5ac9dd3fa549fe6db#diff-0b83aabec0f938f17bc82313483a4241ce5057609a2e384929b5963fc82aeacd

testpaul999, 2024-01-20:
:-) god job all new and old switches do what we expect from them. :-)
The Temperatur Sensors are in rigth order now.
now I'll trace the rest parameters who will be numeric inputs

testpaul999, 2024-01-20:
17:10 RCV V - absorbtion voltage 3.454-3.455-3.454 V change order
17:11 SOC100% - 3.450-3.451-3.450 V change order
17:12 SOC 0% - 2.801-2.802-2.801 V change order
17:13 RFV V - Float voltage 3.400-3.401-3.400 V change order
17:14 Smart Sleep V 3.500-3.501-3.500 V change order
17:15 Time Smart sleep 24-23-24 hours change order
17:16 RCV Time - time for absorbtion 1.0-2.0-1.0 hours change order
17:17 RFV Time - time for Float 8.0-9.0-8.0 hours change order
 
the decimal places are as follows like i write it above

[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/13998576/btsnoop_hci.log)

syssi, 2024-01-20:
```
# RVC V
aa:55:90:eb: 09 :04: 7f:0d:00:00 :36:d5:24:4a:68:8e:ad:95:5f:23 # 3455 = 3.455V
aa:55:90:eb: 09 :04: 7e:0d:00:00 :3c:35:b5:c4:8c:dd:6c:11:32:14 # 3454 = 3.454V

# SOC 100%
aa:55:90:eb: 07 :04: 7b:0d:00:00 :3d:e2:b4:b4:59:cf:ce:23:3d:ea # 3451
aa:55:90:eb: 07 :04: 7a:0d:00:00 :27:df:a7:f9:96:fc:1e:e0:66:a8 # 3450

# SOC 0%
aa:55:90:eb: 08 :04: f2:0a:00:00 :2c:0e:7b:8c:ca:30:42:8f:bc:4a # 2802
aa:55:90:eb: 08 :04: f1:0a:00:00 :9f:7b:ce:d1:b8:b1:87:ec:8a:a0 # 2801

# RFV V
aa:55:90:eb: 0a :04: 49:0d:00:00 :d6:bb:2e:15:63:0e:3c:dc:a4:df # 3401
aa:55:90:eb: 0a :04: 48:0d:00:00 :3a:7a:06:20:a7:93:1b:34:dd:1d # 3400

# Smart Sleep V
aa:55:90:eb: 01 :04: ad:0d:00:00 :4c:f5:ec:88:96:68:d6:68:a0:ca # 3501
aa:55:90:eb: 01 :04: ac:0d:00:00 :09:6f:8e:93:47:c9:41:db:ac:a9 # 3500

# Time Smart Sleep (h)
aa:55:90:eb: 39 :01: 17 :cf:97:89:f3:51:05:79:71:2c:0e:0d:60:94 # 0x17 = 23
aa:55:90:eb: 39 :01: 18 :6b:59:d5:59:e1:6c:c1:b9:55:63:42:44:c3 # 0x18 = 24

# RCV time
aa:55:90:eb: b3 :01: 14 :71:55:0b:ba:97:e6:68:67:fe:71:5b:50:33 # 20
aa:55:90:eb: b3 :01: 0a :76:55:c2:22:63:4f:02:ce:a8:d8:a7:0a:9a # 10

# RFV time
aa:55:90:eb: b4 :01: 5a :45:6a:43:1a:4d:85:5f:3e:db:6a:f8:73:b4 # 90
aa:55:90:eb: b4 :01: 50 :d1:61:5a:d0:53:35:21:c9:8a:63:ec:6d:93 # 80
```

syssi, 2024-01-20:
```
# Cell info frame (version 3, 300 bytes) received
55.AA.EB.90.02.2B.C8.0C.CA.0C.CD.0C.CD.0C.CF.0C.D0.0C.D0.0C.CC.0C.C7.0C.CB.0C.CB.0C.CC.0C.CD.0C.D1.0C.D1.0C.CF.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.CD.0C.0A.00.0E.0C.5B.00.3C.00.3E.00.3B.00.3F.00.3C.00.3F.00.3C.00.40.00.3D.00.40.00.3D.00.40.00.3E.00.40.00.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7C.00.00.00.00.00. CD.CC.00.00.38.DF.35.00.2C.F3.FE.FF.71.00.68.00.00.00.08.00.00.00.00.51.A4.75.07.00.C0.27.09.00.02.00.00.00.37.6E.17.00.64.00.00.00.7C.0E.08.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9D.03.00.00.BA.00.8E.8A.40.40.00.00.00.00.7A.14.00.00.00.01.01.01.00.06.00.00.62.88.06.00.00.00.00.00.7C.00.71.00.74.00.A1.03.77.D0.9C.07.3D.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.59

# Cell info frame (version 3, 300 bytes) received
55.AA.EB.90.02.37.C8.0C.CA.0C.CD.0C.CD.0C.CF.0C.D1.0C.D0.0C.CC.0C.C7.0C.CB.0C.CB.0C.CC.0C.C6.0C.CF.0C.D1.0C.CF.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.CD.0C.0B.00.0E.0C.5B.00.3C.00.3E.00.3B.00.3F.00.3C.00.3F.00.3C.00.40.00.3D.00.40.00.3D.00.40.00.3E.00.40.00.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7C.00.00.00.00.00. CF.CC.00.00.17.94.35.00.F4.F8.FE.FF.71.00.68.00.00.00.08.00.00.00.00.51.2A.75.07.00.C0.27.09.00.02.00.00.00.B1.6E.17.00.64.00.00.00.82.0E.08.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9D.03.00.00.B6.00.8E.8A.40.40.00.00.00.00.7B.14.00.00.00.01.01.01.00.06.00.00.A4.88.06.00.00.00.00.00.7C.00.71.00.74.00.A1.03.7D.D0.9C.07.3D.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.0C

# logs_esp32-jkbms-bt_logs (1).txt
# Cell info frame (version 3, 300 bytes) received
55.AA.EB.90.02.45.CB.0C.CA.0C.CC.0C.C8.0C.C8.0C.CD.0C.CF.0C.CC.0C.C6.0C.CB.0C.CC.0C.CF.0C.CD.0C.D0.0C.D0.0C.CB.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.CB.0C.0A.00.0D.08.5B.00.3C.00.3E.00.3B.00.3F.00.3C.00.3F.00.3C.00.40.00.3D.00.40.00.3D.00.40.00.3E.00.40.00.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7D.00.00.00.00.00. B5.CC.00.00.8A.70.36.00.10.F6.FE.FF.71.00.68.00.00.00.08.00.00.00.00.51.2E.60.07.00.C0.27.09.00.02.00.00.00.AE.83.17.00.64.00.00.00.A1.0F.08.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9D.03.00.00.B8.00.8E.8A.40.40.00.00.00.00.78.14.00.00.00.01.01.01.00.06.00.00.D7.93.06.00.00.00.00.00.7D.00.72.00.74.00.A1.03.9C.D1.9C.07.3D.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.DA

# Cell info frame (version 3, 300 bytes) received
55.AA.EB.90.02.5B.CA.0C.CA.0C.CB.0C.CA.0C.CA.0C.CF.0C.D0.0C.CD.0C.CC.0C.CD.0C.CC.0C.CB.0C.C6.0C.CC.0C.D0.0C.CC.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.CB.0C.0B.00.0E.08.5B.00.3C.00.3E.00.3B.00.3F.00.3C.00.3F.00.3C.00.40.00.3D.00.40.00.3D.00.40.00.3E.00.40.00.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7E.00.00.00.00.00. B5.CC.00.00.8A.70.36.00.10.F6.FE.FF.71.00.68.00.00.00.08.00.00.00.00.51.4C.5F.07.00.C0.27.09.00.02.00.00.00.8F.84.17.00.64.00.00.00.AD.0F.08.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9D.03.00.00.B8.00.8E.8A.40.40.00.00.00.00.78.14.00.00.00.01.01.01.00.06.00.00.4F.94.06.00.00.00.00.00.7E.00.72.00.74.00.A1.03.A8.D1.9C.07.3D.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.84

     0  55.55.55.55.55.55.55.55
     1  AA.AA.AA.AA.AA.AA.AA.AA
     2  EB.EB.EB.EB.EB.EB.EB.EB
     3  90.90.90.90.90.90.90.90
     4  02.02.02.02.02.02.02.02
     5  7F.89.93.9B.2B.37.45.5B
     6  AB.AB.AB.AB.C8.C8.CB.CA
     7  0C.0C.0C.0C.0C.0C.0C.0C
     8  9C.9C.9C.9C.CA.CA.CA.CA
     9  0C.0C.0C.0C.0C.0C.0C.0C
    10  97.97.97.97.CD.CD.CC.CB
    11  0C.0C.0C.0C.0C.0C.0C.0C
    12  9E.9E.9E.9E.CD.CD.C8.CA
    13  0C.0C.0C.0C.0C.0C.0C.0C
    14  A0.A0.A0.9F.CF.CF.C8.CA
    15  0C.0C.0C.0C.0C.0C.0C.0C
    16  9C.9C.9C.9C.D0.D1.CD.CF
    17  0C.0C.0C.0C.0C.0C.0C.0C
    18  9E.9E.9E.9E.D0.D0.CF.D0
    19  0C.0C.0C.0C.0C.0C.0C.0C
    20  9E.9F.9F.9E.CC.CC.CC.CD
    21  0C.0C.0C.0C.0C.0C.0C.0C
    22  96.97.96.96.C7.C7.C6.CC
    23  0C.0C.0C.0C.0C.0C.0C.0C
    24  9E.9E.9E.9E.CB.CB.CB.CD
    25  0C.0C.0C.0C.0C.0C.0C.0C
    26  8D.8D.8D.8C.CB.CB.CC.CC
    27  0C.0C.0C.0C.0C.0C.0C.0C
    28  96.95.96.96.CC.CC.CF.CB
    29  0C.0C.0C.0C.0C.0C.0C.0C
    30  A2.A2.A2.A2.CD.C6.CD.C6
    31  0C.0C.0C.0C.0C.0C.0C.0C
    32  9C.9C.9C.9C.D1.CF.D0.CC
    33  0C.0C.0C.0C.0C.0C.0C.0C
    34  96.96.96.96.D1.D1.D0.D0
    35  0C.0C.0C.0C.0C.0C.0C.0C
    36  A7.A7.A7.A9.CF.CF.CB.CC
    37  0C.0C.0C.0C.0C.0C.0C.0C
    38  00.00.00.00.00.00.00.00
    39  00.00.00.00.00.00.00.00
    40  00.00.00.00.00.00.00.00
    41  00.00.00.00.00.00.00.00
    42  00.00.00.00.00.00.00.00
    43  00.00.00.00.00.00.00.00
    44  00.00.00.00.00.00.00.00
    45  00.00.00.00.00.00.00.00
    46  00.00.00.00.00.00.00.00
    47  00.00.00.00.00.00.00.00
    48  00.00.00.00.00.00.00.00
    49  00.00.00.00.00.00.00.00
    50  00.00.00.00.00.00.00.00
[... cut ...]
   284  00.00.00.00.00.00.00.00
   285  00.00.00.00.00.00.00.00
   286  00.00.00.00.00.00.00.00
   287  FE.FE.FE.FE.FE.FE.FE.FE
   288  FF.FF.FF.FF.FF.FF.FF.FF
   289  7F.7F.7F.7F.7F.7F.7F.7F
   290  DD.DD.DD.DD.DD.DD.DD.DD
   291  2F.2F.2F.2F.2F.2F.2F.2F
   292  01.01.01.01.01.01.01.01
   293  01.01.01.01.01.01.01.01
   294  B0.B0.B0.B0.B0.B0.B0.B0
   295  07.07.07.07.07.07.07.07
   296  00.00.00.00.00.00.00.00
   297  00.00.00.00.00.00.00.00
   298  00.00.00.00.00.00.00.00
   299  45.92.DE.23.59.0C.DA.84
```

testpaul999, 2024-01-20:
Hi,
are this meassages nomal?
<html><body>

20:29:29 | [I] | [jk_bms_ble:469] | Cell info frame (version 3, 300 bytes) received
-- | -- | -- | --
20:29:29 | [W] | [component:214] | Component esp32_ble took a long time for an operation (0.27 s).
20:29:29 | [W] | [component:215] | Components should block for at most 20-30ms.
20:29:34 | [I] | [jk_bms_ble:469] | Cell info frame (version 3, 300 bytes) received
20:29:35 | [W] | [component:214] | Component esp32_ble took a long time for an operation (0.27 s).
20:29:35 | [W] | [component:215] |  

</body>
</html>

testpaul999, 2024-01-20:
Hi,
Have new findings about the problem with the temperature sensors.
This is not „your“ fault the bms make it false.
See this video I have touch the sensor with the number 1 ;-)

https://github.com/syssi/esphome-jk-bms/assets/35597907/31d8bc1f-53d0-4a91-b760-584cbdac62c7

syssi, 2024-01-20:
> 20:29:29 | [W] | [component:214] | Component esp32_ble took a long time for an operation (0.27 s).
> 20:29:29 | [W] | [component:215] | Components should block for at most 20-30ms.

It's normal. Please ignore the warnings.

[123 more replies]

closed 2025-07-16 by syssi