syssi/esphome-jk-bms #391: Device won't start or connect (ESP32S3)

manziman, 2023-11-08:
Hi, I'm new to ESPHome and Home Assistant in general. I tried following the tutorials in the ESPHome docs and built my config using the example files here, but I cannot get the device to start properly or connect. Here's my config (you can see which board I'm using there):

```yaml
substitutions:
  name: jk-bms
  bms0: "${name} Battery 1"
  bms1: "${name} Battery 2"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:80:01:CF:7B
  bms1_mac_address: C8:47:80:01:C6:83
  # Defaults to "JK02" (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
  bms1_protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.5.0

esp32:
  board: adafruit_feather_esp32s3_tft
  framework:
    type: esp-idf
  variant: esp32s3

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: VERY_VERBOSE
    jk_bms_ble: VERY_VERBOSE
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    api.service: INFO
    api: INFO

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
[... 426 lines cut ...]
[17:47:41]I (415) cpu_start: cpu freq: 160000000
[17:47:41]I (415) cpu_start: Application information:
[17:47:41]I (417) cpu_start: Project name:     jk-bms
[17:47:41]I (422) cpu_start: App version:      2023.10.6
[17:47:41]I (427) cpu_start: Compile time:     Nov  8 2023 17:42:06
[17:47:41]I (433) cpu_start: ELF file SHA256:  287202d9812d6a34...
[17:47:41]I (439) cpu_start: ESP-IDF:          4.4.5
[17:47:41]I (444) cpu_start: Min chip rev:     v0.0
[17:47:41]I (449) cpu_start: Max chip rev:     v0.99 
[17:47:41]I (454) cpu_start: Chip rev:         v0.1
[17:47:41]I (458) heap_init: Initializing. RAM available for dynamic allocation:
[17:47:41]I (466) heap_init: At 3FCA3E30 len 000458E0 (278 KiB): D/IRAM
[17:47:41]I (472) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
[17:47:41]I (479) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
[17:47:41]I (485) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
[17:47:41]I (492) spi_flash: detected chip: generic
[17:47:41]I (496) spi_flash: flash io: dio
[17:47:41]I (501) sleep: Configure to isolate all GPIO pins in sleep state
[17:47:41]I (507) sleep: Enable automatic switching of GPIO sleep configuration
[17:47:41]I (514) coexist: coexist rom version e7ae62f
[17:47:41]I (519) cpu_start: Starting scheduler on PRO CPU.
[17:47:41]I (0) cpu_start: Starting scheduler on APP CPU.
```

Anyone have any ideas what's going on?

syssi, 2023-11-09:
Please replace the ESP32S3 with a full featured ESP32. I guess this will fix the issue. Unfortunately I don't know the relevant differences of the ESP models to explain the root cause.

giangy11, 2023-11-16:
Hi,
for me it's working with esp32-s3 (this model: https://www.waveshare.com/wiki/ESP32-S3-Pico)
if you want I can attach here my yaml config

closed 2024-01-13 by syssi

taeasy, 2026-02-27:
> Hi, for me it's working with esp32-s3 (this model: https://www.waveshare.com/wiki/ESP32-S3-Pico) if you want I can attach here my yaml config

Hello, I have the same hardware as you. Canyou send me a YAML config file?