syssi/esphome-jk-bms #395: binary_sensor not working when having 2 ble connected  victron shunt and jk bms

riker65, 2023-11-20:
Hi
maybe you have an idea,

I am reading the victron shunt and the jk bms with the ESP
so mixing 2 sources,

in general I can connect to shunt and jk, but binary actions are not displayed

maybe there is a conflict with the 2 sources 
[  external_components_source_shunt: github://Fabian-Schmidt/esphome-victron_ble 
](url)

radio button for on and off balancing, charging, is not appearing.

`binary_sensor:
  - platform: jk_bms_ble
    balancing:
        #      name: "${bmsname} balancing"
      name: "1 ${bmsname} balancing"      
      id: balancing_BMS
      
    charging:
      name: "1 ${bmsname} charging"
      id: charging
    discharging:
      name: "1 ${bmsname} discharging"
      id: discharging `

Action Radio button not available

[image]

maybe I need to define mot on the binary_sensor?

Thanks Thomas

riker65, 2023-11-24:
user error, all working, forgot the switch

closed 2023-11-24 by riker65

adrusa, 2023-11-25:
> Hi maybe you have an idea,
>
> I am reading the victron shunt and the jk bms with the ESP so mixing 2 sources,
>
> in general I can connect to shunt and jk, but binary actions are not displayed
>
> maybe [...]

Please how did you combine 2 sources? Please share your yaml. Thanks.

riker65, 2023-11-26:
Hi

just added 
and used components from both sources:

```
`  external_components_source_shunt: github://Fabian-Schmidt/esphome-victron_ble 
  mac_address_shunt_format: xxx 
 
  mac_address_shunt: "xxx"
 
  external_components_source_jk: github://syssi/esphome-jk-bms@main
  mac_address_bms: xx  
  protocol_version: JK02_32S
```

`does this help?