syssi/esphome-jk-bms #396: JK-BMS without obtaining data

doreks, 2023-11-21:
Any idea why I am not able to get any data?

`Processing jk-bms (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 2.0.0
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  49.6% (used 40636 bytes from 81920 bytes)
Flash: [=====     ]  46.5% (used 485877 bytes from 1044464 bytes)
========================= [SUCCESS] Took 3.02 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.140
INFO Uploading /data/build/jk-bms/.pioenvs/jk-bms/firmware.bin (490032 bytes)
INFO Compressed to 327520 bytes
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from jk-bms/debug
INFO Connected to MQTT broker!
[16:47:24][I][app:134]: Rebooting safely...
[16:47:45][I][mqtt:274]: MQTT Connected!
[16:47:46][I][app:062]: setup() finished successfully!
[16:47:46][I][app:102]: ESPHome version 2023.11.2 compiled on Nov 21 2023, 16:46:29
[16:47:46][I][app:104]: Project syssi.esphome-jk-bms version 1.5.0
[16:47:46][C][wifi:559]: WiFi:
[16:47:46][C][wifi:391]:   Local MAC: 34:94:54:62:7E:C3
[16:47:46][C][wifi:396]:   SSID: [redacted]
[16:47:46][C][wifi:397]:   IP Address: 192.168.1.140
[16:47:46][C][wifi:398]:   BSSID: [redacted]
[16:47:46][C][wifi:400]:   Hostname: 'jk-bms'
[16:47:46][C][wifi:402]:   Signal strength: -89 dB ▂▄▆█
[16:47:46][C][wifi:406]:   Channel: 3
[16:47:46][C][wifi:407]:   Subnet: 255.255.255.0
[16:47:46][C][wifi:408]:   Gateway: 192.168.1.1
[16:47:46][C][wifi:409]:   DNS1: 192.168.1.1
[16:47:46][C][wifi:410]:   DNS2: 0.0.0.0
[16:47:46][C][logger:416]: Logger:
[16:47:46][C][uart.arduino_esp8266:102]: UART Bus:
[16:47:46][C][uart.arduino_esp8266:103]:   TX Pin: GPIO3
[16:47:46][C][uart.arduino_esp8266:104]:   RX Pin: GPIO1
[16:47:46][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[16:47:46][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[16:47:46][C][uart.arduino_esp8266:109]:   Data Bits: 8
[16:47:46][C][uart.arduino_esp8266:110]:   Parity: NONE
[16:47:46][C][uart.arduino_esp8266:111]:   Stop bits: 1
[16:47:46][C][uart.arduino_esp8266:115]:   Using software serial
[16:47:46][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:47:46][C][jk_modbus:113]: JkModbus:
[16:47:46][C][jk_modbus:114]:   RX timeout: 50 ms
[16:47:46][C][jk_bms:593]: JkBms:
[16:47:46][C][jk_bms:594]:   Address: 0x4E
[16:47:46][C][jk_bms:595]:   Fake traffic enabled: NO
[16:47:46][C][jk_bms:596]: Minimum Cell Voltage 'jk-bms min cell voltage'
[16:47:46][C][jk_bms:596]:   Device Class: 'voltage'
[16:47:46][C][jk_bms:596]:   State Class: 'measurement'
[16:47:46][C][jk_bms:596]:   Unit of Measurement: 'V'
[16:47:46][C][jk_bms:596]:   Accuracy Decimals: 3
[16:47:46][C][jk_bms:597]: Maximum Cell Voltage 'jk-bms max cell voltage'
[16:47:46][C][jk_bms:597]:   Device Class: 'voltage'
[16:47:46][C][jk_bms:597]:   State Class: 'measurement'
[16:47:46][C][jk_bms:597]:   Unit of Measurement: 'V'
[16:47:46][C][jk_bms:597]:   Accuracy Decimals: 3
[... 399 lines cut ...]
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 10': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 11': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 12': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 13': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 14': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 15': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 16': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 17': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 18': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 19': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 20': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 21': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 22': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 23': Sending state nan V with 3 decimals of accuracy
[16:47:56][D][sensor:093]: 'jk-bms cell voltage 24': Sending state nan V with 3 decimals of accuracy
[16:47:56][W][component:214]: Component jk_bms took a long time for an operation (0.19 s).
[16:47:56][W][component:215]: Components should block for at most 20-30ms.
[16:47:56][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:48:01][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:48:06][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:48:11][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:48:16][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:48:21][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:48:26][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[16:48:31][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28`

doreks, 2023-11-23:
HI please big boss @syssi , could you give me some of your words of wisdom?  :-)

syssi, 2023-11-23:
Could you provide a photo of your wiring? Please make sure `GND` is connected properly. If all three wires are connected properly to GPIO1, GPIO3 and GND and aren't swapped (RX vs. TX) there are a few options left:

1. Sometimes GPIO1 and GPIO3 cannot be used because the USB-to-TTL chip of the dev board interferes. Please try GPIO4 + GPIO5 (this is possible because ESPHome uses SoftwareSerial if there is no hardware UART available) instead.
2. Please try a ESP32. I'm aware of many people without success using an ESP8266 dev board.

doreks, 2023-11-23:
thanks @syssi  for your words, time and and and support!

finally I could solve the problem by changing the HW to another ESP :-/
I now have a couple of questions, 
the first one I don't see any switch for balancing, in the photos of the caputaras of this project I see it but I don't observe in the code anything about this. The same thing happens for the value of resistance Ω, I don't see any reference so it doesn't load anything. Is there any way to get this data?

[image]
[image]

again thank so much for support!

syssi, 2023-11-23:
Bluetooth and serial communication are using different protocols. The protocol of the serial communication doesn't provide any details about the resistance and balancing current. You have to use BLE if you need the values.

The balancer switch isn't implemented yet (via serial connection). It's supported by the BLE component only at the moment.

syssi, 2023-11-23:
Could you provide some details about your working setup? Which kind of dev board do you use and do you use a hardware UART or a SoftwareSerial connection.

syssi, 2023-11-25:
@doreks Friendly reminder.

doreks, 2023-11-25:
> @doreks Friendly reminder.

So sorry, just give me this weekend I have plenty of family here.. On Monday will be ready..

virus100b, 2024-02-02:
I can confirm that using GPIO4 + GPIO5  on ESP 8266 NodeMCU that has USB to TTL chip present on board (has micro usb port)  it woks. On D1 Mini the USB to TTL chip is not present on board and probably many people that use code for D1Mini and use Node MCU boards failed to communicate with BMS.

doreks, 2024-02-02:
@virus100b how is your setup?

virus100b, 2024-02-02:
RX, TX from GPS port JK Bms connected to GPIO5, GPIO 4 of ESP 8266 Node MCU board. External 5V supply for Node MCU.
Do not use Vcc on GPS port of JK because tis Vcc is the 16S battery voltage (48-55V)  

substitutions:
  name: jk-bms
  device_description: "Monitor a JK-BMS via UART-TTL"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO4
  rx_pin: GPIO5

baturinivan, 2024-04-25:
I have similar issue. esp32 dev kit was working fine for some months, then communication starts to break down. Change of the esp to new one solved the problem, but after a month of usage comes again. Inverter communication is ok, but bms is not. Changing of the port pins doesn't solve the problem either....

baturinivan, 2024-05-15:
> I have reflashed couple of times, seems to be working, so I will monitor it.
Not for a long time. Starts to have invalid header again and then seems to gone completely. 
Anyone have experienced esp32 malfuntion after some usage with jk bms?

virus100b, 2024-05-15:
Mine is working fine for 3 months now. The only problem i have is that sometimes is reporting balance ON when actually balance is OFF, checked by connecting tp the BMS over bluetooth. If i restart the ESP the Balance reporting is showing correct state. I don't know why is doing that.

syssi, 2024-07-24:
Resolved.

closed 2024-07-24 by syssi