syssi/esphome-jk-bms #397: Want to send the data received serially via SPI

OjasPatkar, 2023-11-25:
I am using esp32 in a ev project car and want to have that data transmitted 3 kms away at my pits using Ra-02 Ai Thinker LoRa module using semtech sx1278 chipset and I am programming esp 32 using arduino IDE so want to print the same data on serial monitor.

syssi, 2023-11-25:
You could describe your setup in detail? How many ESPs do you want to use and what kind of data do you want to transfer using SPI?

OjasPatkar, 2023-11-25:
So basically I'm competing in completion known as eBAJA where we make Electric ATV so I used your codes and we were successful in getting the bms data onto the esphome logger. But since we would be in pits and the driver would be 3-4 kms away from us and we don't have reliable display so we want to tell the driver via radio to come to pits for Battery swapping.
We are thinking of having 1 ESP 32 in the car which is connected to LoRa module using SPI protocol which would transmitted to receiver side having another LoRa module connected via SPI protocol to the ESP 32 which would be connected to the esphome. 
I would like to get all the data which is provided by the bms but specifically 
1. Total Voltage
2. Ampere
3. Remaining battery capacity
4. Battery temperature
5. Battery SoC

syssi, 2023-11-27:
Sounds like a pragmatic solution because there is no LoRa support in ESPHome yet. Passing some measurements via UART or SPI to a second ESP which is responsible for the radio stuff should be feasible. How can I support you?

OjasPatkar, 2023-11-28:
Actually we are short of space in the vehicle. So what if we add header files of lora in the configuration files ?

syssi, 2023-11-28:
> Actually we are short of space in the vehicle. So what if we add header files of lora in the configuration files.

To be able to use a single ESP?

OjasPatkar, 2023-11-28:
No no for the second ESP as you said

closed 2025-02-15 by syssi