syssi/esphome-jk-bms #398: Multiple external_components_source

adrusa, 2023-11-25:
Can I combine external_components_source? Can I for instance use a single ESP32 for both JBD and JK BSMes or JBD and Victron etc?

syssi, 2023-11-26:
Yes. This is possible if you merge the relevant parts of both examples. What do you want to combine exactly?

adrusa, 2023-11-26:
> Yes. This is possible if you merge the relevant parts of both examples. What do you want to combine exactly?

Thanks for your response. I'm looking to combine your JK solar and ANT BMS external components into a single ESP32 using BLE.

syssi, 2023-11-26:
BLE in both cases?

closed 2023-11-27 by syssi

adrusa, 2023-11-27:
> I've attached a YAML how it should look like: [esp32-jk-ant-combined.yaml.txt](https://github.com/syssi/esphome-jk-bms/files/13473437/esp32-jk-ant-combined.yaml.txt)

This did not work. The ANT bms did not respond.

syssi, 2023-11-27:
Did you update the MAC address? Does the BMS respond if the YAMLs aren't combined?

adrusa, 2023-11-27:
> Did you update the MAC address? Does the BMS respond if the YAMLs aren't combined?

Yes, I updated the mac address and yes, the ANT code worked when used alone.

syssi, 2023-11-27:
Which ESPHome version do you use?

adrusa, 2023-11-27:
The latest version.

syssi, 2023-11-28:
Which version exactly?

adrusa, 2023-11-28:
> Which version exactly?

2023.11.5

syssi, 2023-11-28:
Perfect. It's still strange that there are no multiple connections possible. In general it should work.