syssi/esphome-jk-bms #404: Can it add "Charge&Discharge Switch" function?

zjhcr, 2023-12-29:
the UART command is below:
------------------
"Turn ON Charge": [0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03,0X00, 0XAB, 0X01, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00, 0X00, 0X01, 0XD2]);

"Turn OFF Charge":[0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03, 0X00, 0XAB, 0X00, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00,0X00, 0X01, 0XD1]);

"Turn ON Discharge":[0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03, 0X00, 0XAC, 0X01, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00, 0X00, 0X01, 0XD3]);

"Turn OFF Discharge":[0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03, 0X00, 0XAC, 0X00, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00, 0X00, 0X01, 0XD2]);
-------------------
I have been tested these commands works well. I'd appreciate if you could add these function into esphome firmware? Thanks a lot.

syssi, 2023-12-29:
These commands are already implemented as `switch` entities: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/switch/__init__.py#L18-L19

closed 2023-12-29 by syssi

AlbertWete, 2024-01-19:
Hi, I get this.
INFO ESPHome 2023.12.5
INFO Reading configuration esp8266-example.yaml...
Failed config

switch.jk_bms: [source <unicode string>:265]

  Platform not found: 'switch.jk_bms'.
  platform: jk_bms
  charging:
    name: jk-bms charging
  discharging:
    name: jk-bms discharging

syssi, 2024-01-19:
@AlbertWete Please create a new issue and provide your complete configuration YAML (`esp8266-example.yaml`).