syssi/esphome-jk-bms #406: Can I use this integration to connect 3 BMS boards JK-BD6A20S10P

klugev303, 2024-01-03:
_**Good afternoon and Happy New Year!**_

Can I use this integration to connect **3*BMS boards JK-BD6A20S10P** ?
For this I will need 3*ESP32 boards? 
Or is it possible to code several BMS using one ESP32 board?
If it's not difficult, can you provide examples of writing code for 3 BMS boards? 
What is the best (more reliable) way to connect via Bluetooth or RS485?

_**Thank you!**_

_PS: I use ESP Home v. 2023.12.5
HomeAssistant:
Core 2023.12.1
Supervisor 2023.12.0
Operating System 11.2_

syssi, 2024-01-21:
These combinations are possible:

1. A single ESP32 which talks using three different `ble_client` connections to 3 BLE devices. Crashs are very likely.
2. A single ESP32 using three different hardware UARTs connected to 3 different BMS. Should be rock solid.
3. One ESP8266/ESP32 per device (BLE vs. RS485) for improved stability.

klugev303, 2024-01-26:
> These combinations are possible:
>
>     1. A single ESP32 which talks using three different `ble_client` connections to 3 BLE devices. Crashs are very likely.
>
>     2. A single ESP32 using three [...]

If it's not difficult, can you provide examples of writing code for 3 BMS boards?

syssi, 2024-01-27:
Please try the attached example: [esp32-ble-example-three-devices.yaml.txt](https://github.com/syssi/esphome-jk-bms/files/14072594/esp32-ble-example-three-devices.yaml.txt)

klugev303, 2024-01-27:
> Please try the attached example: [esp32-ble-example-three-devices.yaml.txt](https://github.com/syssi/esphome-jk-bms/files/14072594/esp32-ble-example-three-devices.yaml.txt)

Thank you very much, I will try it, all the best to you!

syssi, 2024-02-23:
IMO this depends how you power the ESP. What do you think?

Fliacik, 2024-02-23:
If you connect 3bms into one ESP you connect all bms grounds together. When one bms go to fail state or disconnect P- all traffic will be redirect to esp ground. Thats do loop gnd.

Woodstockskipper, 2024-07-16:
Hi Syssi, thank you verry much for this rep. I try to connect one ESP8266 to four JK-BMS by adapting your a.m. example code. Can you please help me and give me a hint how this lines:

> esp32_ble_tracker:
>   on_ble_advertise:
>     then:
>       - lambda: |-
>           if (x.get_name().rfind("JK-", 0) == 0 || x.get_name().rfind("JK_", 0) == 0) {
>             ESP_LOGI("ble_adv", "New JK-BMS found");
>             ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
>             ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
>             ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
>             for (auto uuid : x.get_service_uuids()) {
>               ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
>             }
>           }

have to be adapted to the ESP8266 board?
Thank you in advance.

closed 2024-08-13 by syssi