syssi/esphome-jk-bms #408: Could it work with new JK BMS's communication board's RS232 port?

txubelaxu, 2024-01-09:
This is the new JK BMS: https://www.jkbms.com/product/jk-pb2a16s15p-jk-pb2a16s-15p/

It has a comms board with 5 ports:
- 1 RS485 ((RJ45) to be connected to Inverter
- 1 CAN (RJ45) to be connected to Inverter
- 1 RS232 (RJ11)
- 2 RS485 (RJ45) for paralling other (same) BMS

I have given a try to RS232 port but ESP uses modbus protocol. 

Is any way to use serial protocol instead?

txubelaxu, 2024-01-10:
I have tested with no success. ESP32 (Serial2) connected directly to RS232 RJ11 connector. ESP32 RX pin in a pull-up input config.
I send:
`  byte bytesToSend[] = {0x4E, 0x57, 0x00, 0x13, 0x00, 0x00, 0x00, 0x00,
                        0x06, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                        0x68, 0x00, 0x00, 0x01, 0x29};` and... there is no response at all.
I send:
`  byte bytesToSend[] = {0x4E, 0x57, 0x00, 0x13, 0x00, 0x00, 0x00, 0x00,
                        0x06, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                        0x68, 0x00, 0x00, 0x01, 0x29, 0x13, 0x10};` and... no response at all.

I think it is because JK-BMS UART protocol is fixed to a kind of "GPS INPUT" and it cannot be changed from app:
[image]

If you try to change the setting, BMS responses that the new setting can not be saved.

syssi, 2024-01-11:
@txubelaxu How many items / protocols are available at the drop-down?

ibantxo, 2024-01-11:
Hi syssi,

There are 3 lines in the settings menu item (status/**settings**/control) regarding to protocols:
[image]
UART1 and CAN have "OK" button. UART2 has not. Anyway, "OK" button of UART1 does not work with all values, so any selection cannot be saved at this firmware/app version.

UART1 Protocol No.: has the next list inside:
[image]
[image]
[image]
[image]
Some of the items are disabled in the list. The other ones are enabled, so you can select from the list but when applying the change using "OK" button a "Sending failure" message can appears. If you reload the screen you can see that the old setting (000 - 4G...) is the active value. 
So:
- 000 - 4G-GPS Remote module.... is selectable & activable (and default)
- 001 - JK BMS 485 Modbus v1.0... is selectable BUT NOT activable
- 002 - NIU U SERIES... is selectable & activable
- 003 - China tower shared battery cabinet... is selectable & activable
- 004 - PACE_RS485_Modbus_v1.3... is NOT selectable because is disabled
- 005 - PYLON_Low_voltage_Protocol_RS4... is NOT selectable because is disabled
- 006 - Growatt_BMS_RS485_Protocol... is selectable & activable
- 007 - Voltronic_inverter_and_BMS_485... is selectable & activable
- 008 - China tower shared battery cabinet... is selectable & activable
- 009 - WOW_RS485_Modbus_V1.3... is selectable & activable
- 010 - Protocol 10... is NOT selectable because is disabled

UART2 Protocol No.: 1 is fixed and has not a list.

CAN Protocol No. has the next list (I have not tested what values are selectable at this moment):
[image]
[image]
[image]
[image]

txubelaxu, 2024-01-11:
Hi syssi,

There are 3 lines in the settings menu item (status/**settings**/control) regarding to protocols:
[image]
UART1 and CAN have "OK" button. UART2 has not. Anyway, "OK" button of UART1 does not work with all values, so any selection cannot be saved at this firmware/app version.

UART1 Protocol No.: has the next list inside:
[image]
[image]
[image]
[image]
Some of the items are disabled in the list. The other ones are enabled, so you can select from the list but when applying the change using "OK" button a "Sending failure" message can appears. If you reload the screen you can see that the old setting (000 - 4G...) is the active value. 
So:
- 000 - 4G-GPS Remote module.... is selectable & activable (and default)
- 001 - JK BMS 485 Modbus v1.0... is selectable BUT NOT activable
- 002 - NIU U SERIES... is selectable & activable
- 003 - China tower shared battery cabinet... is selectable & activable
- 004 - PACE_RS485_Modbus_v1.3... is NOT selectable because is disabled
- 005 - PYLON_Low_voltage_Protocol_RS4... is NOT selectable because is disabled
- 006 - Growatt_BMS_RS485_Protocol... is selectable & activable
- 007 - Voltronic_inverter_and_BMS_485... is selectable & activable
- 008 - China tower shared battery cabinet... is selectable & activable
- 009 - WOW_RS485_Modbus_V1.3... is selectable & activable
- 010 - Protocol 10... is NOT selectable because is disabled

UART2 Protocol No.: 1 is fixed and has not a list.

CAN Protocol No. has the next list (I have not tested what values are selectable at this moment):
[image]
[image]
[image]
[image]

txubelaxu, 2024-03-11:
See --> https://github.com/txubelaxu/esphome-jk-bms/blob/main/components/jk_rs485_bms/README.md

closed 2024-03-11 by txubelaxu

kptkip, 2024-05-29:
The given link describes RS485 not RS232 definition

gianfrdp, 2024-09-17:
I also am interested in RS232.
I tried to connect PC App BMS Monitor w/o success.
Could 00 protocol be same as old JK TTL one?

markottenheim, 2025-01-02:
I have the same problem, have you found the solution?

xiayu6186, 2025-09-22:
@txubelaxu 
JK BMS   4G-GPS protocol (id 000) How to switch to JK BMS RS485 Modbus V1.0 (id 001)？？？
4E 57 。。。。。。。？
怎么用指令把000的4G-GPS协议切换到001 modbus协议。。。切换后用modbus协议控制。。。。。