syssi/esphome-jk-bms #410: Change Pin code command?

JohnGalt1717, 2024-01-13:
For some reason the most recent JK software on both android and ios won't connect over bluetooth telling in invalid pin when I try and connect (with esphome stuff turned off)

So I'm hoping that I can change the pin to something that it will accept using this. Is there a way to manually set it over Bluetooth with this code?

syssi, 2024-01-13:
This project doesn't support pin codes because read & write is possible without authorization. ESPHome is still able to connect and retrieve measurements, right? Could you provide a copy of the device info frame (from the log)?

JohnGalt1717, 2024-01-13:
Here's the frame:

[15:42:53][I][jk_bms_ble:1158]: Device info frame (320 bytes) received
[15:42:53][I][jk_bms_ble:1210]:   Vendor ID: JK-B2A24S20P
[15:42:53][I][jk_bms_ble:1211]:   Hardware version: 10.X-W
[15:42:53][I][jk_bms_ble:1212]:   Software version: 10.02
[15:42:53][I][jk_bms_ble:1213]:   Uptime: 55481100 s
[15:42:53][I][jk_bms_ble:1214]:   Power on count: 22
[15:42:53][I][jk_bms_ble:1215]:   Device name: Battery 1
[15:42:53][I][jk_bms_ble:1216]:   Device passcode: 1234
[15:42:53][I][jk_bms_ble:1217]:   Manufacturing date: 220403
[15:42:53][I][jk_bms_ble:1218]:   Serial number: 1120303195
[15:42:53][I][jk_bms_ble:1219]:   Passcode: 0000
[15:42:53][I][jk_bms_ble:1220]:   User data: Input Userdata
[15:42:53][I][jk_bms_ble:1221]:   Setup passcode: !MyCode!

The !MyCode! is what I set it to before.

Yes, I can use this project to connect. It's just their app that won't connect anymore.

syssi, 2024-01-13:
> [15:42:53][I][jk_bms_ble:1221]: Setup passcode: !MyCode!

Alright! You are still remembering your changed pin code but the App doesn't accept the pin anymore? That's bad. Did you try to talk to the manufacturer / seller?

JohnGalt1717, 2024-01-13:
According to the solar forums, you have to contact them, then they give you some backdoor code after enabling you to use the backdoor in the software for a window of time. If you don't respond to the chinese response in time, it disables the backdoor again.

Trying to avoid going through that by resetting it myself.  Even if I could remote reset to factory defaults that would be a win.

syssi, 2024-01-13:
It looks like there was an version of the app some day in the past with a factory reset button. Another idea is to implement the "set password" frame. Both needs to be captured first. I don't know the commands yet.

23900, 2024-01-14:
If you can capture the MCU to Bluetooth UART data, then you may be able to see your pin code.
I saw the pin code returned by the MCU in the V11 version of BMS.

JohnGalt1717, 2024-01-14:
How do I go about doing that?

23900, 2024-01-14:
This is my BMS. In the MCU TX > Bluetooth module, you can capture the pin code using the serial cable. There is no need to connect a mobile phone, the pin code is at the end of the AT command.

**Be careful with the power supply! ! !** , the computer power supply and the BMS power supply cannot form a loop, otherwise the UART or BMS will be damaged.

[image]

syssi, 2024-01-14:
I will try to prepare a test setup later the week to be able to provide some more details.

23900, 2024-01-14:
I'm sorry, I made a mistake. In reality, the first response for the BLE connection is the pin code.
Hence, the log you provided earlier includes the pin code.

JohnGalt1717, 2024-01-14:
Neither 0000 nor the one that I commented out actually work in the app, hence why I'm at a loss.

syssi, 2024-01-14:
To be more explicit: You can still see measurements using the app but you cannot change the settings anymore because a authorization is required using the 6 digits pin code. Correct?

JohnGalt1717, 2024-01-14:
It's worse than that. With the latest versions of the app, what happens is that as soon as you tap on any device from the scan devices list it immediately asks for your pin code. Which it rejects so you never see anything.

JohnGalt1717, 2024-02-25:
Anyone find the password reset function? I have 6 jk bmses and all of them refuse to connect to the app anymore.

JohnGalt1717, 2024-02-25:
Update (weird):
This link:
https://mirofromdiro.github.io/JK-firmware-code/

Which allows you to force a firmware update on the new units, also works to get me into the app and see all of the settings.

Then the password that I had set works for changing the settings.

I changed the password in the app, closed the app, then restarted it, and it prompted for the password immediately again, for which I tried my changed password which didn't work.  I then put in the code from that website and it works again and the new password works for changing settings in the app.

Which leaves me really perplexed because the app always does this.

Any ideas?

closed 2024-02-25 by JohnGalt1717