syssi/esphome-jk-bms #414: Issue with multiple BMS over BLE & MQTT

luckylinux, 2024-01-15:
My goal was to put both BMS I have (JK BMS HW v11 & v10) for a 16s setup over MQTT using the esphome-jk-bms tool.

To recap:
- 1 x ESP32 for JK BMS Battery 01 (jk-bms-bat01)
- 1 x ESP32 for JK BMS Battery 02 (jk-bms-bat02)

I did that first for jk-bms-bat02. Everything worked. I get continuous MQTT messages every 10s or so.

This morning I flashed jk-bms-bat01 using the same config (minus the different hostname & MAC address of the BMS) and ... well ... both jk-bms-bat01 and jk-bms-bat02 sent ONE SINGLE SET of messages (all cell voltages, cell resistances, ...) AND THEN NOTHING.

When I programmed `jk-bms-bat01` this morning it seemed even finicky and the reboot loop afterwards threw yet another error message (cannot remember what exactly)

It's been over an hour and they are not sending any more responses.

What could be the cause of this ?

I can oberse this:
- When using PAHO Python MQTT client library
- When using
    - `mosquitto_sub -i MyTest -h 192.168.X.Y -p 1883 -t jk-bms-bat01/#`
    - `mosquitto_sub -i MyTest -h 192.168.X.Y -p 1883 -t jk-bms-bat02/#`
- MQTT topics should be different
- Mosquitto MQTT server shows NO FURTHER PUBLISH from either BMS

Possible causes:
- Long USB cable (IIRC 3 m). This was actually causing an issue immediately after programming. Don't remember the exact error, but looking online seemed to point to a bad 3.3V/5V supply
- EMI/EMC of Wireless ? 
- EMI/EMI of BLE ?

Both BMS are quite close together (~ 30 cm).
Both ESP32 are quite close together (~ 20 cm).

Wifi Access Point is some Hi-Power Alfa Wireless Adapter running HostAPd on Linux. The reception should be very good.

My `esp32-ble.yaml` file
```
substitutions:
  name: !secret host_name
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: !secret jk_mac_address
  # Defaults to "JK02" (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: !secret jk_protocol

esphome:
  name: ${name}
  platformio_options:
    build_flags:
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.5.0

esp32:
  #board: wemos_d1_mini32
  board: nodemcu-32s
  framework:
    type: esp-idf
  #  version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
[... 336 lines cut ...]
wifi_password: XXXXXXXXXXXXXXXXXX

jk_mac_address: $mac
jk_protocol: $protocol

mqtt_host: 192.168.X.Y
mqtt_port: 1883
mqtt_username: ""
mqtt_password: ""
EOF

# Validate the configuration, create a binary, upload it, and start logs
# If you use a esp8266 run the esp8266-examle.yaml
cp $currentpath/$esphomeconfig ./$esphomeconfig
esphome run $esphomeconfig

# Change back to currentpath
cd $currentpath

```

It's actually quite weird that it worked in the first place, because the access point is NOT supposed to have routing enabled between interfaces. And the subnet the Wifi Adapters should receive is 172.22.1.1/16 actually.
Maybe `mqtt_host: 172.22.1.1` would work better.

But why did it work in the first place ? And why it's not working with 2 adapters now ?

syssi, 2024-01-15:
Please attach one of the ESPs via USB to your PC and take a look at the local log book using:

```
esphome logs esp32-ble.yaml
```

Do you see any warnings? Please provide these logs. In best case one per ESP. All different root causes should be visible here:

1. Boot loop
2. Unable to setup the WiFi connection
3. Unable to establish a connection to the MQTT broker
4. Unable to connect to the BMS using BLE
5. ...

luckylinux, 2024-01-15:
Now I seem to receive continuously messages from both. Very weird

syssi, 2024-01-15:
Please try to replace the ESP32 if the setup remains unreliable.

luckylinux, 2024-01-15:
> Please try to replace the ESP32 if the setup remains unreliable.

Do you think it's a hardware issue?

I'm not sure it's maybe also triggered if I try to start the access point or Mqtt server.

Maybe I have to put the power supply for these Esp32 under a timer that automatically resets power ever few hours or so. Something like tasmota sonoff or similar. 

If seems one Esp32 was stuck in safe mode.

By the way, Jk-bms-bat01 had also some trouble flashing at 460000 (?), so it tried again at 115200 (?) serial communication and that I believe succeeded. 

I had used another desktop pc for that. Maybe I'll try reflashing with the old one, same as jk-bms-bat02.

syssi, 2024-01-15:
> Should I buy some other boards and try & see with those? Or do you habe other suggestions before I do that?

You could also wait some days. May be it's reliable now? In general this project / component doesn't need to be restarted periodically. My ESP wasn't touched for months now.

luckylinux, 2024-01-15:
> > Should I buy some other boards and try & see with those? Or do you habe other suggestions before I do that?
> 
> You could also wait some days. May be it's reliable now? In general this project / component doesn't need to be restarted periodically. My ESP wasn't touched for months now.

*RIGHT* now I am continuously receiving data after I gathered the logs where jk-bms-bat01 was stuck at boot ....
So BOTH jk-bms-bat01 and jk-bms-bat02 are periodically sending data right now.

What would it happen though:
- If Access Point (Wifi) disconnects or reboots ? Does that trigger a restart (and after 5 times it locks into Safe Mode) ?
- If MQQT / Mosquitto Server disconnects or reboots ? Does that trigger a restart (and after 5 times it locks into Safe Mode) ?

luckylinux, 2024-01-16:
Well it kinda worked for a while.

Today it seems it's jk-bms-bat02 this time that stopped sending new data.

From the access point I cannot pint the device any longer. I can ping jk-bms-bat01 without problem though.

syssi, 2024-01-20:
I'm using the debug level `INFO`:

```
logger:
  level: INFO
```

and the RTT is pretty constant:

```
$ ping -c10 attic-bms-ble.local
PING attic-bms-ble.local (192.168.1.32) 56(84) bytes of data.
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=1 ttl=255 time=18.2 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=2 ttl=255 time=13.4 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=3 ttl=255 time=13.4 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=4 ttl=255 time=14.5 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=5 ttl=255 time=15.2 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=6 ttl=255 time=15.6 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=7 ttl=255 time=11.6 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=8 ttl=255 time=16.1 ms
64 bytes from 192.168.1.32 (192.168.1.32): icmp_seq=9 ttl=255 time=14.3 ms
64 bytes from 192.168.1.32: icmp_seq=10 ttl=255 time=15.0 ms

--- attic-bms-ble.local ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 45161ms
rtt min/avg/max/mdev = 11.586/14.738/18.189/1.687 ms
```

luckylinux, 2024-02-25:
I put the ESP32 directly ON TOP of the BMS. That seemed to work for a few weeks.

Out of the blue, this morning, the ESPHome Reported "Offline" to MQTT (picked up by Home Assistant for Data Logging Purposes).

I'm still not sure whether this is just another EMC/EMI issue (causing the bluetooth connection to be interrupted) or the BMS shut down all of its functions while still allowing charge/discharge.

This is where I tried to ask for help assuming it is a hardware issue: https://diysolarforum.com/threads/urgent-jk-bms-went-offline-in-the-middle-of-the-night-soc-stuck-but-still-discharging.79532/

jogybaer0815, 2024-03-02:
Bluetooth and WiFi on the esp32 often seem to get in the way.
I therefore reduced the WiFi performance and run 5 bms and 5 esp32 here and I have a better result:
`wifi:'
...
'# output_power (Optional, string): The amount of TX power for the WiFi interface from 8.5dB to 20.5dB. Default for ESP8266 is 20dB, 20.5dB might cause unexpected restarts.'
 output_power: 8.5`
Please test

luckylinux, 2024-03-02:
> Bluetooth and WiFi on the esp32 often seem to get in the way. I therefore reduced the WiFi performance and run 5 bms and 5 esp32 here and I have a better result: `wifi:' ... '# output_power [...]

Right now (knock on wood) it has ran OK for a week or two DIRECTLY ON TOP of the BMS.

Can you please share the exact config and value for the `output_power` parameter ?

closed 2024-07-20 by syssi