syssi/esphome-jk-bms #415: Compiling not possible

Uli1900, 2024-01-16:
I want to create 

I tried it 30 times with different conditions:

1. restart (real restart) Raspi
2. Delete
3. Add Limit the amount of resources used for compiling

Here is the log:

INFO ESPHome 2023.12.6
INFO Reading configuration /config/esphome/dgzi3bildschirm.yaml...
INFO Updating https://github.com/Blackymas/NSPanel_HA_Blueprint@main
INFO Updating https://github.com/esphome/esphome.git@pull/6051/head
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing dgzi3bildschirm (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/dgzi3bildschirm/src/main.o
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program as
compilation terminated.
*** [.pioenvs/dgzi3bildschirm/src/main.o] Error 1
========================= [FAILED] Took 183.90 seconds =========================

syssi, 2024-01-16:
It looks like the as available memory (RAM) is too small.

Uli1900, 2024-01-16:
How can I make it bigger at the Raspberry Pi 4?

syssi, 2024-01-16:
I would spin up a virtual machine on a full blown PC. If you still want to use a Raspi probably a swap partition or compressed memory (zram) could be a workaround.