syssi/esphome-jk-bms #416: I struggle to flash OTA over network (using bluetooth)

tomatensaus, 2024-01-18:
Thank you for a brilliant project! I do not think people say this enought!

I am running a V11 of the JKBMS, flashing OTA over network only gets to 55% and then times out.
Running it 25 times maybe 1 will succeed.

The moment I disable the Bluetooth I can flash it over the network so I added the configuration below.
Maybe this can help someone? Maybe I can submit it as a PR to become part of the standard example?

Is this timeout a bug? Or just the way ESP32 bluetooth works? Bluetooth and Wifi compete for the same antenna.
Perhaps the WDT is timing out?

my workaround:
```
  - platform: ble_client
    ble_client_id: client0
    id: bluetooth_enabled_switch   # <--- add an ID so we can disable it from the OTA
    name: "${name} enable bluetooth connection"
```

and also

```
ota:
  on_begin:
    then:
      - lambda: |-
          id(bluetooth_enabled_switch).turn_off();
      - logger.log: "BLE shutdown for flashing"
```

syssi, 2024-01-18:
If the ESP to too busy processing stuff OTA updates doesn't succeed unfortunately. IMO it's a general issue of the ESPHome core and cannot be resolved here but I like your workaround! Could you prepare a pull request?

I've the same issue on a ESP32 node while receiving serial traffic on three different UARTs. I've added the `safe mode switch` component here to reboot the node into safe mode before doing OTAs: https://esphome.io/components/switch/safe_mode.html

tomatensaus, 2024-01-18:
I would gladly prepare a PR to include the workaround in the standard examples

testpaul999, 2024-01-18:
@tomatensaus 
THX cool feature run's like a charm

tomatensaus, 2024-01-19:
PR created https://github.com/syssi/esphome-jk-bms/pull/417

commit 8220ec80 (2024-01-20): Suspend BLE connection during OTA updates (Closes: #416) (#417)
closed 2024-01-20 by syssi