syssi/esphome-jk-bms #426: Pre-discharge via MQTT

AlbertWete, 2024-01-20:
I got the switches working via MQTT but found my JK_Bms also having a Pre-charge switch. This is a FET with a resistor in parallel over the discharge FET, the purpose of this is to charge capacitors in your load before switching on the discharge port and create a high current peak. It would be great to be able to have the possibility to switch on this pre-discharge circuit over MQTT
My BMS is:
JK_BD4A24S4P
HW Ver. V11.XW
SOFT Ver.11.38

syssi, 2024-01-21:
The command isn't supported yet. Do you own an Android device and could you prepare a btsnoop capture of the BLE traffic while turning the featre on/off?

syssi, 2024-01-21:
My BMS

```
[13:13:28][I][jk_bms_ble:1223]:   Vendor ID: JK_BD6A17S6P
[13:13:28][I][jk_bms_ble:1224]:   Hardware version: 11.XW
[13:13:28][I][jk_bms_ble:1225]:   Software version: 11.27
```

doesn't show this feature at the Android app. Could you provide a screenshot how it looks like on your site?

syssi, 2024-01-21:
Dischrg. Pre. Chrg. T(S) setting

```
aa:55:90:eb: 25 :04: 04:00:00:00 :36:d5:24:4a:68:8e:ad:95:5f:b7 # 0x04 = 4S
aa:55:90:eb: 25 :04: 05:00:00:00 :3c:35:b5:c4:8c:dd:6c:11:32:aa # 0x05 = 5S
```

closed 2025-07-25 by syssi