syssi/esphome-jk-bms #433: Heating switch not showing correct status

RFDarter, 2024-01-22:
I gave your latest update a try.
I enabled heating via the app and then started the esp.
The heating switch is turn off but it is activated in the app.
[image]

Hardware version: V11.XW
Software version: V11.287H

syssi, 2024-01-22:
Could you provide an ESPHome log with log level `VERY_VERBOSE` enabled? One log while the heating switch is enabled and a second one with a disabled heating switch? This will help to identify the position of the byte holding the state.

RFDarter, 2024-01-22:
Sure, what would you need? A part like this? 
`[21:46:01][V][esp32_ble_client:121]: [0] [C8:47:80:03:D2:09] gattc_event_handler: event=10 gattc_if=3
[21:46:01][D][esp32_ble_client:110]: [0] [C8:47:80:03:D2:09] ESP_GATTC_NOTIFY_EVT
[21:46:01][VV][jk_bms_ble:221]: Notification received: 55.AA.EB.90.02.61.F6.0C.F6.0C.F6.0C.F6.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0F.00.00.00.F6.0C.00.00.00.00.31.00.34.00.32.00.32.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[21:46:01][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[21:46:01][V][esp32_ble_client:121]: [0] [C8:47:80:03:D2:09] gattc_event_handler: event=10 gattc_if=3
[21:46:01][D][esp32_ble_client:110]: [0] [C8:47:80:03:D2:09] ESP_GATTC_NOTIFY_EVT
[21:46:01][VV][jk_bms_ble:221]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.72.00.00.00.00.00 (22)
[21:46:01][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[21:46:01][V][esp32_ble_client:121]: [0] [C8:47:80:03:D2:09] gattc_event_handler: event=10 gattc_if=3
[21:46:01][D][esp32_ble_client:110]: [0] [C8:47:80:03:D2:09] ESP_GATTC_NOTIFY_EVT
[21:46:01][VV][jk_bms_ble:221]: Notification received: D8.33.00.00.28.11.00.00.B5.FE.FF.FF.40.00.47.00.00.00.08.00.00.00.00.5E.FE.05.04.00.C0.45.04.00.00.00.00.00.FB.1E.03.00.64.00.00.00.B6.E4.2C.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.3D.03.00.00.04.00.88.64.40.40.00.00.00.00.2F.05.00.00.00.01.00.01.C8.05.00.00.3C.F3.D4.00.00.00.00.00.72.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[21:46:01][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[21:46:01][V][esp32_ble_client:121]: [0] [C8:47:80:03:D2:09] gattc_event_handler: event=10 gattc_if=3
[21:46:01][D][esp32_ble_client:110]: [0] [C8:47:80:03:D2:09] ESP_GATTC_NOTIFY_EVT
[21:46:01][VV][jk_bms_ble:221]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.01.00.00.00.00.1D (22)`

for how long would I have to log it to save a full frame?

RFDarter, 2024-01-22:
[turned_off2 copy.log](https://github.com/syssi/esphome-jk-bms/files/14015888/turned_off2.copy.log)
[turned_on2 copy.log](https://github.com/syssi/esphome-jk-bms/files/14015889/turned_on2.copy.log)

syssi, 2024-01-23:
I will have a look this evening and try to provide some details/answers.

RFDarter, 2024-01-23:
I did a Bt_ snoop
[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/14028766/btsnoop_hci.log)

`aa5590eb270401000000580de144323db10d61be
aa5590eb270400000000cae722b2cab7837b6a13`

syssi, 2024-01-23:
Controlling the heating switch should work already. Just the switch state doesn't support feedback at the moment. I'm trying to identify the `heating status` byte in the cell info frame at the moment. You capture is very helpful already. I've extracted a few `cell info frames` (each bunch after heating turned on/off).

The first line (`-1`) indicates the heating was turned on (`01`) or turned off (`00`). Hopefully we can identify the `heating status` and the `heating current`  somewhere at the payload:

```
    -1  00 00 00 00 01 01 01 01 00 00 00 00 01 01 01 01 00 00 00 00 01 01 01 01
     0  55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55
     1  aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa
     2  eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb
     3  90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90
     4  02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02
     5  bd be bf c0 c3 c4 c5 c6 d3 d4 d5 d6 df e0 e1 e2 eb ec ed ee f7 f8 f9 fa
     6  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
     7  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
     8  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
     9  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    10  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    11  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    12  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    13  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    14  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    15  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    16  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    17  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    18  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    19  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    20  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    21  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    22  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    23  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    24  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    25  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    26  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    27  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    28  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    29  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    30  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    31  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    32  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    33  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    34  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    35  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    36  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    37  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    38  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    39  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    40  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    41  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    42  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    43  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    44  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    45  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    46  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    47  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    48  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    49  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    50  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    51  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    52  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    53  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    54  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    55  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    56  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    57  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    58  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    59  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    60  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
   295  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   296  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   297  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   298  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   299  97 9d a5 ab c1 c9 cf d7 33 39 41 47 8b 91 99 9f de e6 ec f4 34 3a 42 49
```

Do you have a heating element attached? Is there a `heating current` shown the at android app? How big is the heating current approx?

RFDarter, 2024-01-23:
> Do you have a heating element attached? Is there a `heating current` shown the at android app? How big is the heating current approx?

Give me a sec i will set it up.
In the postred snoop the heating control was just toggled.
Heating itself was never turned on because the temp abd the setting was to hight for it to trigger heating

syssi, 2024-01-23:
The extracted `cell info frames` above are strange. It looks like they doesn't include any details about the `heating status`. Furthermore it looks like you haven't attached any cells to the BMS at the moment. The large amount of zeros is unusual.

I was looking for a pattern matching with the indicated state of line `-1`. If I remove all equal bytes/lines these bytes remain:

```
    -1  00 00 00 00 01 01 01 01 00 00 00 00 01 01 01 01 00 00 00 00 01 01 01 01

   194  0f 0f 10 10 12 13 13 14 1b 1b 1c 1c 22 22 23 23 28 29 29 2a 2f 2f 30 31

   246  98 9d a3 a8 b9 bf c4 ca 11 16 1c 21 56 5b 61 66 97 9d a2 a8 da df e5 ea
   247  1e 1e 1e 1e 1e 1e 1e 1e 1f 1f 1f 1f 1f 1f 1f 1f 1f 1f 1f 1f 1f 1f 1f 1f
```

RFDarter, 2024-01-23:
> Furthermore it looks like you haven't attached any cells to the BMS at the moment

yep, just B+ and B- is connected to power up the bms

syssi, 2024-01-23:
All `settings frame` from your btsnoop capture above:

```
     0  55 55 55
     1  aa aa aa
     2  eb eb eb
     3  90 90 90
     4  01 01 01
     5  b0 b4 b5
     6  58 58 58
     7  02 02 02
     8  00 00 00
     9  00 00 00
    10  04 04 04
    11  0b 0b 0b
    12  00 00 00
    13  00 00 00
    14  22 22 22
    15  0b 0b 0b
    16  00 00 00
    17  00 00 00
    18  68 68 68
    19  10 10 10
    20  00 00 00
    21  00 00 00
    22  54 54 54
    23  10 10 10
    24  00 00 00
    25  00 00 00
    26  0a 0a 0a
    27  00 00 00
    28  00 00 00
    29  00 00 00
    30  54 54 54
    31  10 10 10
    32  00 00 00
    33  00 00 00
    34  54 54 54
    35  0b 0b 0b
    36  00 00 00
    37  00 00 00
    38  00 00 00
    39  00 00 00
    40  00 00 00
    41  00 00 00
    42  00 00 00
    43  00 00 00
    44  00 00 00
    45  00 00 00
    46  f0 f0 f0
    47  0a 0a 0a
    48  00 00 00
    49  00 00 00
    50  a8 a8 a8
    51  61 61 61
    52  00 00 00
    53  00 00 00
    54  1e 1e 1e
    55  00 00 00
    56  00 00 00
    57  00 00 00
    58  3c 3c 3c
    59  00 00 00
    60  00 00 00
    61  00 00 00
    62  40 40 40
    63  0d 0d 0d
    64  03 03 03
    65  00 00 00
    66  2c 2c 2c
    67  01 01 01
    68  00 00 00
    69  00 00 00
    70  3c 3c 3c
    71  00 00 00
    72  00 00 00
    73  00 00 00
    74  3c 3c 3c
    75  00 00 00
    76  00 00 00
    77  00 00 00
    78  d0 d0 d0
    79  07 07 07
    80  00 00 00
    81  00 00 00
    82  bc bc bc
    83  02 02 02
    84  00 00 00
    85  00 00 00
    86  58 58 58
    87  02 02 02
    88  00 00 00
    89  00 00 00
    90  bc bc bc
    91  02 02 02
    92  00 00 00
    93  00 00 00
    94  58 58 58
    95  02 02 02
    96  00 00 00
    97  00 00 00
    98  38 38 38
    99  ff ff ff
   100  ff ff ff
   101  ff ff ff
   102  9c 9c 9c
   103  ff ff ff
   104  ff ff ff
   105  ff ff ff
   106  e8 e8 e8
   107  03 03 03
   108  00 00 00
   109  00 00 00
   110  20 20 20
   111  03 03 03
   112  00 00 00
   113  00 00 00
   114  14 14 14
   115  00 00 00
   116  00 00 00
   117  00 00 00
   118  00 00 00
   119  00 00 00
   120  00 00 00
   121  00 00 00
   122  00 00 00
   123  00 00 00
   124  00 00 00
   125  00 00 00
   126  00 00 00
   127  00 00 00
   128  00 00 00
   129  00 00 00
   130  40 40 40
   131  9c 9c 9c
   132  00 00 00
   133  00 00 00
   134  dc dc dc
   135  05 05 05
   136  00 00 00
   137  00 00 00
   138  b8 b8 b8
   139  0b 0b 0b
   140  00 00 00
   141  00 00 00
   142  00 00 00
   143  00 00 00
   144  00 00 00
   145  00 00 00
   146  00 00 00
   147  00 00 00
   148  00 00 00
   149  00 00 00
   150  00 00 00
   151  00 00 00
   152  00 00 00
   153  00 00 00
   154  00 00 00
   155  00 00 00
   156  00 00 00
   157  00 00 00
   158  00 00 00
   159  00 00 00
   160  00 00 00
   161  00 00 00
   162  00 00 00
   163  00 00 00
   164  00 00 00
   165  00 00 00
   166  00 00 00
   167  00 00 00
   168  00 00 00
   169  00 00 00
   170  00 00 00
   171  00 00 00
   172  00 00 00
   173  00 00 00
   174  00 00 00
   175  00 00 00
   176  00 00 00
   177  00 00 00
   178  00 00 00
   179  00 00 00
   180  00 00 00
   181  00 00 00
   182  00 00 00
   183  00 00 00
   184  00 00 00
   185  00 00 00
   186  00 00 00
   187  00 00 00
   188  00 00 00
   189  00 00 00
   190  00 00 00
   191  00 00 00
   192  00 00 00
   193  00 00 00
   194  00 00 00
   195  00 00 00
   196  00 00 00
   197  00 00 00
   198  00 00 00
   199  00 00 00
   200  00 00 00
   201  00 00 00
   202  00 00 00
   203  00 00 00
   204  00 00 00
   205  00 00 00
   206  00 00 00
   207  00 00 00
   208  00 00 00
   209  00 00 00
   210  00 00 00
   211  00 00 00
   212  00 00 00
   213  00 00 00
   214  00 00 00
   215  00 00 00
   216  00 00 00
   217  00 00 00
   218  00 00 00
   219  00 00 00
   220  00 00 00
   221  00 00 00
   222  00 00 00
   223  00 00 00
   224  00 00 00
   225  00 00 00
   226  00 00 00
   227  00 00 00
   228  00 00 00
   229  00 00 00
   230  00 00 00
   231  00 00 00
   232  00 00 00
   233  00 00 00
   234  00 00 00
   235  00 00 00
   236  00 00 00
   237  00 00 00
   238  00 00 00
   239  00 00 00
   240  00 00 00
   241  00 00 00
   242  00 00 00
   243  00 00 00
   244  00 00 00
   245  00 00 00
   246  00 00 00
   247  00 00 00
   248  00 00 00
   249  00 00 00
   250  00 00 00
   251  00 00 00
   252  00 00 00
   253  00 00 00
   254  00 00 00
   255  00 00 00
   256  00 00 00
   257  00 00 00
   258  00 00 00
   259  00 00 00
   260  00 00 00
   261  00 00 00
   262  00 00 00
   263  00 00 00
   264  00 00 00
   265  00 00 00
   266  00 00 00
   267  00 00 00
   268  00 00 00
   269  00 00 00
   270  00 00 00
   271  00 00 00
   272  00 00 00
   273  00 00 00
   274  00 00 00
   275  00 00 00
   276  00 00 00
   277  00 00 00
   278  b0 b0 b0
   279  71 71 71
   280  0b 0b 0b
   281  00 00 00
   282  00 00 00
   283  00 00 00
   284  3c 3c 3c
   285  32 32 32
   286  00 00 00
   287  fc fc fc
   288  f9 f9 f9
   289  ff ff ff
   290  ff ff ff
   291  9f 9f 9f
   292  2d 2d 2d
   293  00 00 00
   294  00 00 00
   295  00 00 00
   296  00 00 00
   297  00 00 00
   298  00 00 00
   299  8c 90 91
```

RFDarter, 2024-01-23:
I cant seem to get the heating status to change from OFF to ON even it the temp is below my UTP.
It works that way in my 8s bms
Maybe they canged the software to require  a charge current to be present to switch the heating on? Or its ist just bc. I have no cells connected? On my other bms it definitly Switches heating on without a charge current present 
[image]

[image]

syssi, 2024-01-23:
I would be happy about another btsnoop capture with a working heating element attached. Please switch the views each time your change the heating switch. I'm interested in a `cell info frame` (retrieved at the status page of the app) and `settings frame` (retrieved if you enter the settings view) per (heating) state.

syssi, 2024-01-23:
Are you able to do the btsnoop capture of your "working setup"? A `heating current` + `heating status = on` included in the capture would be helpful.

RFDarter, 2024-01-23:
> Are you able to do the btsnoop capture of your "working setup"? A `heating current` + `heating status = on` included in the capture would be helpful.

My working setup it outside, not access to a pc.
Is the bt snoop only captured while if i have my phone connected to the pc, or could I get there, do the stuff and the come back, connect my phone to the pc and get the bt log?

RFDarter, 2024-01-23:
It turnes out 2 AA Cells can be uses to get the bms to funktion normaly :D

[Uploading btsnoop_hci.log…]()

Main Page, heating OFF,
Still Main Page heating ON , 0,5A
Still Main Page heating OFF
Still Main Page heating ON , 0,5A
Switch to Settings Page
Switch to Control Page
Switch to Settings Page
Switch to Main Page
Still Main Page heating OFF
Disconnect

RFDarter, 2024-01-23:
Another go

[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/14029594/btsnoop_hci.log)

Connect-> Settings->Password->Control->Main
Heating turns on 0.5A
->Settings -> Control -> Main

syssi, 2024-01-23:
> My working setup it outside, no access to a pc.
Is the bt snoop only captured while if i have my phone connected to the pc, or could I get there, do the stuff and the come back, connect my phone to the pc and get the bt log?

The btnsoop capture is stored on the phone, truncated/overridden on every "bluetooth on/off" and can be retrieved using adb / as part of the bug report bundle.

syssi, 2024-01-23:
> Connect-> Settings->Password->Control->Main
Heating turns on 0.5A
->Settings -> Control -> Main

Good job! It's late. I will have a look tomorrow!

RFDarter, 2024-01-24:
I tried to make sense of what i am seeing here.
These are the first few paketes after switching from control to Main with heating off and the same for wehen heating is on with 0.5A

```
//off
830004001b120055aaeb9002 52 bc05ba0500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000bb0502000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
830004001b12 00 76 0b 00 00 00 00 00 00 00 00 00 00 e5 00 ee 00 01 00 08 000000000000000000409c0000000000005700000064000000 bb 1d 000001010000000000000000000000000000ff0001000000da0300000000f2d74040000000002501000000010001000500005229010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f0101010000000042

830004001b120055aaeb9002 53 bc05b90500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000ba0502000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
830004001b12 00 75 0b 00 00 00 00 00 00 00 00 00 00 e5 00 ee 00 01 00 08 000000000000000000409c0000000000005700000064000000 bb 1d 000001010000000000000000000000000000ff0001000000da0300000000f2d74040000000002501000000010001000500005729010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f0101010000000045

830004001b120055aaeb9002 54 bc05ba0500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000bb0502000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
830004001b12 00 76 0b 00 00 00 00 00 00 00 00 00 00 e5 00 ee 00 01 00 08 000000000000000000409c0000000000005700000064000000 bc 1d 000001010000000000000000000000000000ff0001000000da0300000000f2d74040000000002501000000010001000500005d29010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f0101010000000050

830004001b120055aaeb9002 55 bc05ba0500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000bb0502000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
830004001b12 00 76 0b 00 00 00 00 00 00 00 00 00 00 e5 00 ee 00 01 00 08 000000000000000000409c0000000000005700000064000000 bd 1d 000001010000000000000000000000000000ff0001000000da0300000000f2d74040000000002501000000010001000500006229010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f0101010000000057

830004001b120055aaeb9002 56 bc05b90500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000ba0502000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
830004001b12 00 75 0b 00 00 00 00 00 00 00 00 00 00 e5 00 ee 00 01 00 08 000000000000000000409c0000000000005700000064000000 bd 1d 000001010000000000000000000000000000ff0001000000da0300000000f2d74040000000002501000000010001000500006729010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f010101000000005a

//on
830004001b120055aaeb9002 89 b105ba0500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000b50508000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
830004001b12 00 6b 0b 00 00 3f 02 00 00 3b ff ff ff 97 00 96 00 01 02 08 000000000000000000409c0000000000005900000064000000 da 1d 000000010000000000000000000000000000ff0101000000da0300000100f2d7404000000000240100020001000100050000 8c 2a 010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f0101010000000096

830004001b120055aaeb9002 8a b105ba0500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000b50508000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
830004001b12 00 6b 0b 00 00 3f 02 00 00 3b ff ff ff 97 00 96 00 01 02 08 000000000000000000409c0000000000005900000064000000 db 1d 000000010000000000000000000000000000ff0101000000da0300000100f2d7404000000000240101020001000100050000 92 2a 010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f010101000000009f

830004001b120055aaeb9002 8b b105b90500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000000b50508000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
190004001b120000000000000000000000000000000000010103000000
[... cut ...]
830004001b12 00 6b 0b 00 00 3f 02 00 00 3b ff ff ff 97 00 96 00 01 02 08 000000000000000000409c0000000000005900000064000000 da 1d 000000010000000000000000000000000000ff0101000000da0300000100f2d7404000000000240100020001000100050000 8c 2a 010000000000010100000000000000000000000000000000000000000000
190004001b1200000000000000000000feff7fdc2f0101010000000096
```

This part seems to be allways the same
```
830004001b120055aaeb9002
```
The following
```52```
```53```
```54```
```..```
seems to be just a counter.
Since you can allready parse most of the infos out of these packets you must understand the structur of the data, do you have any documentation about the bt protocol?

syssi, 2024-01-24:
> Since you can allready parse most of the infos out of these packets you must understand the structur of the data, do you have any documentation about the bt protocol?

There is no protocol description unfortunately. This is what you need to know:

```
  // Byte Len  Payload                Content              Coeff.      Unit        Example value
  // 0     2   0x55 0xAA 0xEB 0x90    Header
  // 4     1   0x02                   Record type
  // 5     1   0x8C                   Frame counter
```
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L313C24-L586C2

A frame starts with a 4 byte preamble. A better name for byte 5 (`record type`) would be `frame type`. I'm aware of 3 different frame types:

```
0x01 = settings frame
0x02 = cell info frame
0x03 = device info frame
``` 

You can find a decoder/method per frame type here: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp

In your case we are talking about the `jk02` protocol/methods and because of the firmware version greater 11 the payload supports 32 cells (protocol flavor = `FRAME_VERSION_JK02_32S`). This means you have to use an offset of `32`.

For example:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L440

The 4 bytes holding the `charge current` are at position `126` (of older firmware versions). Since firmware version 11 the BMS supports up to 32 cells. Therefore the `charge current` has shifted a bit and is available at position `126 + 32` now.

syssi, 2024-01-24:
I will share this evening some details how to extract the most important parts from the btsnoop capture: 

```
btatt.opcode == 0x1b || btatt.opcode == 0x52
File -> Export Packet Dissection -> as JSON

cat packets.jsom | grep btatt.value | cut -d\" -f4
```

syssi, 2024-01-24:
```
     0  55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55:55
     1  aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa:aa
     2  eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb:eb
     3  90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90:90
     4  02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02:02
     5  f7:f8:f9:fa:fe:ff:00:91:92:93:94:95:06:07:08:09:ab:ac:ad:ae:af:b0:f0:f1:f2:f3:f4:f5:58:59:5a:5b:5c:8d:8e:8f:90:91:fa:fb:fc:9e:9f:a0:a1:75:76:77:78:79:2a:2b:2c:2d:2e:23:24:25
     6  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:bb:ba:ba:ba:af:af:b0:b0:b0:b0:b0:b0:b0:b0:bb:bb:bb
     7  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05
     8  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:b9:b9:b9:b6:b6:b7:b6:b7:b7:b7:b7:b7:b7:b7:b7:b9:b9:b9:b9:b9
     9  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05:05
    10  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    11  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    12  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    13  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    14  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    15  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    16  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    17  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    18  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    19  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    20  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    21  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    22  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    23  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    24  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    25  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    26  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    27  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    28  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
    29  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
   297  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
   298  00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
   299  34:3a:42:4f:56:5d:65:5e:65:6d:73:7b:94:9b:a2:a9:1c:22:2a:31:37:3f:04:0a:11:18:1f:26:e6:ed:f5:fd:04:5c:64:6a:72:78:26:2e:34:6d:72:7d:80:97:a9:b5:c6:cd:46:4d:52:5d:63:5a:5d:64
```

RFDarter, 2024-01-25:
I made little Pyhton Script that takes the Jason file , takes the "btatt.value", combines the 4 Paketes and formates them into a vertical string for easy comparison.

left is heaing off, right is heating on. Could it be the Byte 249?

```
000 		55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 -> 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55
001 		aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa -> aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa
002 		eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb -> eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb eb
003 		90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 -> 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90
004 		02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 -> 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02
005 		c2 c3 c4 c5 c6 c7 c8 c9 ca cb cf e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed -> 89 8a 8b 8d 8e 8f 90 91 92 93 94 95 aa ab ac ad ae af b0 b1 b2 b3 b4 b5 b6
006 		bb bb bb bb bb bb bb bb bb bb bb b0 b0 b0 b0 b0 af b0 b0 b0 b0 b0 af -> b1 b1 b1 b1 b1 b1 b1 b1 b1 b1 b1 b1 b1 b1 bc bb bc bc bc bc bc bc bc bc bc
007 		05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 -> 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05
008 		b7 b7 b9 b7 b9 b7 b7 b7 b7 b7 b7 b7 b7 b7 b7 b7 b7 b9 b7 b7 b7 b7 b7 -> ba ba b9 ba ba ba ba ba ba b9 b9 ba ba ba ba ba ba b9 ba ba ba b9 ba ba ba
009 		05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 -> 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05
010 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
011 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
012 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
013 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
014 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
015 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
016 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
017 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
018 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
019 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
020 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
021 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
022 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
023 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
024 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
025 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
026 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
027 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
028 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
029 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
030 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
031 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
032 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
033 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
296 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
297 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
298 		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
299 		95 9c a8 ac b6 ba c0 c7 ce d5 f1 be b1 b8 3b 43 46 50 53 57 5d 66 6b -> 96 9f a3 b2 ba c2 c8 d1 d7 de e4 ee d9 0a 2e 32 da dd f9 00 07 25 2f 4a 51
```

RFDarter, 2024-01-25:
```
import argparse
import json
import os

def extract_btatt_values(json_data):
  extracted_data = []

  for entry in json_data:
    if '_source' in entry and 'layers' in entry["_source"] and 'btatt' in entry["_source"]["layers"] and 'btatt.value' in entry["_source"]["layers"]["btatt"]:
        btatt_value = entry["_source"]["layers"]["btatt"]["btatt.value"]
        extracted_data.append(''.join(btatt_value))
  return extracted_data

def append_values(values): #beginng of a packet is "55:aa:eb:90", append every packet after that till we hit "55:aa:eb:90" again
  appended_values = []
  current_packet = None
  
  for value in values:
      if value.startswith("55:aa:eb:90"):
          if current_packet: #add the previus combined value
              appended_values.append(''.join(current_packet))
          current_packet = [value]
      else:
          current_packet.append(':' + value)

  print("current_packet")
  print(len(current_packet))
  if current_packet and len(current_packet) == 4:
      appended_values.append(''.join(current_packet))
      
  return appended_values
         

def values_to_vertical_string(data):
  hex_values = []
  output_string = ""
  for value in data:
    hex_values.append(value.split(":"))
  matrix = [[0 for _ in range(len(hex_values))] for _ in range(len(hex_values[0]))] 
  
  for i_col, row in enumerate(hex_values):
    for i_row, hex_val in enumerate(row):
      matrix[i_row][i_col] = hex_val +" "

  for i, col in enumerate(matrix):
    output_string += str(i).zfill(3) + " \t\t"
    for row in col:
      output_string += row
    output_string += "\n"
  return output_string

def load_json_from_file(file_path):
    with open(file_path, 'r') as file:
        data = json.load(file)
    return data
  
def save_to_file(file_name, data):
  with open(file_name + "_output", 'w') as file:
      file.write(data)
      file.close()

parser = argparse.ArgumentParser(description='')
parser.add_argument('file_path', type=str, help='Path to JSON-Datei')
args = parser.parse_args()
file_path = args.file_path

json_data = load_json_from_file(file_path)
btatt_values = extract_btatt_values(json_data)
appended_values = append_values(btatt_values)
vertival_string = values_to_vertical_string(appended_values)

file_name, file_extension = os.path.splitext(file_path)
save_to_file(file_name, vertival_string)
```

RFDarter, 2024-01-25:
heating switch on-off-on-off
https://drive.google.com/file/d/1BeKPQGYpS2A9Vjrc1Z5FDd2hd1jE5b9l/view?usp=sharing
[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/14057493/btsnoop_hci.log)

heating is on. heating current is changing between 0.5A and 1.0A
https://drive.google.com/file/d/1Bh-ygBfquGwDdEiYad-obvXYupxzWWnA/view?usp=sharing
[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/14057495/btsnoop_hci.log)

RFDarter, 2024-01-25:
225 0x00  -> heating off
225 0x01  -> heating on

237/238  -> heating current in mA

syssi, 2024-01-26:
Wow! Awesome findings. My week was pretty busy. I will try to implement the new measurements this weekend!

RFDarter, 2024-01-26:
I allready implemented it last night.
The Current works fine, but it seems 224 is not correct for the heating status.

Also your version assumtion is not working correctly.
Since my models (B2A20S20P and B1A8S20P) are also Hardware Version 11 H.X your Version assumtion sets the protocol to the 32S version and the offset to 16.

But these models are the old style Bms, seem to have the same Hardware version as the New style, but only support max. 24S.

I have the day off today. I will try to find the correct position for the heating status and prepare a PR.

syssi, 2024-01-26:
I've removed the auto-detection. You have to choose the protocol version now (`JK02_24S` vs. `JK02_32S`).

RFDarter, 2024-01-26:
```c++
  // 224   1   0x01                   Heating status          0x00: off, 0x01: on 
  uint16_t raw_heating_status = (((224 + offset) != 0x00));
  ESP_LOGD(TAG, "  offset: %d", offset);
  ESP_LOGD(TAG, "  Heating status: %s", (raw_heating_status > 0) ? "on" : "off");
  this->publish_state_(this->heating_binary_sensor_, (bool)raw_heating_status);

   // 236   2   0x01 0xFD              Heating current        0.001         A
  this->publish_state_(this->heating_current_sensor_, (float) ((int16_t) jk_get_16bit(236 + offset)) * 0.001f);
```

RFDarter, 2024-01-26:
[heating_off_on_off_on_off_on_output.log](https://github.com/syssi/esphome-jk-bms/files/14064018/heating_off_on_off_on_off_on_output.log)

224 definitly turns to 0x01 when 236 shows the current and goes back to 0x00 when no currents is present at 236

syssi, 2024-01-26:
Okay. I guess you have found the positions of `heating status` and `heating current` but we cannot use `heating status` to maintain the switch state of the `heating switch` (because there is some more logic involved and the `heating status` isn't always in sync with the `heating switch` state).

Could you try to analyse the `settings frame` too? Is the state of the `heating switch` persisted here?

RFDarter, 2024-01-26:
> ```c++
>   // 224   1   0x01                   Heating status          0x00: off, 0x01: on
>   uint16_t raw_heating_status = (((224 + offset) != 0x00));
>   ESP_LOGD(TAG, "  offset: %d", offset); [...]

```c++
uint16_t raw_heating_status = (((224 + offset) != 0x00));
``` 
stupid me, this can not work

RFDarter, 2024-01-26:
> Could you try to analyse the `settings frame` too? Is the state of the `heating switch` persisted here?

`settings frame` is recived when first connecting to the bms?
And also when switching to the settings page?
Do you know what my phone sends to the bms when switching to the settings page?
I get a lot of paketes send from my phone to the bms and i can not clearly identify which one is the one i switched to the settings page.
I also have the problem that the btsnoop is not reset after turning my bt on and off, so i get a lot of old date i have to be aware of.

syssi, 2024-01-26:
> settings frame is recived when first connecting to the bms?
> And also when switching to the settings page?

Correct.

> Do you know what my phone sends to the bms when switching to the settings page?

Yes. There is a `retrieve settings` button entity to retrieve/update the settings: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/button/__init__.py#L20

> I also have the problem that the btsnoop is not reset after turning my bt on and off, so i get a lot of old date i have to be aware of.

That's strange. Normally the file is moved and re-created on bluetooth off+on but it's pretty dependend on your Android version and manufacturer.

RFDarter, 2024-01-26:
[heating_setting_switch_on_off_main_settings_main_settings._output.log](https://github.com/syssi/esphome-jk-bms/files/14065170/heating_setting_switch_on_off_main_settings_main_settings._output.log)

282 seems to be the only thing changing if i activate or deactivate heating

RFDarter, 2024-01-26:
OK, it seems to be taht bit 1 in 282 is heating. Bit set -> heating enabled

RFDarter, 2024-01-26:
Got i all working. Preparing a PR now.

But i think you need to overthink the version control, because this is definitly not true anymore

```
  # Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
```

`
JK02_24S` would need to include Hardware version > 11 now

And the new Style bms definitly supports 32 cells and that's why the `JK02_32S` is needed at all?

PR #436 (2024-01-27): expose heating status/current implements #420 fixes #433
implements #420
fixes #433
commit f828d7b7 (2024-01-27): Expose heating status/current (#436)

Implements #420 
Fixes #433
closed 2024-01-27 by syssi