syssi/esphome-jk-bms #437: instable connecting to more than one BMS over BLE

octane321, 2024-01-27:
Hi there,
i tried to get multiple BMSs connecting over BLE - but if a activate all (for me) relevant sensors with two JKs, the esp is getting very instable and crashes every 5-20 minutes. Goal would be to connect to 3 JKs

i tried to debug the situation and think the heap is the problem.
Could it be a possible solution using an ESP-CAM or WROVER with 4/8MB PSRAM to increase the available heap?

octane321, 2024-01-27:
I also added delta Filter on all sensors to reduce publishing to mqtt broker wich helps a bit

`    - platform: jk_bms_ble
      jk_bms_ble_id: jk_bms_ble_JK_BMS2
    # voltages
      total_voltage:
        id: total_voltage_bms2
        name: "BMS2 Voltage"
        filters:
          - delta: 0.01
        state_topic: ${prefix}/${subtopic}/JK_BMS2/01_Battery/01_Voltage`

syssi, 2024-01-27:
Decreasing the log level to `INFO` or `WARN` should free some resources too. I don't if there are ESP32 flavors which will improve the situation. If you have the chance you could try to avoid WiFi and use a ethernet connection instead. IMO concurrency (WiFi vs. BLE) is also responsible for stability issues.

Are you aware of the `throttle` parameter to drop readings? The JK-BMS publishes frames/notifications in a fixed interval (0.5s) which cannot be controlled. Throwing frames away is the only option to reduce the update interval.

syssi, 2024-01-27:
Some more options you could play around with:

```
esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
```

```
esp32_ble_tracker:
  scan_parameters:
    interval: 500ms
```

octane321, 2024-01-27:
thx for the quick reply
loglevel is already on WARN

with the throttle i played from 1s-60s but everything above 5s does not show further improvements

i already played around with the sdkconfig_options but couldnt see any improvment

    sdkconfig_options:
      CONFIG_BT_BLE_ACT_SCAN_REP_ADV_SCAN: y
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"

using LAN would be a good idea - but i have only a W5500 ethernet shield (not supported by esphome) but i will give it a try anyway

Also i´m a bit confused, because switching between boards with esp32dev 320kb SRAM or az-delivery-devkit-v4 520kb SRAM does not make a difference in heap usage (always 21kb free heap).
Does that make sense to you?

syssi, 2024-01-27:
> Does that make sense to you?

I guess this is the answer: https://community.platformio.org/t/esp32-boards-with-320kb-or-520kb/18700/2

TL&DR:  A board definition error in PlatformIO.

octane321, 2024-01-27:
thx for your reply - okay that sounds logical.

I realy have trouble picking the right board definition for the ARCELI ESP32-WROVER-DEV (Amazon ASIN B0CP7SJ4P2)
 could manage to upload a compiled bin which gives me around 60kb free heap instead of around 20kb. Thats was looking promising, but damnit...i cant remember the settings - trying to find them again.
The board should have 4MB flash + 4MB PSRAM but i cant find the right preconfigured board in esphome

23900, 2024-01-27:
I think Bluetooth consumes the most resources. If you skip Bluetooth and use uart directly, 1/3 of the resources will be saved.

23900, 2024-01-28:
> using LAN would be a good idea - but i have only a W5500 ethernet shield (not supported by esphome) but i will give it a try anyway

https://github.com/syssi/esphome-zb-gw03/blob/main/packages/ethernet.yaml

octane321, 2024-01-28:
> I think Bluetooth consumes the most resources. If you skip Bluetooth and use uart directly, 1/3 of the resources will be saved.
i know - one of my BMSs has a broken UART Port - so BLE is the only way. Another benefit is the galvanically isolation using BLE

> > using LAN would be a good idea - but i have only a W5500 ethernet shield (not supported by esphome) but i will give it a try anyway
> 
> https://github.com/syssi/esphome-zb-gw03/blob/main/packages/ethernet.yaml
What are you pointing at?

octane321, 2024-01-28:
I have tried for hours and hours to find the right configuration to reproduce the mentioned bin file...but i cant get this done.

If i use the already compiled bin file, everything runs perfectly fine and rock solid over 8h without any error or warning - still lots of ressources available.

[image]

DutchKillerbee, 2024-02-21:
@octane321 I cannot help you with finding your crucial parameter but I had good results by using different throttle times for both BLE clients:

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: **3s**
    id: bms0
  - ble_client_id: client1
    throttle: **5s**
    id: bms1

Even with the enabled webserver running on a wemos S3-mini
For more than 24 hours a constant flow of these messages:
[image]

Wiki591, 2024-06-25:
> [octane321 wrote:]
> I also added delta Filter on all sensors to reduce publishing to mqtt broker wich helps a bit
> 
> ```
> - platform: jk_bms_ble
>   jk_bms_ble_id: jk_bms_ble_JK_BMS2
> # voltages
>   total_voltage:
>     id: total_voltage_bms2
>     name: "BMS2 Voltage"
>     filters:
>       - delta: 0.01
> ```
> [...]

I've struggled around with reading three Neey balancer using one esp32-C3 getting the data continously into homeassistant with almost no luck. The ESP was sending the data correctly, but HA was not able to get and display them in time neither by api nor by MQTT. Setting the filter especially for the cell voltages (filters set to - delta: 0.001) did the job. Thanks for sharing it.

holle75, 2024-07-19:
> I've struggled around with reading three Neey balancer using one esp32-C3 getting the data continously into homeassistant with almost no luck. The ESP was sending the data correctly, but HA was not [...]

isn´t 0,001 the smallest unit the BMS is transmitting anyhow? So in theory this filter wouldn´t reduce data? Just curious.

octane321, 2024-07-20:
it reduces data - cause the value has to be changed by 0.001 - so if the value hasn't changed, there will be no update

holle75, 2024-07-20:
Ah, sure, got it.
So you would have to add

filters:
      - delta: 0.001
      
separately to every (cell voltages) sensor? Isn´t there a wildcard (to match all cell voltage sensors) or similar?

syssi, 2024-07-20:
There is no wildcard. A filter must be applied individually per sensor. If you don't need a fast update interval of the sensors you should prefer the global `throttle` option.

syssi, 2024-07-20:
Let's close this issue. Feel free to re-open the issue if the topic isn't resolved.

closed 2024-07-20 by syssi