syssi/esphome-jk-bms #441: Informer LCD ST7789V display on one ESP32 with bluetuls connection

odis68, 2024-02-01:
Please help me solve the problem of simultaneously connecting JK BMS and LCD ST7789V display.
Separately, the display works and separately bluetuls interclogation on Esphome.
At the same time, the display and Blutuls do not work, they do not work.
ESP 32 is constantly rebooted.
I would like to make a screen on the battery itself and receive full detailed data on Home Assistant

syssi, 2024-02-01:
Please provide your complete YAML configuration and a ESPHome log which contains the crash/exception causing the boot loop.

syssi, 2024-02-01:
This is the important part:

```
[16:45:58][E][display:017]: Could not allocate buffer for display!Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[16:45:58]
[16:45:58]Core  1 register dump:
[16:45:58]PC      : 0x4000c46c  PS      : 0x00060e30  A0      : 0x800e6491  A1      : 0x3ffcf3e0  
[16:45:58]A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0001c200  A5      : 0x00000000  
[16:45:58]A6      : 0x000000f1  A7      : 0x00001c20  A8      : 0x801af29c  A9      : 0x3ffcf3b0  
[16:45:58]A10     : 0x0000e100  A11     : 0x3f4013c9  A12     : 0x00000011  A13     : 0x3f4011ae  
[16:45:58]A14     : 0x00000028  A15     : 0x00020000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
[16:45:58]EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00001c1f  
[16:45:58]
[16:45:58]
[16:45:58]Backtrace: 0x4000c469:0x3ffcf3e0 0x400e648e:0x3ffcf3f0 0x400ea610:0x3ffcf410 0x401af515:0x3ffcf430 0x400ea16c:0x3ffcf450 0x400eeb43:0x3ffcf480 0x400d8c7b:0x3ffcf570
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400e648e: esphome::st7789v::ST7789V::setup() at /tmp/.esphome/build/jkbms-monitor-ble/src/esphome/components/st7789v/st7789v.cpp:120
WARNING Decoded 0x400ea610: esphome::PollingComponent::call_setup() at /tmp/.esphome/build/jkbms-monitor-ble/src/esphome/core/component.cpp:189
WARNING Decoded 0x401af515: esphome::Component::call() at /tmp/.esphome/build/jkbms-monitor-ble/src/esphome/core/component.cpp:98
WARNING Decoded 0x400ea16c: esphome::Application::setup() at /tmp/.esphome/build/jkbms-monitor-ble/src/esphome/core/application.cpp:38
WARNING Decoded 0x400eeb43: setup() at /tmp/.esphome/build/jkbms-monitor-ble/<unicode string>:1716
WARNING Decoded 0x400d8c7b: esphome::loop_task(void*) at /tmp/.esphome/build/jkbms-monitor-ble/src/esphome/components/esp32/core.cpp:67
```

It looks like there is not enough memory available.

odis68, 2024-02-01:
RAM?

syssi, 2024-02-01:
Yes. The display component tries to allocate some bytes as frame buffer but there is no free RAM available anymore.

syssi, 2024-02-01:
Probably this hint could help you: https://github.com/esphome/issues/issues/3877#issuecomment-1338893224

odis68, 2024-02-01:
Thank you !!!!!!
EightBitcolor: True
Helped

closed 2024-02-01 by syssi