syssi/esphome-jk-bms #442: Gathering Multiple JK-BMS' status info via RS-485 in a 1x MASTER (address 0x00) and 1x CLIENT configuration

txubelaxu, 2024-02-01:
This is the new JK BMS: https://www.jkbms.com/product/jk-pb2a16s15p-jk-pb2a16s-15p/

It has a comms board with 5 ports:

1 RS485 ((RJ45) to be connected to Inverter
1 CAN (RJ45) to be connected to Inverter
1 RS232 (RJ11)
2 RS485 (RJ45) for paralling other (same) BMS

I have given a try to RS485 ports using a linux ubuntu + USB2RS485 converter + cable attached to any of the RS485 ports of the BMSs.
The BMS number 0 has address=0x00 (MASTER)
The BMS number 1 has address=0x01 (SLAVE)

The JK Windows program is working with this configuration and wiring. So, I have given a try to linux and python3 serial... with some success? I don't know. In a listening/sniffing mode, all the communication in the RS485 bus is received by python program:

THIS IS THE python3 code
```
import serial
import sys

def read_serial_port(serial_port, baud_rate):
    with serial.Serial(serial_port, baud_rate, timeout=0) as ser:
        while True:
            # Read all available bytes from the serial port
            data = ser.read_all()

            # If there is data, process and display in hexadecimal format
            if data:
                hex_data = data.hex()
                print(hex_data, end='\n')
                sys.stdout.flush()

        # If there is no more data, exit the loop
        sys.stdout.flush()

if __name__ == "__main__":
    # Serial port configuration (adjust the port and baud rate based on your setup)
    serial_port = '/dev/ttyUSB0'  # Change this to the serial port you are using
    baud_rate = 115200  # Change this to the baud rate of your device

    read_serial_port(serial_port, baud_rate)
```
Is esphome-jk-bms compatible with this configuration in order to gathering information?

This is some of the traffic in hex:
```
55aaeb900200120d120d110d110d110d110d110d110d110d110d110d110d110d110d120d100d0000000000000000000000000000000000000000000000000000000000000000ffff0000110d01000002530050004f004a004d004b004d004d0053004e004d004a004c004d00520051000000000000000000000000000000000000000000000000000000000000000000dd000000000014d100000000000000000000c700cd000000000000000064aa9e040080a3040000000000b30c00006400000048a3290001010000000000000000000000000000ff00010000009a030000000060543f4000000000e81400000001010100060000f2765c0000000000dd00
c700ce009a03453f09007f0000008051010000000000000000000000000000feff7fdd2f0101b0070000008c001016200001059a
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456
011016200001020000d6f1
55aaeb900205120d120d120d120d120d120d120d120d120d110d120d120d120d110d120d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d0100000a4d004b004d004b004d004a004c0049004c004b004e004c0050004e0050004e000000000000000000000000000000000000000000000000000000000000000000ce00000000001fd100000000000000000000c000c7000000000000000064aa9e040080a3040000000000df04000064000000451f0c0001010000000000000000000000000000ff00010000009303000000000bc23f4000000000e914000000010101000600009c6d5c0000000000ce00
bf00c1008d03553e0900090000008051010000000000000000000000000000feff7fdd2f0101b00700000017
011016200001044b
0110161e0001020000d22f
55aaeb900105ac0d0000280a00005a0a00001a0e0000780d000005000000790d0000500a00007a0d0000160d0000c4090000a8610000030000003c000000400d03002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000010000000500000060e3160010023c3218feffffffbfe901020000000000
0110161e00016587
021016200001020000c201
031016200001020000cf91
041016200001020000e9a1
051016200001020000e431
061016200001020000f0c1
071016200001020000fd51
081016200001020000bca1
091016200001020000b131
0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
[... 160 lines cut ...]
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456
011016200001020000d6f1
55aaeb900205120d120d120d120d120d110d120d110d120d120d110d110d120d110d120d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d010000064d004b004d004b004d004a004c0049004c004b004e004c0050004e0050004e000000000000000000000000000000000000000000000000000000000000000000ce00000000001fd100000000000000000000c000c7000000000000000064aa9e040080a3040000000000df04000064000000771f0c0001010000000000000000000000000000ff00010000009303000000000bc23f4000000000e91400000001010100060000946f5c0000000000ce00
bf00c1008d03873e0900090000008051010000000000000000000000000000feff7fdd2f0101b0070000006e
011016200001044b
0110161e0001020000d22f
55aaeb900105ac0d0000280a00005a0a00001a0e0000780d000005000000790d0000500a00007a0d0000160d0000c4090000a8610000030000003c000000400d03002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000010000000500000060e3160010023c3218feffffffbfe901020000000000
0110161e00016587
021016200001020000c201
031016200001020000cf91
041016200001020000e9a1
051016200001020000e431
061016200001020000f0c1
071016200001020000fd51
081016200001020000bca1
091016200001020000b131
0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
0e10162000010200009701
0f10162000010200009a91
```

jogybaer0815, 2024-02-08:
i have to adress bms1 to adress 1, bms2 to adress 2 and so on
The JK programm also cannot connect, if there is the first bms adressed to 0

txubelaxu, 2024-02-23:
[image]

In the "internal" JK RS485 network (using right connectors), 1 of the JK-BMS must be "MASTER" (RS485 0x00) and other JK-BMS must be "SLAVE" (RS485 address inside interval: [0x01,0x0f]). MASTER asks in a pooling mode to each address from 0x01 to 0x0f (always). Slave receiving "asking message" answers.
So, in RS485 network cannot be another master node asking anything. The only way is to insert a "sniffer" and listen to the network gathering information.
Now I am using one ESP32 via bluetooth for one BMS, and another ESP32 via bluetooth for the other BMS. Both ESP32 are integrated with Home Assistant. The target is to have only one ESP32 gathering information from this RS485 network. This way, bluetooth is free to connect with mobile when needed.

In my previous messages the info was fragmented because a bug in my capture script. 

These are some examples of captured information, if can be of your interest:

[20240223_172400.log](https://github.com/syssi/esphome-jk-bms/files/14388928/20240223_172400.log)
[20240223_172455.log](https://github.com/syssi/esphome-jk-bms/files/14388929/20240223_172455.log)
[20240223_172751.log](https://github.com/syssi/esphome-jk-bms/files/14388930/20240223_172751.log)
[20240223_172756.log](https://github.com/syssi/esphome-jk-bms/files/14388931/20240223_172756.log)
[20240223_173223.log](https://github.com/syssi/esphome-jk-bms/files/14388932/20240223_173223.log)

denveronly, 2024-03-01:
Great research!.

So the problem is as understood that there could be only one MASTER that is querying SLAVES.
If we add ESP32 it cannot be Master, and as a SLAVE it cannot query other Status?

The only working way is Bluetooth ESP32 for multiple BMS (have 3)
Is there more efficient way then 3x ESP32, Maybe there are some Multi Bluetooth versions or something?

txubelaxu, 2024-03-01:
>If we add ESP32 it cannot be Master, and as a SLAVE it cannot query other SLAVES and MASTER bms Status?
That is correct. One of the BMSs in the network is the Master. ESP32 cannot be another Master. Only 1 Master in a RS485/Modbus network. An slave cannot query other slaves.
The only way I have found is to "listening to" or "sniffing" the traffic.
I am developing a code "jk_modbus2" that is a mixture of jk_bms_ble & jk_modbus.
I have got compiled OK, but it is not working by the moment.
Anyway, I could upload that code.

denveronly, 2024-03-01:
> If we add ESP32 it cannot be Master, and as a SLAVE it cannot query other SLAVES and MASTER bms Status?
> 
> That is correct. One of the BMSs in the network is the Master. ESP32 cannot be another Master. Only 1 Master in a RS485/Modbus network. An slave cannot query other slaves. The only way I have found is to "listening to" or "sniffing" the traffic. I am developing a code "jk_modbus2" that is a mixture of jk_bms_ble & jk_modbus. I have got compiled OK, but it is not working by the moment. Anyway, I could upload that code.
> 
> The main advantage of RS485 is that one ESP32 could manage all the BMSs in the network and the bluetooth could it be available to connect with other device to control it (change a setting...)

Sounds extremely fantastic! 
Will be waiting for any updates

txubelaxu, 2024-03-03:
There is a new pull request that has the code. https://github.com/syssi/esphome-jk-bms/pull/464

txubelaxu, 2024-03-11:
See --> https://github.com/txubelaxu/esphome-jk-bms/blob/main/components/jk_rs485_bms/README.md

closed 2024-03-17 by txubelaxu