syssi/esphome-jk-bms #444: Neey EK-24S4EB just "NAN" in MQTT

kthemall, 2024-02-08:
hi, could you add support for:

EK-24S4EB
Hardware Version: HW-3.2.0
Software Version: ZH-1.3.2
Production Date: 20230829

is there anything i can help?

thx tom

syssi, 2024-02-08:
Could you provide an `VERY_VERBOSE` ESPHome log?

kthemall, 2024-02-08:
hope that file helps:
[debug.txt](https://github.com/syssi/esphome-jk-bms/files/14209439/debug.txt)

syssi, 2024-02-08:
```
[13:37:50][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[13:37:50][VV][esp32_ble_tracker:423]: Parse Result:
[13:37:50][VV][esp32_ble_tracker:440]:   Address: 3C:A5:51:97:4F:1B (PUBLIC)
[13:37:50][VV][esp32_ble_tracker:442]:   RSSI: -86
[13:37:50][VV][esp32_ble_tracker:443]:   Name: 'neey'
[13:37:50][VV][esp32_ble_tracker:451]:   Ad Flag: 6
[13:37:50][VV][esp32_ble_tracker:454]:   Service UUID: 0xFFE0
[13:37:50][VV][esp32_ble_tracker:454]:   Service UUID: 0xFEE7
[13:37:50][VV][esp32_ble_tracker:457]:   Manufacturer data: 88.A0.3C.A5.51.97.4F.1B (8)
[13:37:50][VV][esp32_ble_tracker:473]: Adv data: 02.01.06.05.02.E0.FF.E7.FE.0B.FF.0D.72.88.A0.3C.A5.51.97.4F.1B.0B.09.6E.65.65.79.00.00.00.00.00.00 (33)
[13:37:50][D][esp32_ble_client:053]: [0] [3C:A5:51:97:4F:1B] Found device
...
[13:37:50][I][esp32_ble_client:069]: [0] [3C:A5:51:97:4F:1B] 0x00 Attempting BLE connection
...
[13:38:06][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 40
[13:38:06][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=40 gattc_if=3
[13:38:06][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 2
[13:38:06][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=2 gattc_if=3
[13:38:06][V][esp32_ble_client:134]: [0] [3C:A5:51:97:4F:1B] ESP_GATTC_OPEN_EVT
[13:38:06][V][esp32_ble:314]: (BLE) gap_event_handler - 20
[13:38:07][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 46
[13:38:07][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=46 gattc_if=3
[13:38:07][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[13:38:07][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=7 gattc_if=3
[13:38:07][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[13:38:07][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=7 gattc_if=3
[13:38:07][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 6
[13:38:07][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=6 gattc_if=3
[13:38:07][V][esp32_ble_client:194]: [0] [3C:A5:51:97:4F:1B] ESP_GATTC_SEARCH_CMPL_EVT
[13:38:07][V][esp32_ble_client:197]: [0] [3C:A5:51:97:4F:1B] Service UUID: 0x1800
[13:38:07][V][esp32_ble_client:199]: [0] [3C:A5:51:97:4F:1B]  start_handle: 0x1  end_handle: 0x7
[13:38:07][V][esp32_ble_client:197]: [0] [3C:A5:51:97:4F:1B] Service UUID: 0xFFE0
[13:38:07][V][esp32_ble_client:199]: [0] [3C:A5:51:97:4F:1B]  start_handle: 0x8  end_handle: 0xffff
[13:38:07][I][esp32_ble_client:201]: [0] [3C:A5:51:97:4F:1B] Connected
[13:38:07][V][esp32_ble_client:069]: [0] [3C:A5:51:97:4F:1B]  characteristic 0xFFE1, handle 0xa, properties 0x1c
[13:38:07][V][esp32_ble_client:069]: [0] [3C:A5:51:97:4F:1B]  characteristic 0xFFE2, handle 0xd, properties 0x1c
[13:38:07][V][esp32_ble_client:069]: [0] [3C:A5:51:97:4F:1B]  characteristic 0xFFE3, handle 0x10, properties 0xc
[13:38:07][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 18
[13:38:07][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=18 gattc_if=3
[13:38:07][V][esp32_ble_client:165]: [0] [3C:A5:51:97:4F:1B] cfg_mtu status 0, mtu 244
[13:38:07][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 38
[13:38:07][V][esp32_ble_client:119]: [0] [3C:A5:51:97:4F:1B] gattc_event_handler: event=38 gattc_if=3
[13:38:07][I][jk_bms_ble:213]: Request device info
[13:38:07][D][jk_bms_ble:1131]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
...
```

kthemall, 2024-02-08:
i can see the blue led icon shining (not blinking) so i think bluetooth is connected. isn´t it?

thx tom

syssi, 2024-02-08:
Yes. But it looks like the device doesn't respond to the requests/commands of other models.

kthemall, 2024-02-08:
ok, i understand, anything i can assist you in getting it to work with the new firmware? thx tom

syssi, 2024-02-08:
Are you able to prepare a btsnoop capture of the BLE traffic between the Android App and your balancer?

kthemall, 2024-02-08:
oh no, i do not have android smartphones, i use iphone with the enerkey app, sorry. i don´t even know what btsnoop capture is.

well i think we have to wait for someone that is able to do that
hmmm
thx tom

syssi, 2024-02-08:
Could you provide a link to the Android app which should be able to talk to your device model?

kthemall, 2024-02-08:
i´m no android user but i found that:

[https://github.com/NEEY-electronic/NEEY-electronic]

syssi, 2024-02-11:
I've collected some details:

1. The [NEEY V3.1 android app](https://github.com/NEEY-electronic/NEEY-electronic/blob/main/NEEY_V3.1_2021-11-01.apk)  cannot communicate with the new hardware model => this project isn't compatible at the moment too
2. A android app called `neey-v1.2.14.apk` or `zy-v1.2.24.apk` must be used instead. 
3. Some Aliexpress product pages are pointing to www.enerkey.cn (but the page is unavailable ATM)

Kippervink, 2024-02-14:
Hi, I have this balancer and would love to get it working in home assistant. I have Android, and the app that works with the balancer. How can I help get support added?

Kippervink, 2024-02-14:
Hi @syssi Is this what you need? 
[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/14283506/btsnoop_hci.log)

syssi, 2024-02-14:
@Kippervink What's the MAC address of your balancer? The btsnoop capture contains just some communication with a Wacom MAC address. A device named `SPEN NA`. I guess this is a pen. Could you try to prepare another capture?

Kippervink, 2024-02-14:
[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/14284787/btsnoop_hci.log)
@syssi sorry, that was my first time creating a log. MAC is 3C:A5:51:95:7D:90. New log file attached. If that doesn't contain the right info, is there some step I am missing? I just turn on the logging, connect to the balancer, then disconnect and create a bugreport.

syssi, 2024-02-14:
The new capture looks pretty good.

```
Filter: bluetooth.addr == 3C:A5:51:95:7D:90 and (btatt.opcode == 0x1b || btatt.opcode == 0x52)
```

Extracted traffic:

```
>> aa:55:11:01:01:00:14:00:00:00:00:00:00:00:00:00:00:00:26:ff
<< 55:aa:11:01:01:00:64:00:45:4b:2d:32:34:53:34:45:42:00:00:00:00:00:00:00:48:57:2d:33:2e:32:2e:30:5a:48:2d:31:2e:32:2e:39:56:31:2e:32:2e:39:00:00:32:30:32:33:30:36:30:38:07:00:00:00:f1:17:85:00:33:43:41:35:35:31:39:35:37:44:39:30:31:32:33:34:35:36:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:7b:ff
>> aa:55:11:01:04:00:14:00:00:00:00:00:00:00:00:00:00:00:29:ff
<< 55:aa:11:01:04:00:64:00:10:0a:d7:a3:3b:00:00:80:40:00:00:20:40:01:01:02:18:01:00:00:66:66:26:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:b7:ff:55:aa:11:01:02:00:2c:01:ca:4e:78:42:40:96:2a:46:40:47:2e:46:40:e7:26:46:40:d1:2b:46:40:71:24:46:40:d1:2b:46:40:47:2e:46:40:21:28:46:40:96:2a:46:40:0c:2d:46:40:e7:26:46:40:18:41:45:40:ac:25:46:40:0c:2d:46:40:ff:7b:42:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6b:c3:26:3e:cc:93:28:3e:4e:68:29:3e:91:da:27:3e:67:e2:27:3e:c9:12:28:3e:74:ec:28:3e:13:99:26:3e:d9:bc:26:3e:ac:f1:28:3e:6c:8a:28:3e:c8:a1:26:3e:b2:9e:27:3e:b9:36:25:3e:b5:3d:25:3e:0b:0d:26:3e:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:8e:a5:45:42:8e:a5:45:40:40:7e:6d:3d:07:00:0f:05:ae:72:81:c0:3e:0a:62:42:3e:0a:62:42:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:27:32:85:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:a9:ff
>> aa:55:11:01:02:00:14:00:00:00:00:00:00:00:00:00:00:00:27:ff
>> aa:55:11:01:04:00:14:00:00:00:00:00:00:00:00:00:00:00:29:ff
<< 55:aa:11:01:04:00:64:00:10:0a:d7:a3:3b:00:00:80:40:00:00:20:40:01:01:02:18:01:00:00:66:66:26:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:b7:ff
<< 55:aa:11:01:02:00:2c:01:cb:4e:78:42:40:96:2a:46:40:47:2e:46:40:e7:26:46:40:d1:2b:46:40:71:24:46:40:d1:2b:46:40:47:2e:46:40:21:28:46:40:96:2a:46:40:0c:2d:46:40:e7:26:46:40:18:41:45:40:ac:25:46:40:0c:2d:46:40:ff:7b:42:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6b:c3:26:3e:cc:93:28:3e:4e:68:29:3e:91:da:27:3e:67:e2:27:3e:c9:12:28:3e:74:ec:28:3e:13:99:26:3e:d9:bc:26:3e:ac:f1:28:3e:6c:8a:28:3e:c8:a1:26:3e:b2:9e:27:3e:b9:36:25:3e:b5:3d:25:3e:0b:0d:26:3e:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:8e:a5:45:42:8e:a5:45:40:40:7e:6d:3d:07:00:0f:05:bf:f0:81:c0:3e:0a:62:42:3e:0a:62:42:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:28:32:85:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:3a:ff
>> aa:55:11:01:02:00:14:00:00:00:00:00:00:00:00:00:00:00:27:ff
<< 55:aa:11:01:02:00:2c:01:cc:4e:78:42:40:96:2a:46:40:47:2e:46:40:e7:26:46:40:d1:2b:46:40:71:24:46:40:d1:2b:46:40:47:2e:46:40:21:28:46:40:96:2a:46:40:0c:2d:46:40:e7:26:46:40:18:41:45:40:ac:25:46:40:0c:2d:46:40:ff:7b:42:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6b:c3:26:3e:cc:93:28:3e:4e:68:29:3e:91:da:27:3e:67:e2:27:3e:c9:12:28:3e:74:ec:28:3e:13:99:26:3e:d9:bc:26:3e:ac:f1:28:3e:6c:8a:28:3e:c8:a1:26:3e:b2:9e:27:3e:b9:36:25:3e:b5:3d:25:3e:0b:0d:26:3e:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:8e:a5:45:42:8e:a5:45:40:40:7e:6d:3d:07:00:0f:05:2f:df:82:c0:3e:0a:62:42:3e:0a:62:42:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:28:32:85:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:9b:ff
<< 55:aa:11:01:02:00:2c:01:cd:4e:78:42:40:96:2a:46:40:47:2e:46:40:e7:26:46:40:d1:2b:46:40:71:24:46:40:d1:2b:46:40:47:2e:46:40:21:28:46:40:96:2a:46:40:0c:2d:46:40:e7:26:46:40:18:41:45:40:ac:25:46:40:0c:2d:46:40:ff:7b:42:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6b:c3:26:3e:cc:93:28:3e:4e:68:29:3e:91:da:27:3e:67:e2:27:3e:c9:12:28:3e:74:ec:28:3e:13:99:26:3e:d9:bc:26:3e:ac:f1:28:3e:6c:8a:28:3e:c8:a1:26:3e:b2:9e:27:3e:b9:36:25:3e:b5:3d:25:3e:0b:0d:26:3e:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:8e:a5:45:42:8e:a5:45:40:40:7e:6d:3d:07:00:0f:05:f2:47:80:c0:3e:0a:62:42:3e:0a:62:42:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:29:32:85:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:c6:ff
<< 55:aa:11:01:02:00:2c:01:d1:4e:78:42:40:96:2a:46:40:47:2e:46:40:e7:26:46:40:d1:2b:46:40:71:24:46:40:d1:2b:46:40:47:2e:46:40:21:28:46:40:96:2a:46:40:0c:2d:46:40:e7:26:46:40:18:41:45:40:ac:25:46:40:0c:2d:46:40:ff:7b:42:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:6b:c3:26:3e:cc:93:28:3e:4e:68:29:3e:91:da:27:3e:67:e2:27:3e:c9:12:28:3e:74:ec:28:3e:13:99:26:3e:d9:bc:26:3e:ac:f1:28:3e:6c:8a:28:3e:c8:a1:26:3e:b2:9e:27:3e:b9:36:25:3e:b5:3d:25:3e:0b:0d [... cut ...] 9:26:3e:d9:bc:26:3e:ac:f1:28:3e:6c:8a:28:3e:c8:a1:26:3e:b2:9e:27:3e:b9:36:25:3e:b5:3d:25:3e:0b:0d:26:3e:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:53:a5:45:42:53:a5:45:40:00:07:6f:3d:0a:00:0f:05:52:ef:7e:c0:52:b8:60:42:52:b8:60:42:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:2f:32:85:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:83:ff
```

syssi, 2024-02-14:
Short responses (100 bytes):

```
     0  55 55 55 55 55
     1  aa aa aa aa aa
     2  11 11 11 11 11
     3  01 01 01 01 01
     4  04 04 04 04 04
     5  00 00 00 00 00
     6  64 64 64 64 64
     7  00 00 00 00 00
     8  10 10 10 10 10
     9  0a 0a 0a 0a 0a
    10  d7 d7 d7 d7 d7
    11  a3 a3 a3 a3 a3
    12  3b 3b 3b 3b 3b
    13  00 00 00 00 00
    14  00 00 00 00 00
    15  80 80 80 80 80
    16  40 40 40 40 40
    17  00 00 00 00 00
    18  00 00 00 00 00
    19  20 20 20 20 20
    20  40 40 40 40 40
    21  01 00 00 01 01
    22  01 01 01 01 01
    23  02 02 02 02 02
    24  18 18 18 18 18
    25  01 01 01 01 01
    26  00 00 00 00 00
    27  00 00 00 00 00
    28  66 66 66 66 66
    29  66 66 66 66 66
    30  26 26 26 26 26
    31  40 40 40 40 40
    32  00 00 00 00 00
    33  00 00 00 00 00
    34  00 00 00 00 00
    35  00 00 00 00 00
    36  00 00 00 00 00
    37  00 00 00 00 00
    38  00 00 00 00 00
    39  00 00 00 00 00
    40  00 00 00 00 00
    41  00 00 00 00 00
    42  00 00 00 00 00
    43  00 00 00 00 00
    44  00 00 00 00 00
    45  00 00 00 00 00
    46  00 00 00 00 00
    47  00 00 00 00 00
    48  00 00 00 00 00
    49  00 00 00 00 00
    50  00 00 00 00 00
    51  00 00 00 00 00
    52  00 00 00 00 00
    53  00 00 00 00 00
    54  00 00 00 00 00
    55  00 00 00 00 00
    56  00 00 00 00 00
    57  00 00 00 00 00
    58  00 00 00 00 00
    59  00 00 00 00 00
    60  00 00 00 00 00
    61  00 00 00 00 00
    62  00 00 00 00 00
    63  00 00 00 00 00
    64  00 00 00 00 00
    65  00 00 00 00 00
    66  00 00 00 00 00
    67  00 00 00 00 00
    68  00 00 00 00 00
    69  00 00 00 00 00
    70  00 00 00 00 00
    71  00 00 00 00 00
    72  00 00 00 00 00
    73  00 00 00 00 00
    74  00 00 00 00 00
    75  00 00 00 00 00
    76  00 00 00 00 00
    77  00 00 00 00 00
    78  00 00 00 00 00
    79  00 00 00 00 00
    80  00 00 00 00 00
    81  00 00 00 00 00
    82  00 00 00 00 00
    83  00 00 00 00 00
    84  00 00 00 00 00
    85  00 00 00 00 00
    86  00 00 00 00 00
    87  00 00 00 00 00
    88  00 00 00 00 00
    89  00 00 00 00 00
    90  00 00 00 00 00
    91  00 00 00 00 00
    92  00 00 00 00 00
    93  00 00 00 00 00
    94  00 00 00 00 00
    95  00 00 00 00 00
    96  00 00 00 00 00
    97  00 00 00 00 00
    98  b7 b6 b6 b7 b7
    99  ff ff ff ff ff
```

syssi, 2024-02-14:
Long cell info frames (300 bytes):

```
     0  55 55 55 55 55 55 55 55 55 55
     1  aa aa aa aa aa aa aa aa aa aa
     2  11 11 11 11 11 11 11 11 11 11
     3  01 01 01 01 01 01 01 01 01 01
     4  02 02 02 02 02 02 02 02 02 02
     5  00 00 00 00 00 00 00 00 00 00
     6  2c 2c 2c 2c 2c 2c 2c 2c 2c 2c
     7  01 01 01 01 01 01 01 01 01 01
     8  ca cb cc cd d1 d2 d3 d4 d7 d8
     9  4e 4e 4e 4e 4e 14 65 65 2a 65
    10  78 78 78 78 78 77 73 73 72 73
    11  42 42 42 42 42 42 42 42 42 42
    12  40 40 40 40 40 40 40 40 40 40
    13  96 96 96 96 96 81 96 d1 0c 0c
    14  2a 2a 2a 2a 2a 2f 2a 2b 2d 2d
    15  46 46 46 46 46 46 46 46 46 46
    16  40 40 40 40 40 40 40 40 40 40
    17  47 47 47 47 47 47 0c 47 81 47
    18  2e 2e 2e 2e 2e 2e 2d 2e 2f 2e
    19  46 46 46 46 46 46 46 46 46 46
    20  40 40 40 40 40 40 40 40 40 40
    21  e7 e7 e7 e7 e7 21 96 5c 5c 21
    22  26 26 26 26 26 28 2a 29 29 28
    23  46 46 46 46 46 46 46 46 46 46
    24  40 40 40 40 40 40 40 40 40 40
    25  d1 d1 d1 d1 d1 47 47 0c 0c 47
    26  2b 2b 2b 2b 2b 2e 2e 2d 2d 2e
    27  46 46 46 46 46 46 46 46 46 46
    28  40 40 40 40 40 40 40 40 40 40
    29  71 71 71 71 71 e7 e7 71 e7 e7
    30  24 24 24 24 24 26 26 24 26 26
    31  46 46 46 46 46 46 46 46 46 46
    32  40 40 40 40 40 40 40 40 40 40
    33  d1 d1 d1 d1 d1 d1 96 96 0c d1
    34  2b 2b 2b 2b 2b 2b 2a 2a 2d 2b
    35  46 46 46 46 46 46 46 46 46 46
    36  40 40 40 40 40 40 40 40 40 40
    37  47 47 47 47 47 b2 62 9c 4d 87
    38  2e 2e 2e 2e 2e 15 19 1a 1e 1f
    39  46 46 46 46 46 46 46 46 46 46
    40  40 40 40 40 40 40 40 40 40 40
    41  21 21 21 21 21 5c d1 96 96 d1
    42  28 28 28 28 28 29 2b 2a 2a 2b
    43  46 46 46 46 46 46 46 46 46 46
    44  40 40 40 40 40 40 40 40 40 40
    45  96 96 96 96 96 96 d1 0c 0c d1
    46  2a 2a 2a 2a 2a 2a 2b 2d 2d 2b
    47  46 46 46 46 46 46 46 46 46 46
    48  40 40 40 40 40 40 40 40 40 40
    49  0c 0c 0c 0c 0c 47 81 47 bc 81
    50  2d 2d 2d 2d 2d 2e 2f 2e 30 2f
    51  46 46 46 46 46 46 46 46 46 46
    52  40 40 40 40 40 40 40 40 40 40
    53  e7 e7 e7 e7 e7 5c 5c 21 5c 96
    54  26 26 26 26 26 29 29 28 29 2a
    55  46 46 46 46 46 46 46 46 46 46
    56  40 40 40 40 40 40 40 40 40 40
    57  18 18 18 18 18 dd 18 18 dd 18
    58  41 41 41 41 41 3f 41 41 3f 41
    59  45 45 45 45 45 45 45 45 45 45
    60  40 40 40 40 40 40 40 40 40 40
    61  ac ac ac ac ac e7 e7 e7 e7 5c
    62  25 25 25 25 25 26 26 26 26 29
    63  46 46 46 46 46 46 46 46 46 46
    64  40 40 40 40 40 40 40 40 40 40
    65  0c 0c 0c 0c 0c 0c 0c 0c 47 d1
    66  2d 2d 2d 2d 2d 2d 2d 2d 2e 2b
[... 209 lines cut ...]
   276  00 00 00 00 00 00 00 00 00 00
   277  00 00 00 00 00 00 00 00 00 00
   278  00 00 00 00 00 00 00 00 00 00
   279  00 00 00 00 00 00 00 00 00 00
   280  00 00 00 00 00 00 00 00 00 00
   281  00 00 00 00 00 00 00 00 00 00
   282  00 00 00 00 00 00 00 00 00 00
   283  00 00 00 00 00 00 00 00 00 00
   284  00 00 00 00 00 00 00 00 00 00
   285  00 00 00 00 00 00 00 00 00 00
   286  00 00 00 00 00 00 00 00 00 00
   287  00 00 00 00 00 00 00 00 00 00
   288  00 00 00 00 00 00 00 00 00 00
   289  00 00 00 00 00 00 00 00 00 00
   290  00 00 00 00 00 00 00 00 00 00
   291  00 00 00 00 00 00 00 00 00 00
   292  00 00 00 00 00 00 00 00 00 00
   293  00 00 00 00 00 00 00 00 00 00
   294  00 00 00 00 00 00 00 00 00 00
   295  00 00 00 00 00 00 00 00 00 00
   296  00 00 00 00 00 00 00 00 00 00
   297  00 00 00 00 00 00 00 00 00 00
   298  a9 3a 9b c6 00 6d c2 4d 85 83
   299  ff ff ff ff ff ff ff ff ff ff
```

syssi, 2024-02-14:
Different commands:

```
>> aa:55:11:00:05:0d:14:00:00:00:00:00:00:00:00:00:00:00:36:ff
>> aa:55:11:00:05:0d:14:00:01:00:00:00:00:00:00:00:00:00:37:ff

>> aa:55:11:01:01:00:14:00:00:00:00:00:00:00:00:00:00:00:26:ff

>> aa:55:11:01:02:00:14:00:00:00:00:00:00:00:00:00:00:00:27:ff

>> aa:55:11:01:04:00:14:00:00:00:00:00:00:00:00:00:00:00:29:ff
```

Kippervink, 2024-02-14:
Glad the trace was OK. If I can help in any way to add support please let me know.

Kippervink, 2024-02-14:
[image]
In case its of any use, this is what the app looks like.

Kippervink, 2024-02-14:
[image]

syssi, 2024-02-14:
Could you provide the name, version and/or a download link of the app?

syssi, 2024-02-14:
The `heltec_balancer_ble` component is able to decode the captured traffic in general:

```
# device info frame
[21:39:07][I][heltec_balancer_ble:649]: Device info frame (100 bytes):
[21:39:07][D][heltec_balancer_ble:650]:   55.AA.11.01.01.00.64.00.45.4B.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.33.2E.32.2E.30.5A.48.2D.31.2E.32.2E.39.56.31.2E.32.2E.39.00.00.32.30.32.33.30.36.30.38.07.00.00.00.F1.17.85.00.33.43.41.35.35.31.39.35.37.44.39.30.31.32.33.34.35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7B.FF (100)
[21:39:07][I][heltec_balancer_ble:667]:   Model: EK-24S4EB
[21:39:07][I][heltec_balancer_ble:669]:   Hardware version: HW-3.2.0
[21:39:07][I][heltec_balancer_ble:671]:   Software version: ZH-1.2.9
[21:39:07][I][heltec_balancer_ble:673]:   Protocol version: V1.2.9
[21:39:07][I][heltec_balancer_ble:675]:   Manufacturing date: 20230608
[21:39:07][I][heltec_balancer_ble:677]:   Power on count: 7
[21:39:07][I][heltec_balancer_ble:679]:   Total runtime: 100d 22h (8722417s)

# cell info frame
[21:39:09][I][heltec_balancer_ble:295]: Cell info frame (300 bytes):
[21:39:09][D][heltec_balancer_ble:296]:   55.AA.11.01.02.00.2C.01.CC.4E.78.42.40.96.2A.46.40.47.2E.46.40.E7.26.46.40.D1.2B.46.40.71.24.46.40.D1.2B.46.40.47.2E.46.40.21.28.46.40.96.2A.46.40.0C.2D.46.40.E7.26.46.40.18.41.45.40.AC.25.46.40.0C.2D.46.40.FF.7B.42.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.6B.C3.26.3E.CC.93.28.3E.4E.68.29.3E.91.DA.27.3E.67.E2.27.3E.C9.12.28.3E.74.EC.28.3E.13.99.26.3E.D9.BC.26.3E.AC.F1.28.3E.6C.8A.28.3E.C8 (150)
[21:39:09][D][heltec_balancer_ble:297]:   A1.26.3E.B2.9E.27.3E.B9.36.25.3E.B5.3D.25.3E.0B.0D.26.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.8E.A5.45.42.8E.A5.45.40.40.7E.6D.3D.07.00.0F.05.2F.DF.82.C0.3E.0A.62.42.3E.0A.62.42.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.28.32.85.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.9B.FF (150)
[21:39:09][D][heltec_balancer_ble:324]:   Frame counter: 204
[21:39:09][D][sensor:094]: 'heltec-balancer total voltage': Sending state 49.41167 V with 2 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer average cell voltage': Sending state 3.08823 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer delta cell voltage': Sending state 0.05798 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer max voltage cell': Sending state 8.00000  with 0 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer min voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 1': Sending state 3.03859 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 1': Sending state 0.16285 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 2': Sending state 3.09635 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 2': Sending state 0.16463 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 3': Sending state 3.09657 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 3': Sending state 0.16544 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 4': Sending state 3.09612 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 4': Sending state 0.16392 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 5': Sending state 3.09642 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 5': Sending state 0.16395 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 6': Sending state 3.09597 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 6': Sending state 0.16413 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 7': Sending state 3.09642 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 7': Sending state 0.16496 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 8': Sending state 3.09657 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 8': Sending state 0.16269 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 9': Sending state 3.09620 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 9': Sending state 0.16283 Ω with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell voltage 10': Sending state 3.09635 V with 3 decimals of accuracy
[21:39:09][D][sensor:094]: 'heltec-balancer cell resistance 10': Sending state 0.16498 Ω with 3 decimals of accuracy
[21:39:10][D][sensor:094]: 'heltec-balancer cell voltage 11': Sending state 3.09650 V with 3 decimals of accuracy
[21:39:10][D][sensor:094]: 'heltec-balancer cell resistance 11': Sending state 0.16459 Ω with 3 decimals of accuracy
[21:39:10][D][sensor:094]: 'heltec-balancer cell voltage 12': Sending state 3.09612 V with 3 decimals of accuracy
[21:39:10][D][sensor:094]: 'heltec-balancer cell resistance 12': Sending state 0.16273 Ω with 3 decimals of accuracy
[... cut ...]
[21:39:10][D][sensor:094]: 'heltec-balancer cell undervoltage bitmask': Sending state 0.00000  with 0 decimals of accuracy
[21:39:10][D][sensor:094]: 'heltec-balancer cell polarity error bitmask': Sending state 0.00000  with 0 decimals of accuracy
[21:39:10][D][sensor:094]: 'heltec-balancer cell excessive line resistance bitmask': Sending state 0.00000  with 0 decimals of accuracy
[21:39:10][I][heltec_balancer_ble:466]:   Uptime: 101d  (8729128s)
```

I guess the bluetooth module has changed / behaves a bit different as in the past. I will have a look.

Kippervink, 2024-02-14:
The app is called neey-v1.2.14.apk, it can be found here: https://www.kolins.cz/share/?dir=neey
I do not know of an official download.

Kippervink, 2024-02-14:
I forgot the most important thing. Thank you for looking at this!

syssi, 2024-02-14:
Let's call it ham spirit! ;-) de DO2KSM

syssi, 2024-02-16:
```
Handle 0x000b: Write Request (Client Characteristic Configuration)
 -> Subscribe to notification

Handle 0x000a: Write Command (0x52)
  Value: aa551101010014000000000000000000000026ff
  Value: aa551101040014000000000000000000000029ff

Incoming notifications on handle 0x000a.
```

syssi, 2024-02-16:
I've prepared a feature branch with a small change. Please update your YAML configuration / retrieve the external component from the feature branch and give it a try:

```
substitutions:
  external_components_source: github://syssi/esphome-jk-bms@add-ek24s4eb-support
```

If it doesn't work out of the box please provide a ESPHome log.

kthemall, 2024-02-17:
hi, just tried the new branch version: no good infos, still not working, see:

[10:51:43][W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[10:51:43][C][mqtt:133]: MQTT:
[10:51:43][C][mqtt:135]:   Server Address: 192.168.188.101:1883 (192.168.188.101)
[10:51:43][C][mqtt:136]:   Username: 'wasser'
[10:51:43][C][mqtt:137]:   Client ID: 'neey1-a842e3ab734c'
[10:51:43][C][mqtt:139]:   Discovery prefix: 'homeassistant'
[10:51:43][C][mqtt:140]:   Discovery retain: YES
[10:51:43][C][mqtt:142]:   Topic Prefix: 'neey1'
[10:51:43][C][mqtt:144]:   Log Topic: 'neey1/debug'
[10:51:43][C][mqtt:147]:   Availability: 'neey1/status'
[10:51:43][C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'neey1 charging':
[10:51:43][C][mqtt.binary_sensor:023]:   State Topic: 'neey1/binary_sensor/neey1_charging/state'
[10:51:43][C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'neey1 discharging':
[10:51:43][C][mqtt.binary_sensor:023]:   State Topic: 'neey1/binary_sensor/neey1_discharging/state'
[10:51:43][C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'neey1 balancing':
[10:51:43][C][mqtt.binary_sensor:023]:   State Topic: 'neey1/binary_sensor/neey1_balancing/state'
[10:51:43][C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'neey1 online status':
[10:51:43][C][mqtt.binary_sensor:023]:   State Topic: 'neey1/binary_sensor/neey1_online_status/state'
[10:51:43][C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'neey1 heating':
[10:51:43][C][mqtt.binary_sensor:023]:   State Topic: 'neey1/binary_sensor/neey1_heating/state'
[10:51:43][C][mqtt.button:029]: MQTT Button 'neey1 retrieve settings':
[10:51:43][C][mqtt.button:030]:   State Topic: 'neey1/button/neey1_retrieve_settings/state'
[10:51:43][C][mqtt.button:030]:   Command Topic: 'neey1/button/neey1_retrieve_settings/command'
[10:51:43][C][mqtt.button:029]: MQTT Button 'neey1 retrieve device info':
[10:51:44][C][mqtt.button:030]:   State Topic: 'neey1/button/neey1_retrieve_device_info/state'
[10:51:44][C][mqtt.button:030]:   Command Topic: 'neey1/button/neey1_retrieve_device_info/command'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 balance trigger voltage':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_balance_trigger_voltage/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 cell count':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_cell_count/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 total battery capacity':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_total_battery_capacity/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 cell voltage overvoltage protection':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_cell_voltage_overvoltage_protection/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 cell voltage overvoltage recovery':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_cell_voltage_overvoltage_recovery/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 cell voltage undervoltage protection':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_cell_voltage_undervoltage_protection/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 cell voltage undervoltage recovery':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_cell_voltage_undervoltage_recovery/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 balance starting voltage':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_balance_starting_voltage/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 voltage calibration':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_voltage_calibration/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 current calibration':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_current_calibration/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 power off voltage':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_power_off_voltage/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 max balance current':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_max_balance_current/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 max charge current':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_max_charge_current/state'
[10:51:44][C][mqtt.number:033]: MQTT Number 'neey1 max discharge current':
[10:51:44][C][mqtt.number:034]:   State Topic: 'neey1/number/neey1_max_discharge_current/state'
[10:51:44][C][mqtt.sensor:028]: MQTT Sensor 'neey1 cell voltage 1':
[10:51:44][C][mqtt.sensor:032]:   State Topic: 'neey1/sensor/neey1_cell_voltage_1/state'
[10:51:44][C][mqtt.sensor:028]: MQTT Sensor 'neey1 cell voltage 2':
[10:51:44][C][mqtt.sensor:032]:   State Topic: 'neey1/sensor/neey1_cell_voltage_2/state'
[10:51:44][C][mqtt.sensor:028]: MQTT Sensor 'neey1 cell voltage 3':
[10:51:44][C][mqtt.sensor:032]:   State Topic: 'neey1/sensor/neey1_cell_voltage_3/state'
[10:51:44][C][mqtt.sensor:028]: MQTT Sensor 'neey1 cell voltage 4':
[10:51:44][C][mqtt.sensor:032]:   State Topic: 'neey1/sensor/neey1_cell_voltage_4/state'
[10:51:44][C][mqtt.sensor:028]: MQTT Sensor 'neey1 cell voltage 5':
[10:51:44][C][mqtt.sensor:032]:   State Topic: 'neey1/sensor/neey1_cell_voltage_5/state'
[10:51:44][C][mqtt.sensor:028]: MQTT Sensor 'neey1 cell voltage 6':
[10:51:44][C][mqtt.sensor:032]:   State Topic: 'neey1/sensor/neey1_cell_voltage_6/state'
[... cut ...]
[10:52:19][W][component:214]: Component jk_bms_ble took a long time for an operation (0.63 s).
[10:52:19][W][component:215]: Components should block for at most 20-30ms.
[10:52:23][I][jk_bms_ble:242]: Request status notification
[10:52:23][D][jk_bms_ble:1132]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[10:52:28][I][jk_bms_ble:242]: Request status notification
[10:52:28][D][jk_bms_ble:1132]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)

syssi, 2024-02-17:
Thanks for testing! I've seen what I missed. I will push another change and ping you again.

syssi, 2024-02-17:
Wait. You are using the wrong component (`jk_bms_ble`). Please use `heltec_balancer_ble`.

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-heltec-balancer-ble-example.yaml

kthemall, 2024-02-17:
new test with the heltec yaml file (sorry my fault), but still lot of nan

C:\Users\x>esphome run c:\neey\esphome-jk-bms-2.0.0\neey.yaml
INFO ESPHome 2023.12.9
INFO Reading configuration c:\neey\esphome-jk-bms-2.0.0\neey.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@add-ek24s4eb-support
INFO Generating C++ source...
INFO Compiling app...
Processing neey2 (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (5.4.0) > WEMOS D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40405.230623 (4.4.5)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.18.5
Building in release mode
Compiling .pioenvs\neey2\src\main.o
Linking .pioenvs\neey2\firmware.elf
Retrieving maximum program size .pioenvs\neey2\firmware.elf
Checking size .pioenvs\neey2\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.6% (used 54500 bytes from 327680 bytes)
Flash: [=======   ]  71.9% (used 1319737 bytes from 1835008 bytes)
Building .pioenvs\neey2\firmware.bin
esptool.py v4.4
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs\neey2\firmware.bin"], [".pioenvs\neey2\firmware.elf"])
esptool.py v4.6.2
Wrote 0x1539b0 bytes to file c:\neey\esphome-jk-bms-2.0.0\.esphome\build\neey2\.pioenvs\neey2/firmware-factory.bin, ready to flash to offset 0x0
============================================= [SUCCESS] Took 44.24 seconds =============================================
INFO Successfully compiled program.
INFO Resolving IP address of neey2.local
INFO  -> 192.168.188.140
INFO Uploading c:\neey\esphome-jk-bms-2.0.0\.esphome\build/neey2\.pioenvs\neey2\firmware.bin (1325488 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 9.34 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from neey2/debug
INFO Connected to MQTT broker!
[11:41:06][D][esp-idf:000]: I (141920) MQTT_CLIENT: Client asked to disconnect

[11:41:17][I][mqtt:274]: MQTT Connected!
[11:41:17][I][app:062]: setup() finished successfully!
[11:41:18][I][app:102]: ESPHome version 2023.12.9 compiled on Feb 17 2024, 11:40:16
[11:41:18][I][app:104]: Project syssi.esphome-jk-bms version 2.0.0
[11:41:18][C][wifi:573]: WiFi:
[11:41:18][C][wifi:405]:   Local MAC: A8:42:E3:AB:73:4C
[11:41:18][C][wifi:410]:   SSID: 'K'
[11:41:18][C][wifi:411]:   IP Address: 192.168.188.140
[11:41:18][C][wifi:413]:   BSSID: 5C:49:79:46:9F:09
[... 577 lines cut ...]
[11:41:56][I][heltec_balancer_ble:242]: Request status notification
[11:41:56][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:41:56][W][component:214]: Component heltec_balancer_ble took a long time for an operation (0.36 s).
[11:41:56][W][component:215]: Components should block for at most 20-30ms.
[11:42:01][I][heltec_balancer_ble:242]: Request status notification
[11:42:01][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:06][I][heltec_balancer_ble:242]: Request status notification
[11:42:06][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:11][I][heltec_balancer_ble:242]: Request status notification
[11:42:11][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:16][I][heltec_balancer_ble:242]: Request status notification
[11:42:16][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:19][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:21][I][heltec_balancer_ble:242]: Request status notification
[11:42:21][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:26][I][heltec_balancer_ble:242]: Request status notification
[11:42:26][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:31][I][heltec_balancer_ble:242]: Request status notification
[11:42:31][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:36][I][heltec_balancer_ble:242]: Request status notification
[11:42:42][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:42][I][heltec_balancer_ble:242]: Request status notification
[11:42:42][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[11:42:46][I][heltec_balancer_ble:242]: Request status notification
[11:42:46][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.0

syssi, 2024-02-17:
Differences of the commands / request payloads:

```
# device info
aa551101010014000000000000000000000026ff # btsnoop EK-24S4EB -> different checksum
aa5511010100140000000000000000000000faff # btsnoop GW-24S4EB
aa5511010100140000000000000000000000faff # heltec_balancer_ble implementation

# cell info
aa551101020014000000000000000000000027ff # btsnoop EK-24S4EB -> shifted date_len position fixed
aa5511010200001400000000000000000000f9ff # btsnoop GW-24S4EB
aa5511010200001400000000000000000000f9ff # heltec_balancer_ble implementation

# settings
aa551101040014000000000000000000000029ff # btsnoop EK-24S4EB -> different checksum
aa5511010400140000000000000000000000ffff # btsnoop GW-24S4EB
aa5511010400140000000000000000000000ffff # heltec_balancer_ble implementation
```

Kippervink, 2024-02-17:
I would love to help with testing but I am away this weekend. I will try it as soon as I get back!

syssi, 2024-02-17:
I've pushed another change. In best case it's working now.

kthemall, 2024-02-17:
oh yesssss it is. thx

kthemall, 2024-02-17:
that´s from console:

[12:23:53][I][heltec_balancer_ble:493]:   Uptime: 16d 23h (1465356s)
[12:23:53][W][component:214]: Component esp32_ble took a long time for an operation (0.69 s).
[12:23:53][W][component:215]: Components should block for at most 20-30ms.
[12:23:58][I][heltec_balancer_ble:322]: Cell info frame (300 bytes):
[12:23:58][D][heltec_balancer_ble:323]:   55.AA.11.01.02.00.2C.01.31.12.B1.51.40.85.C3.51.40.D1.CF.51.40.0C.D1.51.40.24.BC.51.40.CC.29.51.40.CF.19.51.40.D3.64.51.40.69.8D.51.40.AB.C9.51.40.B9.E4.51.40.55.ED.51.40.F0.F5.51.40.6A.E8.51.40.CD.DF.51.40.66.F8.51.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C5.33.48.3E.FC.09.4A.3E.1E.D5.51.3E.61.74.57.3E.9A.86.60.3E.DA.96.64.3E.36.75.66.3E.69.2C.69.3E.E3.75.60.3E.CD.13.62.3E.02.DF.61.3E.F5 (150)
[12:23:58][D][heltec_balancer_ble:324]:   FE.60.3E.F7.3F.61.3E.73.CB.50.3E.7C.13.47.3E.A6.5C.44.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.9B.B5.51.42.9B.B5.51.40.00.97.5E.3C.0F.06.0F.0A.00.00.00.00.33.33.8F.41.33.33.8F.41.00.00.FF.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11.5C.16.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AA.FF (150)
[12:23:58][D][heltec_balancer_ble:351]:   Frame counter: 49
[12:23:58][D][sensor:094]: 'neey2 total voltage': Sending state 52.42735 V with 2 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 average cell voltage': Sending state 3.27671 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 delta cell voltage': Sending state 0.01359 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 max voltage cell': Sending state 16.00000  with 0 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 min voltage cell': Sending state 7.00000  with 0 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 1': Sending state 3.27643 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 1': Sending state 0.19551 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 2': Sending state 3.27756 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 2': Sending state 0.19730 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 3': Sending state 3.27831 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 3': Sending state 0.20491 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 4': Sending state 3.27838 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 4': Sending state 0.21040 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 5': Sending state 3.27711 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 5': Sending state 0.21926 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 6': Sending state 3.26818 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 6': Sending state 0.22323 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 7': Sending state 3.26720 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 7': Sending state 0.22506 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 8': Sending state 3.27178 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 8': Sending state 0.22771 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 9': Sending state 3.27426 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 9': Sending state 0.21920 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 10': Sending state 3.27793 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 10': Sending state 0.22078 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 11': Sending state 3.27959 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 11': Sending state 0.22058 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 12': Sending state 3.28011 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 12': Sending state 0.21972 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 13': Sending state 3.28064 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 13': Sending state 0.21997 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 14': Sending state 3.27981 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 14': Sending state 0.20390 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 15': Sending state 3.27928 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 15': Sending state 0.19441 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 16': Sending state 3.28079 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 16': Sending state 0.19176 Ω with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell voltage 17': Sending state 0.00000 V with 3 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell resistance 17': Sending state 0.00000 Ω with 3 decimals of accuracy
[... cut ...]
[12:23:58][D][sensor:094]: 'neey2 cell undervoltage bitmask': Sending state 0.00000  with 0 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell polarity error bitmask': Sending state 0.00000  with 0 decimals of accuracy
[12:23:58][D][sensor:094]: 'neey2 cell excessive line resistance bitmask': Sending state 0.00000  with 0 decimals of accuracy
[12:23:58][I][heltec_balancer_ble:493]:   Uptime: 16d 23h (1465361s)
[12:23:58][W][component:214]: Component esp32_ble took a long time for an operation (0.69 s).
[12:23:58][W][component:215]: Components should block for at most 20-30ms.

syssi, 2024-02-17:
@kthemall Do you own the old Neey balancer version too? If not we have to wait for @Kippervink. I would like to verify the new behavior is compatible with the old model too.

kthemall, 2024-02-17:
no just the new one sorry

Kippervink, 2024-02-18:
Hi @syssi It is working perfectly with both the new and old versions of the balancer. Thank you so much for getting this going, it makes balancing much much easier.

syssi, 2024-02-18:
@Kippervink Could you provide the `device info` frame (log messages) or both balancer versions? I would like to make sure the old balancer is using protocol version 1.0.0.

Kippervink, 2024-02-18:
Apologies, this must be what you need:

[19:24:06][I][heltec_balancer_ble:676]: Device info frame (100 bytes):
[19:24:06][D][heltec_balancer_ble:677]:   55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.39.31.36.05.00.00.00.90.3F.13.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FA.FF (100)
[19:24:06][I][heltec_balancer_ble:694]:   Model: GW-24S4EB
[19:24:06][I][heltec_balancer_ble:696]:   Hardware version: HW-2.8.0
[19:24:06][I][heltec_balancer_ble:698]:   Software version: ZH-1.2.3
[19:24:06][I][heltec_balancer_ble:700]:   Protocol version: V1.0.0
[19:24:06][I][heltec_balancer_ble:702]:   Manufacturing date: 20220916
[19:24:06][I][heltec_balancer_ble:704]:   Power on count: 5
[19:24:06][I][heltec_balancer_ble:706]:   Total runtime: 1y 37d 23h (34815888s)

syssi, 2024-02-18:
And you are successfully using the new feature branch for both hardware versions? So it's in your opinion safe to merge the new device support with the changes checksum calculation? :-)

Kippervink, 2024-02-18:
Something seems to go wrong after a few minutes.. I'm not sure if its me or hardware or something else.. 

[19:27:32][D][sensor:094]: 'heltec-balancer3 cell voltage 9': Sending state 3.24262 V with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell resistance 9': Sending state 0.16283 Ω with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell voltage 10': Sending state 3.24269 V with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell resistance 10': Sending state 0.16498 Ω with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell voltage 11': Sending state 3.24269 V with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell resistance 11': Sending state 0.16459 Ω with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell voltage 12': Sending state 3.24322 V with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell resistance 12': Sending state 0.16273 Ω with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell voltage 13': Sending state 3.24104 V with 3 decimals of accuracy
[19:27:32][D][sensor:094]: 'heltec-balancer3 cell resistance 13': Sending state 0.16369 Ω with 3 decimals of accuracy
WARNING heltec-balancer: Connection error occurred: [WinError 64] The specified network name is no longer available
INFO Processing unexpected disconnect from ESPHome API for heltec-balancer
WARNING Disconnected from API
INFO Successfully connected to heltec-balancer in 0.006s
INFO Successful handshake with heltec-balancer in 0.085s
[19:28:13][I][heltec_balancer_ble:242]: Request status notification
[19:28:13][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.14.00.00.00.00.00.00.00.00.00.00.00.27.FF (20)
[19:28:13][I][heltec_balancer_ble:242]: Request status notification
[19:28:13][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.14.00.00.00.00.00.00.00.00.00.00.00.27.FF (20)
[19:28:14][I][heltec_balancer_ble:242]: Request status notification
[19:28:14][D][heltec_balancer_ble:759]: Write register: AA.55.11.01.02.00.14.00.00.00.00.00.00.00.00.00.00.00.27.FF (20)
[19:28:19][D][binary_sensor:036]: 'heltec-balancer online status': Sending state OFF
[19:28:19][D][text_sensor:064]: 'heltec-balancer operation status': Sending state 'Offline'
[19:28:19][D][sensor:094]: 'heltec-balancer min cell voltage': Sending state nan V with 3 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer max cell voltage': Sending state nan V with 3 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer min voltage cell': Sending state nan  with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer max voltage cell': Sending state nan  with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer delta cell voltage': Sending state nan V with 3 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer average cell voltage': Sending state nan V with 3 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer total voltage': Sending state nan V with 2 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer temperature sensor 1': Sending state nan °C with 2 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer temperature sensor 2': Sending state nan °C with 2 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer total runtime': Sending state nan s with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer balancing current': Sending state nan A with 2 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer cell detection failed bitmask': Sending state nan  with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer cell overvoltage bitmask': Sending state nan  with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer cell undervoltage bitmask': Sending state nan  with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer cell polarity error bitmask': Sending state nan  with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer cell excessive line resistance bitmask': Sending state nan  with 0 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balancer cell voltage 1': Sending state nan V with 3 decimals of accuracy
[19:28:19][D][sensor:094]: 'heltec-balan

It doesnt seem to recover until I power cycle the device.

Kippervink, 2024-02-18:
I will keep testing, I have another esp32 somewhere I can try.

Kippervink, 2024-02-19:
I have switched to a new board and it has been completely stable for the last 10 hours, connecting to both new and old balancers at the same time. I will continue to test and report back if I find any issues.

closed 2024-02-19 by syssi

ThibautBascoulergue, 2024-03-26:
Hello, I am very interested in this position because I own a Neey EK-24S4EB. Having an esp8266, which bluetooth module should I buy for everything to work? HC-05 or HC-06
THANKS

Kippervink, 2024-03-26:
> Hello, I am very interested in this position because I own a Neey EK-24S4EB. Having an esp8266, which bluetooth module should I buy for everything to work? HC-05 or HC-06 THANKS

I am using a firebeetle 2 which is a really nice board, but I think any esp32 board will work. The esp32 has built in bluetooth so no need for additional modules.

wave-man-wave, 2026-03-06:
I am reaching out to you regarding the ENERKEY 10A 2-24S Active Balancer (Model: EK-24SEB). I am currently integrating this device into my own monitoring system and I need to establish communication via the TTL port.

Could you please provide the technical documentation for the communication protocol? Specifically, I need the details regarding:

1. Serial Port Configuration (Baud rate, Parity, etc.).
2. Command Packet Structure (The exact byte sequences to request data).
3. Data Map/Registers (How to interpret the response packets to get cell voltages and status).

If there is a protocol manual specifically for the EK-24SEB model, please send it to me.

Thank you in advance for your assistance. I look forward to your reply.

commit 249bc479 (2024-02-19): Add Neey/EnerKey EK-24S4EB balancer support (Closes: #444) (#452)