syssi/esphome-jk-bms #449: Total Voltage showing the wronge value

RFDarter, 2024-02-13:
``Model: B2A20S20P``
``Hardware Version: V11 H.X``
``Software Version: V11.288H``
``Connection Mode: BLE``

Total Voltage stays as 0 but the app shows the correct calue.

Here is a already structured bt snoop generated by the script
[tmp_output.log](https://github.com/syssi/esphome-jk-bms/files/14266763/tmp_output.log)
total voltage changing between 1.48V and 2.96V a couple of times

RFDarter, 2024-02-13:
```c++
  // 118   4   0x03 0xD0 0x00 0x00    Battery voltage       0.001        V
  float total_voltage = (float) jk_get_32bit(118 + offset) * 0.001f;
```

``151`` seems to be correct one.

I only have a new `B2A20S20P` and `B1A8S10P` both Hardware Version: V11 H.X

Did `118` work as total voltage on other devices?

RFDarter, 2024-02-15:
Offset 32 would fit it, so maybe my devide is 32s after all? I going to look into that tomorrow. If that is the case my frame positions for heating would need to be adjusted to take the offset into account

syssi, 2024-02-18:
I will double check the total voltage position on a new and old BMS today but I'm pretty sure it's exposed properly if the correct protocol version is selected.

RFDarter, 2024-02-19:
> I will double check the total voltage position on a new and old BMS today but I'm pretty sure it's exposed properly if the correct protocol version is selected.

yep, my bad. It turnes out my bms is a 32S and not a 24 as i thought.
here is a fix for heating
https://github.com/syssi/esphome-jk-bms/pull/457

closed 2024-05-07 by syssi