syssi/esphome-jk-bms #453: Question: Board settings for wemos s3 mini

DutchKillerbee, 2024-02-17:
Hello 

I tried to complie the esp32-heltec-balancer-ble-example.yaml file and modified it a bit for the use of my wemos S3-mini

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    version: recommended
  variant: esp32s3

I am able to compile and load the firmware (no errors). But Nothing happens.
When I reset the device I hear Windows USB connection sounds, over and over again. 
So I assume the device is in a bootloop.

I have to tell I am new to this so could someone help me out?

syssi, 2024-02-17:
May be your USB port isn't able to provide enough power to establish the WiFi connection. Are you able to ping the ESP if you power the device using a USB power plug?

DutchKillerbee, 2024-02-17:
Thanks for your quick reply

FYI
I am able to flash the standard esphome firmware using https://web.esphome.io/ using the same USB cable.
Configure WIFI and the ESP is found by home assistant (using the standard esphome firwmware)

When I power the esp32 using s powerplug (and with my firmware) I cannot find any IP addres when I scan my network.
So I assume the firmware is not correct of it doesn't start at all.

I Will try another USB cable and check again

Do you have another suggestion?

syssi, 2024-02-17:
If there is enough power available you should see log messages / the boot loop on the serial line. One thing is uncommon nevertheless: The USB-to-TTL adapter shouldn't restart periodically.

syssi, 2024-02-17:
In other words: The USB-to-TTL module on your development board runs independent of the ESP. If the ESP crashes the USB interface shouldn't disappear. IMO this happens on low power situations only.

DutchKillerbee, 2024-02-17:
I moved to my linux machine and tried these settings:

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 4MB
  variant: esp32s3
  framework:
    type: esp-idf
    version: recommended

I was able to compile and flash
the terminal spams these messages:

[22:19:05]ESP-ROM:esp32s3-20210327
[22:19:05]Build:Mar 27 2021
[22:19:05]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[22:19:05]Saved PC:0x400454e9
[22:19:05]SPIWP:0xee
[22:19:05]mode:QIO, clock div:1
[22:19:05]load:0x3fce3808,len:0x162c
[22:19:05]ets_loader.c 79 
[22:19:06]ESP-ROM:esp32s3-20210327
[22:19:06]Build:Mar 27 2021
[22:19:06]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[22:19:06]Saved PC:0x400454e9
[22:19:06]SPIWP:0xee
[22:19:06]mode:QIO, clock div:1
[22:19:06]load:0x3fce3808,len:0x162c
[22:19:06]ets_loader.c 79 
[22:19:08]ESP-ROM:esp32s3-20210327
[22:19:08]Build:Mar 27 2021
[22:19:08]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[22:19:08]Saved PC:0x400454e9
[22:19:08]SPIWP:0xee
[22:19:08]mode:QIO, clock div:1
[22:19:08]load:0x3fce3808,len:0x162c
[22:19:08]ets_loader.c 79

syssi, 2024-02-18:
It looks like the ESP doesn't start the application at all. Could you try to reduce the YAML step by step? Are you sure a basic YAML starts properly (if ESP-IDF is used)?

DutchKillerbee, 2024-02-18:
Until now I was only able to flash the standard esphome firmware using https://web.esphome.io/ 
There is not much documentation/examples found on the net.
I found this topic [https://community.home-assistant.io/t/wemos-lolin-esp32-s3-mini-v1-0-0-s3fh4r2/586820/12](url) and try to flash my device using these minimal settings:

substitutions:
  friendly_name: 'Lolin'
  device_name: 'lolin'
  project_base: 'Wemos'
  project_name: 'ESP32'
  project_version: 'S3 mini'
  device_description: 'Lolin Wemos ESP32-S3 mini'

esphome:
  name: lolin-s3-mini
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git
    board: lolin_s3_mini
    framework: espidf
    board_build.flash_mode: dio

esp32:
  variant: esp32s3
  board: lolin_s3_mini
  framework:
    type: esp-idf
    version: latest

logger:
  level: DEBUG

compiling and flashings seems to be fine.
These are the last few lines in my terminal

....
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.1 seconds (effective 467.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00009000 in 0.1 seconds (effective 702.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200

(nothing happens)

As I said I am new to ESPHome and ESP32 devices. 
It seems I bought the wrong device(s) for testing your promissing heltec balancer software.....

DutchKillerbee, 2024-02-18:
UPDATE:
Oke as I said I am new.
To be able to flash the device needs to be in a flash mode. It seems that if the device is not in flash mode there are no errors and flashing seems to work.

To put the device in flash mode you need to press the buttons in this sequence (not verified yet):
press boot
press reset
release boot
release restet

I was able to flash now with wifi and home assistant support.
To be continued....

DutchKillerbee, 2024-02-18:
Sorry for spamming this topic but I was able to compile/flash the WEMOS S3-MINI using this minimal configuration:

substitutions:
  name: heltec-balancer
  device_description: "Monitor and control a Heltec 4A balancer via bluetooth"
  device_name: 'lolin'
  project_base: 'Wemos'
  project_name: 'ESP32'
  project_version: 'S3 mini'
  mac_address: 3C:A5:51:95:FD:5E
  external_components_source: github://syssi/esphome-jk-bms@main

esphome:
  name: ${name}
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git
    board: lolin_s3_mini
    framework: espidf
    board_build.flash_mode: dio

esp32:
  variant: esp32s3
  board: lolin_s3_mini
  flash_size: 4MB
  framework:
    type: esp-idf
    version: latest
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y    

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - logger.log: "BLE connection suspended for OTA update"

logger:
  level: DEBUG

api:

ble_client:
  - mac_address: ${mac_address}
    id: client0

switch:
  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${name} enable bluetooth connection"

I could see the device in home assistant and I could see the switch as well
I could see BT activity as soon as I am close to BLE devices.

But as soon as I enable this part in my yaml file:

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0

I get this compiler errors:
[... 70 lines cut ...]
      |                            ^~~~~~~~~~
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:679:3: note: in expansion of macro 'ESP_LOGI'
  679 |   ESP_LOGI(TAG, "  Total runtime: %s (%ds)", format_total_runtime_(heltec_get_32bit(60)).c_str(), heltec_get_32bit(60));
      |   ^~~~~~~~
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:679:40: note: format string is defined here
  679 |   ESP_LOGI(TAG, "  Total runtime: %s (%ds)", format_total_runtime_(heltec_get_32bit(60)).c_str(), heltec_get_32bit(60));
      |                                       ~^
      |                                        |
      |                                        int
      |                                       %ld
Compiling .pioenvs/heltec-balancer/src/esphome/components/mdns/mdns_component.o
Compiling .pioenvs/heltec-balancer/src/esphome/components/mdns/mdns_esp32.o
Compiling .pioenvs/heltec-balancer/src/esphome/components/mdns/mdns_esp8266.o
Compiling .pioenvs/heltec-balancer/src/esphome/components/mdns/mdns_host.o
Compiling .pioenvs/heltec-balancer/src/esphome/components/mdns/mdns_libretiny.o
Compiling .pioenvs/heltec-balancer/src/esphome/components/mdns/mdns_rp2040.o
Compiling .pioenvs/heltec-balancer/src/esphome/components/network/util.o
Compiling .pioenvs/heltec-balancer/src/esphome/components/number/automation.o
cc1plus: some warnings being treated as errors
*** [.pioenvs/heltec-balancer/src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.o] Error 1
========================================================== [FAILED] Took 38.73 seconds ==========================================================`

Do you have a suggestion what could cause this?

PS is there a better way to show yaml content in a comment in github?

DutchKillerbee, 2024-02-18:
YESS!
[image]

syssi, 2024-02-18:
Good job! Feel free to ask more questions if something isn't obvious.

DutchKillerbee, 2024-02-18:
Almost there...

When I activate the number or button section in the yaml file

number:
  - platform: heltec_balancer_ble
    cell_count:
      name: "${name} cell count"
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    max_balance_current:
      name: "${name} max balance current"
    balance_sleep_voltage:
      name: "${name} balance sleep voltage"
    balance_start_voltage:
      name: "${name} balance start voltage"
    nominal_battery_capacity:
      name: "${name} nominal battery capacity"

I get these errors when linking

.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/heltec-balancer/src/main.o:(.literal._ZN7esphome19heltec_balancer_ble12HeltecNumberC5Ev[_ZN7esphome19heltec_balancer_ble12HeltecNumberC5Ev]+0x4): undefined reference to `_ZTVN7esphome19heltec_balancer_ble12HeltecNumberE'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/heltec-balancer/src/main.o:(.literal._ZN7esphome19heltec_balancer_ble12HeltecNumberC5Ev[_ZN7esphome19heltec_balancer_ble12HeltecNumberC5Ev]+0x8): undefined reference to `_ZTVN7esphome19heltec_balancer_ble12HeltecNumberE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/heltec-balancer/firmware.elf] Error 1
========================================================== [FAILED] Took 18.70 seconds ==========================================================

any suggestions how to fix this?

syssi, 2024-02-18:
You have to clean the build folder: 

```
esphome clean yourconfig.yaml
```

If you use the ESPHome dashboard there should be a `clean` button somewhere at the tile.

DutchKillerbee, 2024-02-18:
Thanks that did the trick!

DutchKillerbee, 2024-02-23:
Hello @syssi 

I am running my S3 device now for more than 3 days with 2 Heltec (Master/Slave) active balancers connected.
I runs very stable and I have some minor addional question
The next two text sensors are not frequently updated in HA

[image]

The log shows other values
[image]

The webserver also shows the "old values"
[image]
[image]

Its not a big deal but I'd like to understand why not.
Is there any configuration option to update these sensors, lets say every day?
Or run an automation to force an update of all sensors?

syssi, 2024-02-23:
> Is there any configuration option to update these sensors, lets say every day?
> Or run an automation to force an update of all sensors?

Interesting. This isn't intended. I will try to get an idea what's going on here.

syssi, 2024-02-23:
This is the background:

https://github.com/syssi/esphome-jk-bms/blob/main/components/heltec_balancer_ble/heltec_balancer_ble.cpp#L481-L482

A `cell info frame` is retrieved periodically and contains a increasing timer. I've labeled the value as `Uptime` but it's probably the `total runtime` too.

The `total runtime` and `total runtime formatted` sensor is upated per `device info frame`:

https://github.com/syssi/esphome-jk-bms/blob/main/components/heltec_balancer_ble/heltec_balancer_ble.cpp#L481-L482

The `device info` frame isn't retrieved periodically. It's published/requested once if the BLE connection gets established. In other words: The `total runtime` updates very infrequent. 

Could you restart your ESP (the trigger a re-connect of the BLE connection / a new device info frame) and make a comparison between these two log messages:

```
ESP_LOGI(TAG, "  Total runtime: %s (%ds)", format_total_runtime_(heltec_get_32bit(60)).c_str(), heltec_get_32bit(60));
# vs.
ESP_LOGI(TAG, "  Uptime: %s (%ds)", format_total_runtime_(heltec_get_32bit(254)).c_str(), heltec_get_32bit(254));
```

If the values are equal we could update the `total runtime` once per cell info frame. This would match with your expectation.

DutchKillerbee, 2024-02-23:
If I understand you correct, before I reboot the device, if I hit the retreive device info button total runtime is shown as well right?
When I hit the retreive device info button (for the master device) the results of **total runtime** and **uptime** are not the same:
[image]
(both are from the master balancer)

DutchKillerbee, 2024-02-23:
I was wondering where the 16 hour difference came from in my last screenshot......
Just a guess:
- Uptime = the total runtime of the balancer
- Total runtime = runtime since the balancer was started

syssi, 2024-02-23:
Could you restart the balancer to proof your guess?

DutchKillerbee, 2024-02-23:
Unfortunately this is difficult since the balancer is build in a battery and there is no possibility to power off this balancer.
A stupid design by GobelPower.
To restart it I should open the battery box and disconnect a balancer cable, which is not easy to access.
I will ckeck if I can restart the balancer by app

DutchKillerbee, 2024-02-23:
Funny after a restart of the S3 both Uptime and Total runtime are the same.....

DutchKillerbee, 2024-02-24:
@syssi 

As test I have added these two line to the decode cell info part of your code:

ESP_LOGI(TAG, "  Uptime: %s (%ds)", format_total_runtime_(heltec_get_32bit(254)).c_str(), heltec_get_32bit(254));
**this->publish_state_(this->total_runtime_sensor_, (float) heltec_get_32bit(254));
this->publish_state_(this->total_runtime_formatted_text_sensor_, format_total_runtime_(heltec_get_32bit(254)));**

I expected that the runtime and runtime_formatted sensors were updated every 3 to 5 seconds in HA.
But again this is not the case. Not in the webserver nor in HA.....

Am I missing something here?

syssi, 2024-07-24:
> I expected that the total_runtime and total_runtime_formatted sensors were updated every 3 to 5 seconds in HA.
> But again this is not the case. Not in the webserver nor in HA.....
> 
> Am I missing something here?

This is a behaviour of the BMS firmware which updates the value only a few time a day if I remember correctly.