syssi/esphome-jk-bms #454: problems with the latest firmware of standard jk

Thedoctor68, 2024-02-18:
I'm having problems with one of the latest standard versions of the jk bms, it doesn't detect several values, the cell delta is wrong, the SOC is not displayed and several other errors, on the contrary the voltage of my 16 cells is displayed correctly
[image]

Thedoctor68, 2024-02-18:
[image]

I have configured this protocol

Thedoctor68, 2024-02-18:
i have other 2 older jk-bms and all perfect, any solution?

RFDarter, 2024-02-19:
Have you tried `protocol_version: JK02_24S` ?

Thedoctor68, 2024-02-24:
> Have you tried `protocol_version: JK02_24S` ?

i'll try thanks

syssi, 2024-07-08:
Did you resolve the issue? The wrong `protocol_version` is the root cause for sure.

closed 2024-07-08 by syssi