syssi/esphome-jk-bms #458: JK-B2A20S20P, hw 11.XW, sw 11.284

martinSezman, 2024-02-19:
I'd like to raport that JK-B2A20S20P BMS works fine using JK02_32S, running on an ESP32 C3 board, connected via both BLE and UART.

BMS details:
JK-B2A20S20P
HW version: V11.XW
SW version: V11.284

Regards,
martinSezman

betmensk, 2024-02-19:
hi, can you please post your code?
I try to connect to this bms via esp32 wroom ble, via wemos d1 rs485 (original converter from jikong) and everything fails.
bms sends data 1x and then communication is lost.
well thank you.

RFDarter, 2024-02-19:
Hi, is it `JK-B2A20S20P` or `JK-B2A20S20P-HC` ?

martinSezman, 2024-02-20:
My code:

substitutions:
  name: "jkbms"
  device_description: "Monitor a JK-BMS via BLE/UART"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO21
  rx_pin: GPIO20
  mac_address: xx:xx:xx:xx:xx:xx
  protocol_version: JK02_32S
  
esphome:
  name: ${name}
  friendly_name: JKBMS
  area: Garaż
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.0.0

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    esp32_ble_tracker: INFO
    jk_bms_ble: INFO
    scheduler: DEBUG
    component: DEBUG
    sensor: INFO
    api.service: INFO
    api: INFO
    number: INFO

debug:
  update_interval: 10s

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxx"
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - logger.log: "BLE connection suspended for OTA update"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 8.5dB

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jkbms Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxx"

captive_portal:

web_server:
  port: 80

status_led:
[... 251 lines cut ...]
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

martinSezman, 2024-02-20:
> Hi, is it `JK-B2A20S20P` or `JK-B2A20S20P-HC` ?
JK-B2A20S20P

betmensk, 2024-02-21:
martin, thank you, it didn't help me, because esp32dev has a lack of memory.

denveronly, 2024-03-01:
> My code:
>
> ```
> substitutions:
>   name: "jkbms"
>   device_description: "Monitor a JK-BMS via BLE/UART"
>   external_components_source: github://syssi/esphome-jk-bms@main
>   tx_pin: GPIO21
>   rx_pin: [...]

Hello,
You are connecting it to UART port of JK (BMS expansion board the two right RJ45 ports, one of them) correct?
Is it possible to receive data about Multiple BMS from a single port the same way as JK Windows Software does Using Device adress list?

Here is a sketch of what i am trying to achieve

[image]

martinSezman, 2024-03-01:
> > My code:
> > ```
> > 
> >       name: "${name} errors bitmask"
> > ```
> 
> Hello, You are connecting it to UART port of JK (BMS expansion board the two right RJ45 ports, one of them) correct? Is it possible to receive data about Multiple BMS from a single port the same way as JK Windows Software does Using Device adress list?
> 
> Here is a sketch of what i am trying to achieve
> 
> ![image](https://private-user-images.githubusercontent.com/66441516/309293625-a197c8f8-b5c0-47ca-9c9f-6206001d344a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkzMDU2ODcsIm5iZiI6MTcwOTMwNTM4NywicGF0aCI6Ii82NjQ0MTUxNi8zMDkyOTM2MjUtYTE5N2M4ZjgtYjVjMC00N2NhLTljOWYtNjIwNjAwMWQzNDRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzAxVDE1MDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4MGZiOWVkODQxZTc5NTFjMzA4ZGVhMjg3Mjk2MWE0ODQyMGM5YWM2ZjFjNzg4N2NlZDUyYTZjNTM0YWMyOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.enab6PLSBhgm4D-K4xRBryiW0uRcvDMU0mp6hd8ZFTk)

This topic is for JK-B2A20S20P BMS which is not stackable I suspect.

denveronly, 2024-03-01:
> This topic is for JK-B2A20S20P BMS which is not stackable I suspect.

Oh, sorry missed that
I have PB2A
Sorry

commit 3004d361 (2024-07-24): Add JK-B2A20S20P as supported (Closes: #458)
closed 2024-07-24 by syssi