syssi/esphome-jk-bms #459: Connecting Neey Active Balancer to NodeMCU 32-S via BLE

trippfam07, 2024-02-20:
I am trying to Program a NodeMCU32S to read the Neey Active Balancer and not having any luck.
Could someone point me in the right direction?

I am using this code....

substitutions:
  name: heltec-balancer
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E1:E2:FF

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }
ble_client:
  - mac_address: ${mac_address}
    id: client0

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: heltec_balancer_ble
    balancing:
      name: "${name} balancing"
    online_status:
      name: "${name} online status"

button:
  - platform: heltec_balancer_ble
    retrieve_settings:
      name: "${name} retrieve settings"
      id: retrieve_settings_button
    retrieve_device_info:
      name: "${name} retrieve device info"
    retrieve_factory_defaults:
      name: "${name} retrieve factory defaults"

And I get these error logs....

Processing heltec-balancer (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.o
Compiling /data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/number/automation.o
Compiling /data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/number/number.o
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp: In member function 'virtual void esphome::heltec_balancer_ble::HeltecBalancerBle::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:194:71: error: 'class esphome::ble_client::BLEClient' has no member named 'get_gattc_if'; did you mean 'gattc_if'?
       auto status = esp_ble_gattc_register_for_notify(this->parent()->get_gattc_if(), this->parent()->get_remote_bda(),
                                                                       ^~~~~~~~~~~~
                                                                       gattc_if
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:194:103: error: 'class esphome::ble_client::BLEClient' has no member named 'get_remote_bda'; did you mean 'remote_bda'?
       auto status = esp_ble_gattc_register_for_notify(this->parent()->get_gattc_if(), this->parent()->get_remote_bda(),
                                                                                                       ^~~~~~~~~~~~~~
                                                                                                       remote_bda
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp: In member function 'bool esphome::heltec_balancer_ble::HeltecBalancerBle::send_command(uint8_t, uint8_t, uint8_t, uint32_t)':
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:768:47: error: 'class esphome::ble_client::BLEClient' has no member named 'get_gattc_if'; did you mean 'gattc_if'?
       esp_ble_gattc_write_char(this->parent_->get_gattc_if(), this->parent_->get_conn_id(), this->char_handle_,
                                               ^~~~~~~~~~~~
                                               gattc_if
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:768:78: error: 'class esphome::ble_client::BLEClient' has no member named 'get_conn_id'; did you mean 'conn_id'?
       esp_ble_gattc_write_char(this->parent_->get_gattc_if(), this->parent_->get_conn_id(), this->char_handle_,
                                                                              ^~~~~~~~~~~
                                                                              conn_id
*** [/data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.o] Error 1
========================== [FAILED] Took 8.38 seconds ==========================

syssi, 2024-02-20:
Please make sure you are using the most recent ESPHome version.

trippfam07, 2024-02-20:
I am using ESPHome on Homeassistant....I keep it up to date when ever there is update available. Is there a manual way to update?

trippfam07, 2024-02-27:
Figured it out...Thanks

closed 2024-02-27 by trippfam07

trippfam07, 2025-07-02:
> [@trippfam07](https://github.com/trippfam07) Would you be so kind to also share _**HOW**_ you "figured it out", to help those people running into same/similar problems?

That was a while ago, but I think I had to uninstall ESP Home and reinstall.

syssi, 2025-07-03:
@jangrewe if it doesn't work for you out of the box feel free to create an issue.