syssi/esphome-jk-bms #463: Working on JK-PB2A16S-20P

txubelaxu, 2024-02-24:
Model JK-PB2A16S-20P
Hardware V =14.XA
Software V=14.20
ESP32

2 units working, gathering data using bluetooth and publishing into Home Assistant.

txubelaxu, 2024-02-24:
Sniffing the RS485 internal network, this is the traffic:
Two types of response messages from each JK-BMS in the network:
55 aa eb 90 **02** 00... <<<-- similar to BLE info
55 aa eb 90 **01** 00... <<<-- CONFIGURATION information the jk-bms. I have detected many fields related to this type of message.
This traffic is in the network because 1 of the BMSs is acting as MASTER. This is used by MASTER to using the whole system information to communicate via CAN with Victron. The rest of BMSs in the network are SLAVEs. This is a RS485 network, so it cannot be another MASTER asking. The only way is "to listen".

```0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
0e10162000010200009701
0f10162000010200009a91
55aaeb900200130d120d110d120d120d110d110d110d110d110d110d110d110d110d110d100d0000000000000000000000000000000000000000000000000000000000000000ffff0000110d0300000f530050004f004a004d004b004d004d0053004e004d004a004c004d00520051000000000000000000000000000000000000000000000000000000000000000000e0000000000015d100000000000000000000c800ce0000000000000000623d8f040080a3040000000000201c00006400000036aa460001010000000000000000000000000000ff00010000009a030000000060543f4000000000e814000000010101000600003dbc7e0100000000e000c800d0009a03c04626007f0000008051010000000000000000000000000000feff7fdd2f0101b0070000000d001016200001059a
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456
011016200001020000d6f1
021016200001020000c201
031016200001020000cf91
041016200001020000e9a1
051016200001020000e431
061016200001020000f0c1
071016200001020000fd51
55aaeb900205140d120d120d120d120d120d120d110d120d110d120d120d120d110d120d100d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d0500000f4d004b004d004b004d004a004c0049004c004b004e004c0050004e0050004e000000000000000000000000000000000000000000000000000000000000000000d0000000000023d100000000000000000000c000c80000000000000000623e8f040080a30400000000004b140000640000003726290001010000000000000000000000000000ff00010000009303000000000bc23f4000000000e9140000000101010006000013b37e0100000000d000c100c2008d03d1452600340000008051010000000000000000000000000000feff7fdd2f0101b007000000ec
071016200001042d
0710161e0001020000f98f
55aaeb900105ac0d0000280a00005a0a00001a0e0000780d000005000000790d0000500a00007a0d0000160d0000c4090000a8610000030000003c000000400d03002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000070000000500000060e3160010023c3218feffffffbfe901020000000006
0710161e000165e1
081016200001020000bca1
091016200001020000b131
0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
0e10162000010200009701
0f10162000010200009a91
```
All the info about the "config fields" is here: [mpp-solar](https://github.com/jblance/mpp-solar/issues/460).

denveronly, 2024-03-01:
Hello,
You are connecting it to UART port of JK (BMS expansion board the two right RJ45 ports, one of them) correct?
Is it possible to receive data about Multiple BMS from a single port the same way as JK Windows Software does Using Device adress list?

Here is a sketch of what i am trying to achieve

[image]

txubelaxu, 2024-03-01:
Yes. The traffic sniffed from the network has the info about every BMS connected to it.
The scheme is perfect.

txubelaxu, 2024-03-03:
There is a new pull request that has the code. https://github.com/syssi/esphome-jk-bms/pull/464

closed 2024-03-17 by txubelaxu