syssi/esphome-jk-bms #477: Can not read data via UART TTL with JKBD6A20S10P Software version 11.42, hardware V11.XW

paulsteigel, 2024-03-14:
Hi All
I tested the [esp32-example.yaml](https://github.com/syssi/esphome-jk-bms/blob/main/esp32-example.yaml) file with esp32S2 mini Lolin and JKBD6A20S10P with software version latter than 11.36 (I tested on 2 different JKs, one with Software version 11.36 and 11.42) Esp32 can not read data from the BMS, it return nan for all sensor.
It might be problem new data protocol!
Can any one be so kind to advise me with thanks
Best

marcjeno1, 2024-04-20:
Hello,

I have the exact same problem. My harware info page says: Hardware Ver: V11.XA and Software Ver: V11.41.

The logging reports only [D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28 over and over.

Now I tried connecting directly to my pc via USB to TTL and noticed 2 settings that I needed to change on my BMS. Otherwise it wouldnt get any data from the BMS via USB.

[image]
[image]

Now with this setting changed to modbus V1.0 I can get data from bms with USB to ttl and see live data on the JK bms monitor software. When I googled this protocol this document showed up: http://www.jk-bms.com/Upload/2023-11-28/1450064500.pdf

Maybe that is helpful?

paulsteigel, 2024-04-21:
Hi @marcjeno1 
Many thanks for sharing, the problem above only happened with 11.XW version 11.36 and 11.42 (as far as I knew with my existing devices). The 11.XA are well connected, no problem at all. With some check I discovered that 11.42 (XW) can still be readable with esp8266 promini D1 using hardware serial port (GPIO1, GPIO3) so I doubt using GPIO9/10 on my Esp32S2 lolin should be problem. I will tried testing when I have the new BMS of those version at hand.
Best

marcjeno1, 2024-04-21:
I am trying to connect it to a esp8266 with no luck. Tried 2 different boards and just doesnt seem to get any data from the BMS. The only thing that works is the usb to ttl to my pc with the JK BMS 485 modbus v1. Bluetooth on my mobile works also fine. 

Any tips on how to maybe read out the bms with esphome jk bms?

paulsteigel, 2024-04-22:
Hi @marjeno1,
The esp8266 promini d1 (12F in fact) connects to BMS via pin 1 and 3 with a pull up on pin 1 to avoid failed booting. The querry command is still the same as what we knew.
[image]
Please try using arduino ide first for parsing the command via uart to check on how it work then! Be minded with baud rate and parity setting for 8266

braincore, 2024-05-03:
Does anyone know if V10 had the "UART1 Protocol No." and "Device Addr" options?

Sleeper85, 2024-05-19:
@marcjeno1 

We would say photos of the new JK-BMS inverter PB series. It is not possible to select the UART or CAN portocole on older JK-BMS.

What is your BMS model?

closed 2024-07-07 by syssi