syssi/esphome-jk-bms #489: Invalid Header Error when discharging

ctaskin, 2024-03-28:
Hi,
First of all thank you for your effort.
Everything is working fine when no load is connected to the battery but it gives Invalid Header Error when I start discharging come current.

My BMS Model is JK_B1A8S20P and I am using ESP32 Dev kit

I am attaching my log, At the top and bottom discharge is disabled, in the middle it is enabled.

It looks like the end of the previous message is added to the start of the next message, and that is causing the problem, may be there is some buffering issue? you can see an example below..

17:56:12][D][uart_debug:114]: <<< _20:55:73:65:72:64:61:4A:4B:5F:42:31:41:B8:53:32:30:50:00:C0:01:00:00:00:00:68:00:00:4B:3F:_**4E:57:00:FD:00:00:00:00:06:**00:01:79:0C:01:0C:AF:02:0C:AE:03:0C:AD:04:0C:AB:80:00:1A:81:00:1A:82:00:19:83:05:11:84:02:F7:85:30:86:02:87:00:00:89:00:00:00:39:8A:00:04:8B:00:00:8C:00:03:8E:05:A0:8F:04:10:90:0E:10:91:0D:DE:92:00:03:93:0A:28:94:0A:5A:95:00:03:96:01:2C:97:00:C8:98:01:2C:99:00:19:9A:00:1E:9B:0B:B8:9C:00:0A:9D:01:9E:00:64:9F:00:50:A0:00:64:A1:00:64

[log.txt](https://github.com/syssi/esphome-jk-bms/files/14791552/log.txt)

syssi, 2024-03-28:
How long is the data cable between the BMS and the ESP?

ctaskin, 2024-03-28:
Not much, 10cm max

syssi, 2024-07-20:
How do you power the ESP? This isn't normal and you are the first user reporting this issue. This could be an hardware issue of your BMS or interferences of your setup and pretty hard to debug here.

closed 2024-08-13 by syssi