syssi/esphome-jk-bms #495: JK-B2A24S15P Many values are 0

Selyria, 2024-04-24:
I have the problem that yesterday I connected to my BMS with my cell phone where the values are correct a few hours later they are correct in the app but with the Esp they are wrong. Many values are 0

[image]

[image]

syssi, 2024-04-24:
Did you set the proper `protocol_version` at your YAML?

Selyria, 2024-04-24:
Ich denke, auf jeden Fall ging es 8 Monate ohne Probleme, dann habe ich die ESP32 getrennt, um einen Blick mit der App zu haben und 1-2 Stunden später bemerkte ich, dass einige Daten auf 0 waren

Selyria, 2024-04-24:
Sorry, the fault lies with me. I still had version 1.5 on it, then disconnected the power from the ESP and then looked at it with the BMS app and then reconnected the ESP to the power supply. After that the values were gone. 

I then thought that the problem was with the ESP, flashed it again with the current version 2.0 and then had the same problems with the values. However, I saw that I now have to use version JK02_32S in the protocol and had JK02 in version 1.5.

Now everything is working again thanks for the tip

closed 2024-04-24 by Selyria