syssi/esphome-jk-bms #499: Sometime the esp32 lost BT connection

testpaul999, 2024-05-02:
Hi,

sometime the esp32 will lost his BT connection to the BMS.
So I have found this code to reset the esp32

online_status:
      name: "status online"                   
      id: bms0_status
      on_release:
        if:
          condition:
            for:
              time: 2min
              condition:
                binary_sensor.is_off: bms0_status
          then:
            - switch.toggle: esp_reset

but this won’t work I see at the logscreen (web ui) that the esp32 try to reconnect but this are not successful.
What I must change here to monitor the BT connection himself?

regards

syssi, 2024-05-03:
What do you mean by "it doesn't work"? Does the ESP reboot but the BLE connection cannot be established again because the BMS rejects the new connection?

testpaul999, 2024-05-03:
> What do you mean by "it doesn't work"? Does the ESP reboot but the BLE connection cannot be established again because the BMS rejects the new connection?

Sorry, the reboot sims not work.
The connectivity los are more them 4h and not 2min.
[image]

testpaul999, 2024-05-12:
No this issue occurs again

Message
21:35:48	[I]	[esp-idf:000]	
[0;33mW (708261604) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:48	[I]	[esp-idf:000]	
[0;33mW (708261721) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:48	[I]	[esp-idf:000]	
[0;33mW (708261829) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:49	[I]	[esp-idf:000]	
[0;33mW (708262248) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e
21:35:49	[I]	[esp-idf:000]	
[0;33mW (708262256) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:49	[W]	[esp32_ble_client:143]	
[0] [C8:47:80:07:A0:CC] Connection failed, status=133
21:35:51	[W]	[jk_bms_ble:238]	
[C8:47:80:07:A0:CC] Not connected
21:35:53	[I]	[esp-idf:000]	
[0;31mE (708266160) BT_BTM: BTM_BleScan scan not ac
21:35:53	[I]	[esp-idf:000]	
[0;33mW (708266166) BT_APPL:  bta_dm_ble_scan stop scan failed, status
21:35:53	[I]	[esp32_ble_client:067]	
[0] [C8:47:80:07:A0:CC] 0x00 Attempting BLE connection
21:35:53	[I]	[esp-idf:000]	
[0;33mW (708266644) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:54	[I]	[esp-idf:000]	
[0;33mW (708268021) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:55	[I]	[esp-idf:000]	
[0;33mW (708268441) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:55	[I]	[esp-idf:000]	
[0;33mW (708268548) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e
21:35:55	[I]	[esp-idf:000]	
[0;33mW (708268552) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:35:55	[W]	[esp32_ble_client:143]	
[0] [C8:47:80:07:A0:CC] Connection failed, status=133
21:35:56	[W]	[jk_bms_ble:238]	
[C8:47:80:07:A0:CC] Not connected
21:36:00	[I]	[esp-idf:000]	
[0;31mE (708273079) BT_BTM: BTM_BleScan scan not ac
21:36:00	[I]	[esp-idf:000]	
[0;33mW (708273083) BT_APPL:  bta_dm_ble_scan stop scan failed, status
21:36:00	[I]	[esp32_ble_client:067]	
[0] [C8:47:80:07:A0:CC] 0x00 Attempting BLE connection
21:36:00	[I]	[esp-idf:000]	
[0;33mW (708273272) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:36:00	[I]	[esp-idf:000]	
[0;33mW (708273365) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:36:00	[I]	[esp-idf:000]	
[0;33mW (708273482) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:36:00	[I]	[esp-idf:000]	
[0;33mW (708273572) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e
21:36:00	[I]	[esp-idf:000]	
[0;33mW (708273576) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
21:36:00	[W]	[esp32_ble_client:143]	
[0] [C8:47:80:07:A0:CC] Connection failed, status=133
21:36:00	[I]	[esp32_ble_client:067]	

a manual reset esp32 works fine but not the automatic one.

regards

syssi, 2024-05-13:
This is probably the most important part of your log:

```
Connection failed, status=133
```

It looks like something of the BLE connection get cached and isn't invalidated on reconnects. If you reboot the device it works because of the empty cache.

testpaul999, 2024-05-13:
Ok and what can I/we do here?
If we can fix the root question about the reset esp32 it my works for me 🤷🏻‍♂️

regards

syssi, 2024-07-13:
Please try this snippet:

```
online_status:
      name: "status online"                   
      id: bms0_status
      on_release:
        if:
          condition:
            for:
              time: 2min
              condition:
                binary_sensor.is_off: bms0_status
          then:
            - button.press: esp_reset

button:
  - platform: restart
    id: esp_reset
    name: "Restart ESP"
```

testpaul999, 2024-07-17:
Thx will try this.
But since 6 weeks no more connection lost‘s.

regards