syssi/esphome-jk-bms #509: Is it compatible with PB BMS version?

sarzyk, 2024-06-19:
Hi is the solution compatible with inverter BMS version like pb2a16s20p?

Thanks

syssi, 2024-06-19:
The short answer is: Yes.

How do you want to talk to your BMS?

sarzyk, 2024-06-19:
I prefer via BT. But I need to read about this because I'm newbie

syssi, 2024-06-19:
You have to use this YAML: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml

Just update the `mac_address` to the one of your BMS and change the `protocol_version` to `JK02_32S`. Make sure there is no other BLE client connected to the BMS (= force-close the Android app). As soon the ESP boots up the BMS should beep and indicate an established connection. HA will auto-discover the ESPHome node if you have the ESPHome integration already enabled. That's all!

sarzyk, 2024-06-19:
I have lilygo can485 to wifi with Bluetooth.
Maybe I can use it?

hungden94, 2024-06-20:
Hello, I'm currently using a PB version bms connect via BLE esp32 but it seem not reliable and some time I also want to use jkbms app on my phone
I want to try comnect by rs485, can you guide me which yaml should I used? How can I connect the cable?

Thank a lot

HA1Andrzej, 2024-06-21:
Hi @syssi ,  super project !!! Good job. 
I have a question for you regarding the communication with the PB series, both over BLE and RS485. 
BLE communication - what does the number of entities detected depend on? I have BMSs where there are 65 entities but I have also had 100 entities. 
RS485 communication - here does the UART and CAN protocol affect the success of the communication? 
Thanks!

syssi, 2024-06-22:
> BLE communication - what does the number of entities detected depend on? I have BMSs where there are 65 entities but I have also had 100 entities.

There is no auto-detection for supported entities/measurements. The number of entities depends on the mentioned sensors at your YAML configuration. Did you remove sensors from the config example?

> RS485 communication - here does the UART and CAN protocol affect the success of the communication?

You can select between different protocols using the Android app. If you want to interface your BMS using Modbus you have to enable this protocol first.

closed 2024-07-07 by syssi