syssi/esphome-jk-bms #510: Other brand bms want to use esphome

shum-elli, 2024-06-23:
I want to adapte lolan bms to esphome，but i meet some problems，Could you help me to analysis

syssi, 2024-06-23:
What do you know about the BMS? Please provide as much details as you can.

shum-elli, 2024-06-23:
[app_unpack.zip](https://github.com/user-attachments/files/15944071/app_unpack.zip)
[wechat_miniprogram_unpack.zip](https://github.com/user-attachments/files/15944072/wechat_miniprogram_unpack.zip)
seller give me two way to view information. I use some unpack tools to unpack it.Seller do some Obfuscation in program. The app is difficult to understand. Wechat Miniprogrammar is easier to understand, but i'm not good at dealing this.I  try to nrf connect to get information directly,but I failed.

syssi, 2024-06-23:
Do you know how to prepare a btsnoop capture of the BLE traffic?

Please give it a try: https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

syssi, 2024-06-23:
Are you able to talk to the BMS using the WeChat examples?

shum-elli, 2024-06-23:
bluetooth log，i have collected.  In the log ,my device name is realmech*****,Bms board name is Baken*****
[btsnoop_hci_20240623_002937.zip](https://github.com/user-attachments/files/15944276/btsnoop_hci_20240623_002937.zip)

syssi, 2024-06-23:
```
-> Write Request (0x12) to handle 0x0011 -> Subscribe notifications
-> Write Command (0x52) to handle 0x0013: c56500bc614e

<- Notification received: 0100030041b8e6b8000000000000000041d59f3041cd0b30429d9320424753ec00343f6e00010027

-> Write Command (0x52) to handle 0x0013: c56500bc614e

<- Notification received: 0100030041b8e75c000000000000000041d59f3041cd10e0429d9320424753ec00343f6f00010027

-> Write Command (0x52) to handle 0x0013: c56500bc614e

<- Notification received: 0100030041b8e70a000000000000000041d5a4e041cd10e0429d9320424753ec00343f7000010027

-> Write Command (0x52) to handle 0x0013: c56500bc614e

<- Notification received: 0100030041b8e666000000000000000041d59f3041cd1680429d9320424753ec00343f7100010027

-> Write Command (0x52) to handle 0x0013: c56500bc614e

<- Notification received: 0100030041b8e615000000000000000041d5997041cd0590429d9320424753ec00343f7300010027
```

[btsnoop-reduced.json](https://github.com/user-attachments/files/15945023/btsnoop-reduced.json)
[btsnoop-reduced.log](https://github.com/user-attachments/files/15945024/btsnoop-reduced.log)

syssi, 2024-06-23:
Could you provide some screenshots of the android app? I would like to know the measurements shown at the app.

shum-elli, 2024-06-23:
Could you please provide a contact for a chat software, similar to Discord? I think it seems more conveniently

shum-elli, 2024-06-23:
this is Wechat MiniProgrammar view，the original is Chinese， i have translated it  to english
[image]