syssi/esphome-jk-bms #511: New JK-BMS inverter missing sensor

grydo, 2024-06-23:
Hello in the yaml current many sensors and switches are missing.

[image]
[image]
[image]

testpaul999, 2024-06-26:
There we will wait for syssi to merge this branch https://github.com/syssi/esphome-jk-bms/tree/add-more-controls to the main version. At this branche the most new switches and sensors are there. :-)

regrads

grydo, 2024-06-26:
what yaml file name?

syssi, 2024-07-07:
This merge request introduces the new tempature sensor: https://github.com/syssi/esphome-jk-bms/pull/525

syssi, 2024-07-08:
The device support is complete. Please re-open the issue if you still miss something!

closed 2024-07-08 by syssi