syssi/esphome-jk-bms #512: Disconnecting from bluetooth

sarzyk, 2024-06-24:
Hi. I bought this esp32 device:

I make everything like in tutorial but I have a problem.
The device randomly disconnected all the time.
What could be a reason this situation?

rt1970, 2024-07-04:
On my ESP32 via BLE too. BMS make frequently a beep...
How can i set MQTT Update-Interval?

syssi, 2024-07-04:
The BMS pushes data every 0.5s. You can use the `throttle` parameter to skip most of the data:

```
jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 60s
```

This setting publishes a set of measurements once per minute.

IMO this setting is useless:

```
    on_passkey_request:
      then:
        - ble_client.passkey_reply:
            id: client0
            passkey: 1234
```

The BLE module of the BMS is unable to do authentication/encryption.

rt1970, 2024-07-05:
Thank you. I have delete Passkey. Works too.

I have many notify in my Log:
[09:49:05][D][jk_bms_ble:525]: Unknown168: 00.00.00.00.00.00.00.00.00.00.00.00.00.00 (14)
[09:49:05][D][jk_bms_ble:539]: Unknown189: 0x00 0x64
[09:49:05][D][jk_bms_ble:543]: Unknown192: 0x00
[09:49:05][D][jk_bms_ble:545]: Unknown193: 0x00 0xFF (0x00 0x8D)
[09:49:05][D][jk_bms_ble:547]: Unknown195: 0x66 0x40 (0x21 0x40)
[09:49:05][D][sensor:094]: 'jk-bms heating current': Sending state 0.00000 A with 2 decimals of accuracy
[09:49:05][I][jk_bms_ble:570]:   Emergency switch: off
[09:49:05][D][sensor:094]: 'jk-bms temperature sensor 4': Sending state 18.70000 °C with 1 decimals of accuracy
[09:49:05][D][sensor:094]: 'jk-bms temperature sensor 3': Sending state 18.60000 °C with 1 decimals of accuracy
[09:49:05][D][esp32_ble_client:110]: [0] [C8:47:80:16:8D:CE] ESP_GATTC_NOTIFY_EVT
[09:49:05][D][esp32_ble_client:110]: [0] [C8:47:80:16:8D:CE] ESP_GATTC_NOTIFY_EVT
[09:49:05][D][esp32_ble_client:110]: [0] [C8:47:80:16:8D:CE] ESP_GATTC_NOTIFY_EVT
[09:49:05][D][esp32_ble_client:110]: [0] [C8:47:80:16:8D:CE] ESP_GATTC_NOTIFY_EVT
[09:49:05][D][esp32_ble_client:110]: [0] [C8:47:80:16:8D:CE] ESP_GATTC_NOTIFY_EVT
[09:49:05][D][esp32_ble_client:110]: [0] [C8:47:80:16:8D:CE] ESP_GATTC_NOTIFY_EVT
[09:49:06][D][esp32_ble_client:110]: [0] [C8:47:80:16:8D:CE] ESP_GATTC_NOTIFY_EVT
[09:49:06][W][component:237]: Component esp32_ble took a long time for an operation (666 ms).
[09:49:06][W][component:238]: Components should block for at most 30 ms.

ESP_GATTC_NOTIFY_EVT are many times in my log of ESPHome

holle75, 2024-07-08:
> The BMS pushes data every 0.5s. You can use the `throttle` parameter to skip most of the data

you mean every 5s (as one would think when seeing 5s) or 0.5s?
5s is fine for me, 0.5s would be overdone. Just asking to clarify.

syssi, 2024-07-08:
Correct. See https://esphome.io/guides/configuration-types#config-time

holle75, 2024-07-08:
ok, **BMS** pushes every 0.5s and with 5s in the yaml we reduce the transmit frequency to 5s. I think i got it.

syssi, 2024-07-08:
Yes. The notification interval of the BMS cannot be changed and is always 0.5s. The data arrives every 0.5s and we discard most of the incoming data using the `throttle` option.

holle75, 2024-07-08:
thank you, Sebastian.

rt1970, 2024-07-08:
Thank you too!

closed 2024-07-13 by syssi