syssi/esphome-jk-bms #515: JK-B2A20S20P-HC temperatures issue

jopl, 2024-06-30:
Hi, I have got an issue with temperatures reading from JK-BMS (JK-B2A20S20P-HC). The ESPHome reads them constantly as 44°C. All other data are perfect. Isn't there any solution for such situation, please?
Kind regards, Josef.

syssi, 2024-06-30:
Which component / connection type do you use? Could you increase the log level and provide a log? I would like to see the raw traffic.

jopl, 2024-06-30:
I am using ESPHome with ESP8266 (ESP-01), directly connected to RX/TX. 2 temperature sensors are connected to appropriate connector on JK/BMS.
It is really interesting ... with "level: VERBOSE" I cannot get any temperature (I can get just voltages) like this:
'''
[16:33:39][I][jk_bms:071]: Status frame received
[16:33:39][V][sensor:043]: 'Nap. článku 1 - BMS01': Received new state 3.276000
[16:33:39][D][sensor:093]: 'Nap. článku 1 - BMS01': Sending state 3.27600 V with 3 decimals of accuracy
[16:33:39][V][sensor:043]: 'Nap. článku 2 - BMS01': Received new state 3.276000
[16:33:39][D][sensor:093]: 'Nap. článku 2 - BMS01': Sending state 3.27600 V with 3 decimals of accuracy
...
'''
With "level: DEBUG":
'''
[16:40:58][D][sensor:093]: 'Max. napětí článku - BMS01': Sending state 3.27600 V with 3 decimals of accuracy
[16:40:58][D][sensor:093]: 'Max. rozdíl na článcích - BMS01': Sending state 0.00300 V with 3 decimals of accuracy
[16:40:58][D][sensor:093]: 'Průměrné napětí článku - BMS01': Sending state 3.27450 V with 3 decimals of accuracy
[16:40:58][D][sensor:093]: 'Teplota - BMS01': Sending state 44.00000 °C with 0 decimals of accuracy
[16:40:58][D][sensor:093]: 'Teplota baterie 1 - BMS01': Sending state 44.00000 °C with 0 decimals of accuracy
'''
I am very sorry for czech language ... "Teplota" = "Temperature"
I hope there will be some local issue, but has got no idea ... :-(

syssi, 2024-06-30:
Please use this configuration:

```
logger:
  baud_rate: 0
  level: DEBUG

uart:
  id: uart_0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
```

and provide a complete log.

jopl, 2024-06-30:
Thanks syssi. I am so sorry, I have got the HW serial port used for JK-BMS communication ... wouldn't be this the problem? Except "debug: direction: BOTH" everzthing is on place ...
I can use just remote logging via WiFi at the moment. If I need the HW port for logging, I have to change configuration, but I am able to do it in 14 days ... I amm on vacation out of home :-(.
Regards, Josef.

jopl, 2024-06-30:
But I am trzing to send you the "complete log" with DEBUG level.
[Log_BMS01.txt](https://github.com/user-attachments/files/16045449/Log_BMS01.txt)
Regards, Josef

syssi, 2024-06-30:
It looks like you haven't enable the `debug` mode of the `uart` component. Could you please update the uart section too and provide another log?

```
uart:
  id: uart_0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
```

The last two lines are the important ones.

jopl, 2024-06-30:
Dear syssi,
I am sorry, I was afraid to allocate serial port to debug, but I did it and everything went.
I am sending the log.
[Log_BMS01.txt](https://github.com/user-attachments/files/16045860/Log_BMS01.txt)
Thx so much.
Regards, Josef.

syssi, 2024-06-30:
Could you guess the current room / battery temperature? It is cold or warm around the battery at the moment? ;-)

jopl, 2024-06-30:
around 20°C, I guess ...

jopl, 2024-06-30:
I asked my friend to measure it ... 24°C.

jopl, 2024-07-01:
Absolutely no change at all, it is constant from beginning.

syssi, 2024-07-01:
If you use the android app there are valid temperature measurements so we can be sure the temperature probes are connected properly?

jopl, 2024-07-01:
It's more complicated ... I'm not able to view temperature(s) in app :-(. Just NA is there :-(
[image]
Regards, Josef.

jopl, 2024-07-01:
I'm using 2 sensors (in one connector), which came in the package with JK-BMS. They are connected to the BMS well (to left upper connector - marked, based on pinout attached), I double checked it.
If you can perceive it as fault, let me know, please.
Thx. Josef.
[image]

syssi, 2024-07-01:
Hmm...  could you ask your seller or manufacturer about this issue? It looks like a hardware issue and cannot be resolved here.

syssi, 2024-07-07:
Let's close this issue because it cannot be resolved here.

closed 2024-07-07 by syssi