syssi/esphome-jk-bms #517: A (maybe|hopefully unqualified) question |  RS485 on new JK inverter bms JKPB2A16SXXP

holle75, 2024-07-06:
Hi, in regards to

https://github.com/syssi/esphome-jk-bms/pull/409
and
https://github.com/syssi/esphome-jk-bms/issues/390

it seems to be (also) about converting a RS485 signal to read data from the new JK inverter bms (JKPB2A16SXXP). The old JK-BMS, as in the readme, using "The UART-TTL (labeled as RS485) socket of the BMS can be attached to any UART pins of the ESP" DIRECTLY.
I´m just wondering why the RS232 Port on the new BMS is not possible to use for a configuration without a RS485 converter module?

I would also be very keen to find a detailed description which parts need to be connected in which way for the new bms and what are the UART settings of the bms are helpfull for (if not to connect something to the RS232 port).
Spent the last 10 hours to follow the description in the readme to find out in the end that attaching my new JK bms needs to happen in a different way ;) 

thank you for any insights

txubelaxu, 2024-07-06:
Hi!!

You could try this one:

https://github.com/txubelaxu/esphome-jk-bms

Regards

holle75, 2024-07-07:
Thank you guys. This one from txubelaxu is about connection via RS485. Thats a viable way to do it. Thanks for that.
Just out of curiosity: why the RS232 port (and the settings behind) can´t be used for a direct connection without the serial to RS485 converter?
Should have asked the initial question in the discussions area. Sorry for that.

syssi, 2024-07-07:
> it seems to be (also) about converting a RS485 signal to read data from the new JK inverter bms (JKPB2A16SXXP). The old JK-BMS, as in the readme, using "The UART-TTL (labeled as RS485) socket of the BMS can be attached to any UART pins of the ESP" DIRECTLY.

It's hard to explain but I will give it a try. :-) If you disassemble the initial/old hardware model and trace the wires of the RS485 port you will find an optical isolator + the microcontroller. Let's ignore the optical isolator. In other words: The RS485 port is labeled "RS485" but it's directly connected to the UART port of the microcontroller. The logic level is 3.3V this is why we are talking about UART-TTL here.

If you want to interface the BMS with another RS485 device you have to convert the signal first using the Dongle of the manufacturer: https://github.com/syssi/esphome-jk-bms/blob/main/images/rs485-adapter/001.jpg

Interfacing the old BMS would look like this:

```
[JK-BMS JK-BD/JK-B] <-- UART-TTL --> [OEM RS485 Dongle] <-- RS485-Signal --> [RS485 port of another device]
```

> I´m just wondering why the RS232 Port on the new BMS is not possible to use for a configuration without a RS485 converter module?

Let's talk about your BMS model now. And be careful there is a important difference between RS232 & RS485.

syssi, 2024-07-07:
If we disassemble your BMS model there will be a RS485 converter on the mainboard of the BMS. The microcontroller will be attached the this RS485 converter and the converter is attached to the RS485 of the BMS. The RS485 transforms the UART-TTL signal of the microcontroller to an differential signal with a new voltage level (-7V / +12V).

The ESP cannot handle RS485 signals directly. A RS485 converter is needed to transform the signal back to the 3.3V logic level (UART-TTL). 

The sample applies for RS232. A RS232 (MAX3232) converter board is needed if you want to talk devices using RS232. 

Did this answer your question?

holle75, 2024-07-07:
Ah, the (not only) voltage Thing ;)
I´m no expert, Probably thats why i was confused. Thank you very much syssi for taking the time to explain (and thank you very much for the whole project).
I will first try the BT way now. Cabling is no fun at all in the place where my battery is stored. I hope it´s stable enough.

... a little side question where i couldn´t find any answer on the Net: In the settings of the new inverter BMS you can set different protocols for UART1 and UART2. I´m wondering for which port exactly are those for?
[image]

txubelaxu, 2024-07-07:
Another important thing is that in RS232 can be only one (slave) BMS. In a
RS485 network can be multiple (slave) BMSs connected.

RS232 has a length limit of about 12m. RS485 has a higher limit.

I tried with RS232 multiple times and I got nothing from that port.

syssi, 2024-07-07:
> ... a little side question where i couldn´t find any answer on the Net: In the settings of the new inverter BMS you can set different protocols for UART1 and UART2. I´m wondering for which port exactly are those for?

Good question. I don't know it. This is how it looks like at the windows application:

[image]

May be @txubelaxu is able to answer the question. This is my guess:

UART2 is attached to the RS485 and cannot be changed because the Modbus Protocol is expected here (always!).
UART1 is attached to the RS232 and you can change the protocol here. If you choose the GPS protocol you have to use this [YAML example](https://github.com/syssi/esphome-jk-bms/blob/main/esp32-example.yaml). If you choose Modbus here too you have to use this [YAML example](https://github.com/syssi/esphome-jk-bms/blob/main/esp32-jk-pb-modbus-example.yaml) and you have to attach a RS232 converter to your ESP.

holle75, 2024-07-07:
i would be interested too in a clarification. But it seems the RS232 port won´t give any advantages in using instead of RS485.

syssi, 2024-07-07:
It depends what you try to achieve and on the complexity of your setup. Where are you from?

holle75, 2024-07-07:
The solar plant (Studer innotec) is in italy. Small test plant 2.4kwp|7KWh battery| 24V ... i´m planning on a bigger one and thats why i got into LiFePo and the technical challenges around. After a lot of initial hickups making the plant talk to the battery and especially working well together (switched a first tried seplos to the JK) now i´m at a point where i can start to optimize. Last little problem is Current reduction when end of charge is around the corner. When i can read the cell voltages (especially max cell voltage) i can set charge current via fhem on the solar charger.

syssi, 2024-07-07:
If you have multiply battery banks they will communicate via RS485 probably. One BMS will be the Modbus master and it will collect periodically data from the slaves. @txubelaxu solution is able to sniff this communication and expose the data as sensor entities. Because the ESP isn't the Modbus Master in this setup *listening only* is allowed/possible here.

If you want to change the settings of your BMS remotely some day you will attach an ESP to the RS232 port probably because you are allowed to act as Modbus Master here.

holle75, 2024-07-07:
Ok, RS232 became interesting again ;)

holle75, 2024-07-07:
I was flashing the BT way in the meantime. Finding out the MAC address of the BMS was a bit complicated but i found information in one of the threads here how to do it. 

This works pretty well. I´m amazed.

Now i have to make the ESP talk to MQTT in fhem.

There are some (minor) error messages in ESPHome:

11:21:23 | [W] | [component:237] | Component esp32_ble took a long time for an operation (155 ms).
-- | -- | -- | --
11:21:23 | [W] | [component:238] | Components should block for at most 30 ms.

and some information (temperature_sensor_3, temperature_sensor_4) is missing. But hey, who´s interested in that.

syssi, 2024-07-07:
The warning can be ignored.

holle75, 2024-07-07:
Wow, MQTT autodetect in fhem works pretty fine. Now i have all info in fhem. This was a fast one. Let the game begin ;)

holle75, 2024-07-07:
sensor_heating | ON

is reported wrongly.

syssi, 2024-07-07:
How many temperature probes are attached to your BMS?

syssi, 2024-07-07:
> sensor_heating | ON
> is reported wrongly.

Which `protocol_version` do you use?

holle75, 2024-07-07:
... to be more precise
jk_bms_XXXXXXXX:jk-bms/binary_sensor/jk-bms_heating/state
is reported wrongly (also in ESPHome Dashboard)

there are 4 temperature probes attached to the bms and reported accurately in the APP
protocol_version: JK02_32S

syssi, 2024-07-07:
Could you create an issue per finding? I would like to track each issue properly.

holle75, 2024-07-07:
yes. see you there

syssi, 2024-07-07:
Lets close this issue because the initial question is resolved.

closed 2024-07-07 by syssi