syssi/esphome-jk-bms #518: new JK inverter BMS JKPB2A16SXXP | Heating binary sensor state reported wrongly

holle75, 2024-07-07:
https://github.com/syssi/esphome-jk-bms/issues/517#issuecomment-2212387860

MQTT 
jk_bms_XXXXXXXX:jk-bms/binary_sensor/jk-bms_heating/state
is reported wrongly (also in ESPHome Dashboard)

holle75, 2024-07-07:
Mmh, now the heating state is reported correctly as OFF. Don´t ask my why.
BUT the switch heating from settings is not shown (and wasn´t before)

i have to refer to the two other issues i opened. I´m new to ESPHome and might made mistakes. This is the yaml file i used. Are all settings/devices are even in here?

`substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:33:10:33:4a
  # Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.0.0

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: ssidxxx
  password: passowrdxxx
#Optional manual IP - damit gelingt ein wireless update in Home Assistant und in ESPHome
  manual_ip:
    static_ip: 192.168.1.60
    gateway: 192.168.1.1
    subnet: 255.255.255.0

web_server:
  port: 80

ota:
  platform: esphome
  password: passwordxxx
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - logger.log: "BLE connection suspended for OTA update"

logger:
  level: INFO

mqtt:
  broker: 192.168.1.55
  port: 13
  username: userxxx
  password: passwordxxx
  id: MQTT2_JK_BMS_Container_ESP32

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0 || x.get_name().rfind("JK_", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
[... 133 lines cut ...]
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"`

syssi, 2024-07-07:
Which example configuration YAML did you choose as starting point? This is the one which fits best with your device version: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v14-example.yaml

holle75, 2024-07-07:
YES, it seems i found the "wrong" yaml file at the beginning (following an older link here in the threads) ;)

holle75, 2024-07-07:
ps: seems also to work for Hardware version 15.x

syssi, 2024-07-07:
I cannot reproduce the issue. The `heating binary sensor`, `heating current` and `heating switch` are working in general. Let's close the issue for now. Feel free to re-open the issue if it's happening again.

closed 2024-07-07 by syssi

holle75, 2024-07-07:
Yes, you can´t reproduce the issue because i used the wrong yaml file.
Thats what i meant and that everything is working now also with hardware version 15.x