syssi/esphome-jk-bms #519: new JK inverter BMS JKPB2A16SXXP | temp sensors 3 and 4 not reported

holle75, 2024-07-07:
https://github.com/syssi/esphome-jk-bms/issues/517#issuecomment-2212388919

temp sensors 3 and 4 (i suppose) are not registered/communicated

syssi, 2024-07-07:
Could you increase the log level to `VERY_VERBOSE` and provide some raw data of the `cell info frame`?

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L329-L331

```
logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: VERY_VERBOSE
    jk_bms_ble: VERY_VERBOSE
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

holle75, 2024-07-07:
Yes, i will.
I´m new to ESPHome, so please bear with me.
Is there any way i can change those settings via OTA? (i never did OTA so not even know what is needed to do so) because the ESP32 is screwed down in a little case close to the solar plant. Wifi yes, easy taking the ESP32 out for flashing, no ;)

holle75, 2024-07-07:
Oh, hold on. On my search for OTA i looked into the yaml file i used for flashing. I don´t know why but two temp sensors seems to  be marked as "do not use"

    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    # temperature_sensor_3:
    #   name: "${name} temperature sensor 3"
    # temperature_sensor_4:
    #   name: "${name} temperature sensor 4"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"

holle75, 2024-07-07:
Wow, i´m embarrassed and amazed the same time. I figured out how OTA works (why is it always the hardest part when entering an area you have no expertise at all to find out the most simplest basics?)
uncommented the two sensors in the yaml
and they are shown in the dashboard now
So the issue is no issue after all.

For followers: OTA works the same as flashing on your computer plus the IP address (probably more complicated than that, but it works in my case)
esphome run mydevice.yaml --device 192.168.1.123

syssi, 2024-07-07:
Perfect! Let's close this issue. :-)

closed 2024-07-07 by syssi